framps/raspiBackup #129: Default Logfile Location

DocYester, 2019-04-28:
Shouldnt the default logfile location be next to the backup's ?
In case the system that made the backup crashes completly and the backup is faulty in some parts or any other failure it would be nice to see the logfile.
Or even when when it crashes during backing up.
So when the backup is placed in /mnt/remote-fs/\<hostname>/\<hostname>-\<backuptype>-\<date>/ 
the logfile would be best in /mnt/remote-fs/\<hostname>/Backup_\<hostname>_\<date>.log
in my opinion.
The logfiles should be deleted in same style like the backups.

framps, 2019-04-28:
I agree with you. Therefore next release 0.6.4.3 which is already [beta](https://github.com/framps/raspiBackup/tree/beta) has the whole logging restructured 

1. Default is to create a debug log all the time ( -l debug)
2. Default log location is in the backup directory (-L backup). If backup fails and the backup directory will be cleaned up the logfiles are saved in /root directory.

See details in closed issues #120, #119, #116

framps, 2019-05-04:
@DocYester Do you have any update on this issue. Otherwise please close.

closed 2019-05-05 by DocYester