framps/raspiBackup #130: Missing option -a and -o in v0.6.4.3

KiefDelicious, 2019-05-01:
Hi,

I've updated from v0.6.4.2 to v0.6.4.3 and the backups aren't working when no parameters are set with -a and -o. When I revert back to v0.6.4.2 it's working again.

> pi@domoticz:/mnt/backup $ sudo /usr/local/bin/raspiBackup.sh -a ":" -o ":"
> --- RBK0009I: domoticz: raspiBackup.sh V0.6.4.3-beta (74e3e4a) started at Wed May  1 08:29:18 CEST 2019.
> !!! RBK0165W: =========> NOTE  <=========
> !!! RBK0165W: This is a betaversion and should not be used in production.
> !!! RBK0165W: =========> NOTE <=========
> ??? RBK0019E: Missing option -a and -o.
> --- RBK0033I: Please wait until cleanup has finished.
> --- RBK0105I: Deleting new backup directory /mnt/backup/domoticz/domoticz-rsync-backup-20190501-082918.
> rm: cannot remove '/mnt/backup/domoticz/domoticz-rsync-backup-20190501-082918': Directory not empty
> ??? RBK0005E: Backup failed. Check previous error messages for details.
> --- RBK0010I: domoticz: raspiBackup.sh V0.6.4.3-beta (74e3e4a) stopped at Wed May  1 08:29:22 CEST 2019.

framps, 2019-05-01:
Thank you very much for testing the beta. You found a valid bug I introduced the last couple of days when I detected an issue with parameters starting with à dash :cry: 
It's fixed now. Please verify and close the issue.

KiefDelicious, 2019-05-05:
Hi, sorry for the late reply. Thanks for fixing, I've tested the new version and the error is gone. Seems like it's working fine.

commit 52e63ce1 (2019-05-01): Fixed #130
commit 8e935813 (2019-05-01): Fixed #130
closed 2019-05-05 by KiefDelicious