framps/raspiBackup #132: TAR Backup fails

SAS-1, 2019-05-01:
Hi 

I have raspibackup running on 2 PI's both running Debian, however one of them is failing. 

The log shows the following:

`--- RBK0151I: Using backuppath /media/Backups/pibackups.
--- RBK0036I: Saving partition layout.
47+1 records in
47+1 records out
50028544 bytes (50 MB, 48 MiB) copied, 6.4088 s, 7.8 MB/s
1+0 records in
1+0 records out
512 bytes copied, 0.000928225 s, 552 kB/s
--- RBK0158I: Creating native tar backup "/media/Backups/pibackups/PINAME/PINAME-tar-backup-20190501-091703/PINAME-tar-backup-20190501-091703.tar".
--- RBK0085I: Backup of type tar started. Please be patient.
tar: Removing leading `/' from member names
tar: Removing leading `/' from hard link targets
tar: /var/lib/samba/winbindd_privileged/pipe: socket ignored
tar: /var/lib/samba/private/msg.sock/693: socket ignored
tar: /var/lib/samba/private/msg.sock/1312: socket ignored
tar: /var/lib/samba/private/msg.sock/599: socket ignored
tar: /var/lib/samba/private/msg.sock/687: socket ignored
tar: /var/lib/samba/private/msg.sock/20506: socket ignored
tar: /var/lib/samba/private/msg.sock/646: socket ignored
tar: /var/lib/samba/private/msg.sock/465: socket ignored
tar: /var/lib/samba/private/msg.sock/20458: socket ignored
tar: /var/lib/samba/private/msg.sock/627: socket ignored
tar: /var/lib/samba/private/msg.sock/20380: socket ignored
tar: /var/lib/samba/private/msg.sock/702: socket ignored
tar: /var/lib/samba/private/msg.sock/638: socket ignored
tar: /var/lib/samba/private/msg.sock/20459: socket ignored
tar: /var/lib/samba/private/msg.sock/20465: socket ignored
tar: /var/lib/samba/private/msg.sock/575: socket ignored
tar: /var/lib/samba/private/msg.sock/776: socket ignored
tar: /var/lib/samba/private/msg.sock/20462: socket ignored
tar: /var/lib/samba/private/msg.sock/652: socket ignored
tar: /var/lib/samba/private/msg.sock/684: socket ignored
tar: /var/lib/samba/private/msg.sock/20463: socket ignored
tar: /var/lib/samba/private/msg.sock/608: socket ignored
tar: /var/lib/samba/private/msg.sock/20381: socket ignored
tar: /var/lib/samba/private/msg.sock/701: socket ignored
tar: /var/lib/samba/private/msg.sock/20457: socket ignored
tar: /var/lib/samba/private/msg.sock/683: socket ignored
tar: Exiting with failure status due to previous errors
??? RBK0021E: Backupprogram for type tar failed with RC 2.
--- RBK0033I: Please wait until cleanup has finished.
Invocation parms: ' -l debug'
`

if I run it with a v parameter, then I can see that the last folder is /Run 

`/run/
tar: Exiting with failure status due to previous errors
??? RBK0021E: Backupprogram for type tar failed with RC 2.
--- RBK0033I: Please wait until cleanup has finished.
Invocation parms: ' -v'`

I compared the config files between the 2 PI's and they are identical. 

Any ideas what I am missing?
Thanks

framps, 2019-05-01:
```
tar RCs:

0 ‘Successful termination’.  
1 ‘Some files differ’. If tar was invoked with ‘--compare’ (‘--diff’,  ‘-d’) command line option, this means that some files in the archive  differ from their disk counterparts (see section Comparing Archive  Members with the File System). If tar was given ‘--create’, ‘--append’  or ‘--update’ option, this exit code means that some files were changed  while being archived and so the resulting archive does not contain the  exact copy of the file set.  
2 ‘Fatal error’. This means that some fatal, unrecoverable error occurred.
```   
Unfortunately that's a very unspecific error from tar used by raspiBackup :disappointed:  

tar usually terminates with RC1 if some file was updated during backup processing. That's not RC2 but please make sure you stopped all services which modify any files. It looks like samba is still active.

Please see [FAQ18](https://www.linux-tips-and-tricks.de/en/faq/#a18)

SAS-1, 2019-05-01:
hi framps,

thanks, The interesting thing is I don't have samba enabled and it's not running :)

is there any other logging I can try that will give me a hint as to what is wrong?

framps, 2019-05-01:
Unfortunately this is a nasty tar error and it's not easy to debug the root cause :disappointed: 

If you enable the debug log with `-l debug` and search for `Command:` in the debug log you will find the tar command executed by raspiBackup which fails with RC2. Just execute this command when you try to debug the issue. You don't need raspiBackup for this.

In order to debug tar the only existing option is -v - unfortunately.

That's what I suggest to do:
1) If you use a remote backup partition: Try to create the backup on a local attached partition, e.g. USB stick to check whether it's related to the remote connection
2) If you use a local backup partition, try another one to check whether the backup device is buggy
3) ... search in the net and ask the community for help

Hope this helps

framps, 2019-05-02:
@SAS-1 Please see [FAQ19](https://www.linux-tips-and-tricks.de/en/faq/#a18) which may also the root cause for your issue if your tar backup becomes greater than 4GB.

SAS-1, 2019-05-02:
Hi Framps, it is around 5GB for the backup, the same size it's been for months now which is strange as nothing other than the version of the backup works. 

I am going to roll back the raspibackup to an earlier version this weekend and see if that works. 

Thanks

framps, 2019-05-02:
I don't think an older version of raspiBackup will solve the issue. It's a tar issue - not a raspiBackup issue. Maybe you use different tar versions and one is buggy? Just guessing ... Check with `tar --version`.

SAS-1, 2019-05-05:
Hi Framps, 

so I restored version 0.6.1.3b which is the earliest I have and the backup worked!

I rolled it forward and again it worked so I have no idea what happened but it's working for now. 

Thanks

closed 2019-05-05 by SAS-1