framps/raspiBackup #133: -0 option does not work in beta 6.4.3

rosch99, 2019-05-02:
i am trying to restore an rsync backup to a smaller sdcard.
OS is raspbian stretch with latest updates.
source (backup) is on a 120GB SSD --> /dev/sda
target is a 64GB sdcard --> /dev/mmcblk0 
Unfortunately neither the "automatic" resizing function, nor the -0 option does work,
i always get the following error messages.

-------------------------------------
root@loxberry:~# raspiBackup.sh -0 -d /dev/mmcblk0 /backup/loxberry/loxberry-rsync-backup-20190427-221539/
--- RBK0009I: loxberry: raspiBackup.sh V0.6.4.3-beta (4b891fe) Don Mai  2 20:40:47 CEST 2019 gestartet.
!!! RBK0165W: =========> HINWEIS <=========
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte.
!!! RBK0165W: =========> HINWEIS <=========
sda1
sda2
!!! RBK0144W: Ziel /dev/mmcblk0 wird nicht partitioniert. Existierende Partitionen werden benutzt.
??? RBK0160E: Ziel /dev/mmcblk0 mit 59.47 GiB ist kleiner als die Backupquelle mit 119.24 GiB.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft mit RC 102 beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: loxberry: raspiBackup.sh V0.6.4.3-beta (4b891fe) Don Mai  2 20:40:48 CEST 2019 beendet.

keep on the good work 
Robert

framps, 2019-05-02:
RBK0160E is written when a partition oriented backup is restored (you used option -P). raspiBackup doesn't support automatic resizing for partitioned backups. 

Do you really need the partition oriented backup? As far as I know Loxberry can be saved with raspiBackup in the normal mode by not using option -P because it just uses the standard Raspbian partitioning. Then automatic resizing will work during restore.

rosch99, 2019-05-02:
thanks for the quick response!

there are 3 partitions on /dev/sda but i want to backup only sda1 and sda2, 
so i set --> DEFAULT_PARTITIONS_TO_BACKUP="1 2"
but i also have to set --> DEFAULT_PARTITIONBASED_BACKUP=1
otherwise i get the following error:

??? RBK0013E: Es existieren mehr als zwei Partitionen, die nur mit dem Backuptype DD oder DDZ oder der Option -P gesichert werden können.

rosch

framps, 2019-05-02:
If you have more than 2 partitions you have to configure raspiBackup the way you did. But even you backup only 2 partitions raspiBackup considers this backup as a partition oriented backup and thus there is no automatic resizing possible. That's a restriction of raspiBackup.

The only way I see to get a Loxberry backup created with raspiBackup to restore on a smaller device is by deleting the third partition on the source system. 

But ... when you use option -0 and you partitioned the target device accordingly the existing partitioning should be used and the actual size of the target should be ignored. That's a bug in raspiBackup.

Are you willing to test a fix I will create? I don't have any image I can test the fix.

rosch99, 2019-05-02:
do i get it right?
Lets say: sda1=100MB, sda2=60GB, sda3=60GB -> target mmcblk0p1=100MB, mmcblk0p2=30GB  should work with -0 but recent version has a bug?
 
anyway, i would be happy to test a fix.

framps, 2019-05-04:
I just created a [fix in this branch](https://github.com/framps/raspiBackup/tree/beta_133) for you which now doesn't check for the correct size if option -0 is used.
Just grab `raspiBackup.sh` from the branch and try restore again to test the fix :smiley:

rosch99, 2019-05-05:
thanks for the quick fix!
Unfortunately it didnt work out, there seems to be a problem with the creation of the restore command.
-----------
/usr/local/bin/raspiBackup.sh: Zeile 922: MSG_DE[]: Falsche Variablenersetzung.
------------

 

root@loxberry:/usr/local/bin# raspiBackup.sh -0 -g -d /dev/mmcblk0 /backup/loxberry/loxberry-rsync-backup-20190
loxberry-rsync-backup-20190427-221539/ loxberry-rsync-backup-20190502-211927/ loxberry-rsync-backup-20190505-124721/
root@loxberry:/usr/local/bin# raspiBackup.sh -0 -g -d /dev/mmcblk0 /backup/loxberry/loxberry-rsync-backup-20190505-124721/
--- RBK0009I: loxberry: raspiBackup.sh V0.6.4.3-beta (13071f8) Son Mai  5 12:50:40 CEST 2019 gestartet.
!!! RBK0165W: =========> HINWEIS <=========
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte.
!!! RBK0165W: =========> HINWEIS <=========
sda1
sda2
!!! RBK0144W: Ziel /dev/mmcblk0 wird nicht partitioniert. Existierende Partitionen werden benutzt.
**/usr/local/bin/raspiBackup.sh: Zeile 922: MSG_DE[]: Falsche Variablenersetzung.**

--- RBK0067I: Momentane Partitionen auf /dev/mmcblk0:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  47,7MB   43,5MB   primary  fat32        lba
 2      48,2MB  63865MB  63816MB  primary  ext4
!!! RBK0100W: Gerät /dev/mmcblk0 wird mit dem Backup beschrieben.
--- RBK0038I: Bist Du sicher? j/N j
--- RBK0009I: loxberry: raspiBackup.sh V0.6.4.3-beta (13071f8) Son Mai  5 12:50:58 CEST 2019 gestartet.
/usr/local/bin/raspiBackup.sh: Zeile 922: MSG_DE[]: Falsche Variablenersetzung.
--- RBK0095I: Backup wird auf Partition /dev/mmcblk0p1 zurückgespielt.
rsync  version 3.1.2  protocol version 31
Copyright (C) 1996-2015 by Andrew Tridgell, Wayne Davison, and others.
Web site: http://rsync.samba.org/
Capabilities:
    64-bit files, 64-bit inums, 32-bit timestamps, 64-bit long ints,
    socketpairs, hardlinks, symlinks, IPv6, batchfiles, inplace,
    append, ACLs, xattrs, iconv, symtimes, prealloc

rsync comes with ABSOLUTELY NO WARRANTY.  This is free software, and you
are welcome to redistribute it under certain conditions.  See the GNU
General Public Licence for details.

rsync is a file transfer program capable of efficient remote update
via a fast differencing algorithm.

Usage: rsync [OPTION]... SRC [SRC]... DEST
  or   rsync [OPTION]... SRC [SRC]... [USER@]HOST:DEST
  or   rsync [OPTION]... SRC [SRC]... [USER@]HOST::DEST
  or   rsync [OPTION]... SRC [SRC]... rsync://[USER@]HOST[:PORT]/DEST
  or   rsync [OPTION]... [USER@]HOST:SRC [DEST]
  or   rsync [OPTION]... [USER@]HOST::SRC [DEST]
  or   rsync [OPTION]... rsync://[USER@]HOST[:PORT]/SRC [DEST]
The ':' usages connect via remote shell, while '::' & 'rsync://' usages connect
to an rsync daemon, and require SRC or DEST to start with a module name.

Options
 -v, --verbose               increase verbosity
     --info=FLAGS            fine-grained informational verbosity
     --debug=FLAGS           fine-grained debug verbosity
     --msgs2stderr           special output handling for debugging
 -q, --quiet                 suppress non-error messages
     --no-motd               suppress daemon-mode MOTD (see manpage caveat)
 -c, --checksum              skip based on checksum, not mod-time & size
 -a, --archive               archive mode; equals -rlptgoD (no -H,-A,-X)

framps, 2019-05-05:
:cry: 

I need a debug log which is create with option `-l debug`. Please either attach it here, upload it somewhere in the net and add a link or send it via eMail to me (See my [Contact page](https://www.linux-tips-and-tricks.de/en/contact/) for the eMail address)

rosch99, 2019-05-05:
restore is ok without the  -g parameter, although the error "Falsche Variablensetzung" is displayed.
So, it looks like a problem with the combination of arguments.
(i will send the logfile via email)

framps, 2019-05-07:
I received a debug log and detected two issues. I fixed them and send a fixed version for verification to @rosch99 .

framps, 2019-05-09:
@rosch99 verified the fix and it's merged into the beta branch.

@rosch99 Thank you very much for you patience on getting this issue fixed :smiley:

closed 2019-05-09 by framps