framps/raspiBackup #134: Option -F disables option -L

DocYester, 2019-05-03:
If you set option -F (Fake Backup) for testing purposals hte option -L (manual Logfile) gets partly ignored/disabled

**root@piDC:/mnt/backup/rsp-backup/piDC# raspiBackup.sh -F -v -l debug -L /mnt/backup/rsp-backup/`hostname`/log/Backup_`date '+%Y-%m-%d_%H-%M'`.log**
05-03-2019 20:52:29 --- RBK0009I: piDC: raspiBackup.sh V0.6.4.2 (74411c9) Fr 3. Mai 20:52:29 CEST 2019 gestartet.
**05-03-2019 20:52:31 --- RBK0128I: Logdatei ist /mnt/backup/rsp-backup/piDC/raspiBackup.log.**
05-03-2019 20:52:31 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
05-03-2019 20:52:33 !!! RBK0124W: Simulationsmodus an.
05-03-2019 20:52:34 --- RBK0151I: Backuppfad /mnt/backup/rsp-backup wird benutzt.

**root@piDC:/mnt/backup/rsp-backup/piDC# raspiBackup.sh -v -l debug -L /mnt/backup/rsp-backup/`hostname`/log/Backup_`date '+%Y-%m-%d_%H-%M'`.log**
05-03-2019 20:54:35 --- RBK0009I: piDC: raspiBackup.sh V0.6.4.2 (74411c9) Fr 3. Mai 20:54:35 CEST 2019 gestartet.
**05-03-2019 20:54:38 --- RBK0128I: Logdatei ist /mnt/backup/rsp-backup/piDC/log/Backup_2019-05-03_20-54.log.**
05-03-2019 20:54:39 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
05-03-2019 20:54:41 --- RBK0151I: Backuppfad /mnt/backup/rsp-backup wird benutzt.

framps, 2019-05-04:
I just create a fix in [this branch](https://github.com/framps/raspiBackup/commit/aa13bd91500940435b43106e0da79af4e0290ab7). Please verify.

DocYester, 2019-05-05:
Verified 👍 
root@mail:~# ./raspiBackup.sh  -F -v -l debug -L /mnt/backup/rsp-backup/`hostname`/log/Backup_`date '+%Y-%m-%d_%H-%M'`.log
05-05-2019 10:09:48 --- RBK0009I: mail: raspiBackup.sh V0.6.4.2 (f5a5309) So 5. Mai 10:09:48 CEST 2019 gestartet.
05-05-2019 10:09:49 --- RBK0128I: Logdatei ist /root/raspiBackup.log.
05-05-2019 10:09:49 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
05-05-2019 10:09:49 --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
05-05-2019 10:09:50 --- RBK0168I: :-D raspiBackup.sh Beta Version 0.6.4.3 ist verfügbar. Hilfe beim Testen dieser Beta ist sehr willkommen. Einfach auf die neue Beta Version mit der Option -U upgraden. Die vorhergehende Version kann mit der Option -V wiederhergestellt werden
05-05-2019 10:09:50 --- RBK0114I: Besuche https://www.linux-tips-and-tricks.de/de/versionshistorie/ um die Änderungen in der neuen Version kennenzulernen.
05-05-2019 10:09:50 !!! RBK0124W: Simulationsmodus an.
05-05-2019 10:09:50 --- RBK0151I: Backuppfad /mnt/backup/rsp-backup wird benutzt.
05-05-2019 10:09:50 --- RBK0008I: Services werden gestoppt: 'service cron stop && service dropbear stop && service lighttpd stop && service php7.0-fpm stop && service mariadb stop'.
05-05-2019 10:09:55 --- RBK0081I: Backup vom Typ rsync wird in /mnt/backup/rsp-backup/mail/mail-rsync-backup-20190505-100948 erstellt.
05-05-2019 10:09:55 --- RBK0036I: Partitionslayout wird gesichert.
05-05-2019 10:09:55 --- RBK0133I: Verzeichnis /mnt/backup/rsp-backup/mail/mail-rsync-backup-20190504-105634 wird für Hardlinks benutzt.
05-05-2019 10:09:55 --- RBK0158I: rsync Backup "/mnt/backup/rsp-backup/mail/mail-rsync-backup-20190505-100948" wird erstellt.
05-05-2019 10:09:55 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
05-05-2019 10:09:55 --- RBK0078I: Backupzeit: 00:00:00.
05-05-2019 10:09:56 --- RBK0007I: Services werden gestartet: 'service mariadb start && service php7.0-fpm start && service lighttpd start && service dropbear start && service cron start'.
05-05-2019 10:10:00 --- RBK0032I: Bitte warten bis aufgeräumt wurde.
05-05-2019 10:10:00 --- RBK0017I: Backup erfolgreich beendet.
05-05-2019 10:10:00 --- RBK0026I: Logdatei wurde in /root/raspiBackup.log gesichert.
05-05-2019 10:10:00 --- RBK0010I: mail: raspiBackup.sh V0.6.4.2 (f5a5309) So 5. Mai 10:10:00 CEST 2019 beendet.

But found another funny "Bug" that isnt documented -> Script uses its own name as Config-Filename - it's intended to work like this ?
Like this its easy to keep and use diffrent config's - but testing with other script names will allways revert to default options

root@mail:~# ./test.sh  -F -v -l debug -L /mnt/backup/rsp-backup/`hostname`/log/Backup_`date '+%Y-%m-%d_%H-%M'`.log
--- RBK0009I: mail: test.sh V0.6.4.2 (f5a5309) So 5. Mai 10:09:04 CEST 2019 gestartet.
--- RBK0128I: Logdatei ist /root/test.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/test.conf wird benutzt.
!!! RBK0124W: Simulationsmodus an.
??? RBK0027E: Kein externes Gerät an /backup verbunden. Die SD Karte würde für das Backup benutzt werden.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
--- RBK0026I: Logdatei wurde in /root/test.log gesichert.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: mail: test.sh V0.6.4.2 (f5a5309) So 5. Mai 10:09:06 CEST 2019 beendet.

DocYester, 2019-05-05:
`
20190505-100905: DBG  -- LOG_OUTPUT: 3
20190505-100905: DBG  -- Using logfile /root/test.log
20190505-100905: DBG  -- --- RBK0009I: mail: test.sh V0.6.4.2 (f5a5309) So 5. Mai 10:09:05 CEST 2019 gestartet.
20190505-100905: DBG  -- BACKUPTARGET_DIR: /backup/mail/mail-dd-backup-20190505-100904
20190505-100905: DBG  -- BACKUPTARGET_FILE: /backup/mail/mail-dd-backup-20190505-100904/mail-dd-backup-20190505-100904.img
20190505-100905: DBG     >> logOptions
20190505-100905: DBG     -- Linux mail 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
20190505-100905: DBG     -- Options:  -F -v -l debug -L /mnt/backup/rsp-backup/mail/log/Backup_2019-05-05_10-09.log
20190505-100905: DBG     -- APPEND_LOG=0
20190505-100905: DBG     -- APPEND_LOG_OPTION=-a
20190505-100905: DBG     -- BACKUPPATH=/backup
20190505-100905: DBG     -- BACKUPTYPE=dd
20190505-100905: DBG     -- CHECK_FOR_BAD_BLOCKS=0
20190505-100905: DBG     -- CONFIG_FILE=
20190505-100905: DBG     -- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20190505-100905: DBG     -- DD_BLOCKSIZE=1MB
20190505-100905: DBG     -- DD_PARMS=
20190505-100905: DBG     -- DEPLOYMENT_HOSTS=
20190505-100905: DBG     -- YES_NO_RESTORE_DEVICE=loop
20190505-100905: DBG     -- EMAIL=
20190505-100905: DBG     -- EMAIL_PARMS=
20190505-100905: DBG     -- EXCLUDE_LIST=
20190505-100905: DBG     -- EXTENSIONS=
20190505-100905: DBG     -- FAKE=1
20190505-100905: DBG     -- HANDLE_DEPRECATED=1
20190505-100905: DBG     -- KEEPBACKUPS=3
20190505-100905: DBG     -- LANGUAGE=DE
20190505-100905: DBG     -- LINK_BOOTPARTITIONFILES=0
20190505-100905: DBG     -- LOG_LEVEL=1
20190505-100905: DBG     -- LOG_OUTPUT=3
20190505-100905: DBG     -- MAIL_ON_ERROR_ONLY=0
20190505-100905: DBG     -- MAIL_PROGRAM=mail
20190505-100905: DBG     -- MSG_LEVEL=1
20190505-100905: DBG     -- NOTIFY_UPDATE=1
20190505-100905: DBG     -- PARTITIONBASED_BACKUP=0
20190505-100905: DBG     -- PARTITIONS_TO_BACKUP=*
20190505-100905: DBG     -- RESIZE_ROOTFS=1
20190505-100905: DBG     -- RESTORE_DEVICE=
20190505-100905: DBG     -- ROOT_PARTITION=
20190505-100905: DBG     -- RESTORE_DEVICE=
20190505-100905: DBG     -- ROOT_PARTITION=
20190505-100905: DBG     -- RSYNC_BACKUP_ADDITIONAL_OPTIONS=
20190505-100905: DBG     -- RSYNC_BACKUP_OPTIONS=-aHAx
20190505-100905: DBG     -- RSYNC_IGNORE_ERRORS=
20190505-100905: DBG     -- SENDER_EMAIL=
20190505-100905: DBG     -- SKIPLOCALCHECK=0
20190505-100905: DBG     -- STARTSERVICES=
20190505-100905: DBG     -- STOPSERVICES=
20190505-100905: DBG     -- SYSTEMSTATUS=0
20190505-100905: DBG     -- TAR_BACKUP_ADDITIONAL_OPTIONS=
20190505-100905: DBG     -- TAR_BACKUP_OPTIONS=-cpi --one-file-system
20190505-100905: DBG     -- TAR_BOOT_PARTITION_ENABLED=0
20190505-100905: DBG     -- TAR_IGNORE_ERRORS=
20190505-100905: DBG     -- TAR_RESTORE_ADDITIONAL_OPTIONS=
20190505-100905: DBG     -- TIMESTAMPS=0
20190505-100905: DBG     -- USE_HARDLINKS=1
20190505-100905: DBG     -- VERBOSE=1
20190505-100905: DBG     -- ZIP_BACKUP=0
20190505-100905: DBG     << logOptions
20190505-100905: DBG     >> logSystem
20190505-100905: DBG     -- PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
[... 84 lines cut ...]
20190505-100906: DBG           << exitError 102
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
20190505-100906: MSG        --- RBK0032I: Bitte warten bis aufgeräumt wurde.
20190505-100906: DBG           >> cleanupBackup
20190505-100906: DBG           -- Got trap EXIT
20190505-100906: DBG           -- rc: 102
Invocation parms: ' -F -v -l debug -L /mnt/backup/rsp-backup/mail/log/Backup_2019-05-05_10-09.log'
20190505-100906: DBG              >> sendEMail
20190505-100906: DBG              << sendEMail
20190505-100906: DBG              >> cleanupBackupDirectory
20190505-100906: DBG              -- BackupDir created: 0
--- RBK0026I: Logdatei wurde in /root/test.log gesichert.
20190505-100906: MSG              --- RBK0026I: Logdatei wurde in /root/test.log gesichert.
20190505-100906: DBG              << cleanupBackupDirectory
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
20190505-100906: MSG           ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
20190505-100906: DBG           << cleanupBackup
20190505-100906: DBG           >> cleanupTempFiles
20190505-100906: DBG           -- Removing mailfile /tmp/test.maillog
20190505-100906: DBG           << cleanupTempFiles
20190505-100906: DBG        -- Terminate now with rc 102
--- RBK0010I: mail: test.sh V0.6.4.2 (f5a5309) So 5. Mai 10:09:06 CEST 2019 beendet.
20190505-100906: MSG        --- RBK0010I: mail: test.sh V0.6.4.2 (f5a5309) So 5. Mai 10:09:06 CEST 2019 beendet.

`

framps, 2019-05-05:
Please open another issue for the new issue you discovered :smiley: Please also explain why you need this change. I don't consider it as a bug - but a feature :laughing:

closed 2019-05-05 by framps