framps/raspiBackup #138: No error message

biohazardxxx, 2019-05-06:
After I restored a broken SD on a new boot SD (thanks to the working backups by this script!)  I can't run the script to create backups anymore. Any idea what could be wrong? The only error I see is RBK0016E but no idea how to fix that.
```
pi@rpi2a:/usr/local/bin $ sudo /usr/local/bin/raspiBackup.sh
--- RBK0009I: rpi2a: raspiBackup.sh V0.6.4.3-beta (c042ea1) started at Mo 6. Mai 15:12:05 CEST 2019.
!!! RBK0165W: =========> NOTE  <=========
!!! RBK0165W: This is a betaversion and should not be used in production.
!!! RBK0165W: =========> NOTE <=========
--- RBK0128I: Using logfile /var/log/raspiBackup/rpi2a.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
??? RBK0016E: No sd card /dev/mmcblk found.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0105I: Deleting new backup directory /mnt/backup_router/RPI/rpi2a/rpi2a-rsync-backup-20190506-151205.
--- RBK0049I: Messages saved in /var/log/raspiBackup/rpi2a.msg.
--- RBK0026I: Debug logfile saved in /var/log/raspiBackup/rpi2a.log.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0167I: Sending email.
Reading message body from STDIN because the '-m' option was not used.
If you are manually typing in a message:
  - First line must be received within 60 seconds.
  - End manual input with a CTRL-D on its own line.

May 06 15:12:08 rpi2a sendEmail[20670]: Message input complete.
May 06 15:12:10 rpi2a sendEmail[20670]: Email was sent successfully!
--- RBK0010I: rpi2a: raspiBackup.sh V0.6.4.3-beta (c042ea1) stopped at Mo 6. Mai 15:12:11 CEST 2019.
```

framps, 2019-05-06:
Glad the restore was successfull :smiley: 

`??? RBK0016E: No sd card /dev/mmcblk found.` is the root cause for the failure. 

Please create a debug log with option `-l debug` and send it to me (see my eMail address on my [contact page](https://www.linux-tips-and-tricks.de/en/contact/))

biohazardxxx, 2019-05-07:
full log send to your mail here the the relevant snippet from the log:
```
20190507-122021 DBG 4290:        -> doitBackup 0
20190507-122021 DBG 4050:           -> getRootPartition 
20190507-122021 DBG 4055:           -- cat /proc/cmdline
20190507-122021 DBG 4055:           -- 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=tty1 root=UUID=ddaf2604-fb7c-4d13-86db-414ee4143ed9 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait rootdelay=5
20190507-122021 DBG 4058:           -- RootPartition: UUID=ddaf2604-fb7c-4d13-86db-414ee4143ed9
20190507-122021 DBG 4062:           <- getRootPartition UUID=ddaf2604-fb7c-4d13-86db-414ee4143ed9
20190507-122021 DBG 4087:           -> inspect4Backup 
20190507-122021 DBG 4089:           -- ls /dev/mmcblk*:
20190507-122021 DBG 4089:           -- /dev/mmcblk0
20190507-122021 DBG 4090:           -- ls /dev/sd*:
20190507-122021 DBG 4090:           -- /dev/sda
20190507-122021 DBG 4090:           -- /dev/sda1
20190507-122021 DBG 4091:           -- mountpoint /boot: 179:0 mountpoint /: 8:1
20190507-122021 DBG 4098:           -- Legacy boot discovery
20190507-122021 DBG 4101:           -- part: /dev/mmcblk0
20190507-122021 DBG 4104:           -- bootDeviceNumber: 179:0
20190507-122021 DBG 4105:           -- rootDeviceNumber: 8:1
20190507-122021 DBG 4116:           -- Starting alternate boot discovery
20190507-122021 DBG 4121:           -- /boot mounted? /dev/mmcblk0
20190507-122021 DBG 4132:           -- / mounted? /dev/sda1
20190507-122021 DBG 4074:              -> deviceInfo /dev/mmcblk0
20190507-122021 DBG 4082:              <- deviceInfo mmcblk 0
20190507-122021 DBG 4074:              -> deviceInfo /dev/sda1
20190507-122021 DBG 4082:              <- deviceInfo sda 1
20190507-122021 DBG 4153:           -- boot: mmcblk
20190507-122021 DBG 4154:           -- root: sda
20190507-122021 DBG 4171:           -- BOOT_DEVICE: mmcblk
20190507-122021 DBG 4175:           -- BOOT_DEVICENAME: /dev/mmcblk
20190507-122021 DBG 1583:              -> getPartitionPrefix mmcblk
20190507-122021 DBG 1592:              <- getPartitionPrefix mmcblkp
20190507-122021 DBG 4179:           -- BOOT_PARTITION_PREFIX: mmcblkp
20190507-122021 DBG 4182:           <- inspect4Backup 
20190507-122021 DBG 4025:           -> commonChecks 
20190507-122021 DBG 4044:           <- commonChecks 
20190507-122021 DBG 2063:           -> getFsType /mnt/backup_router/RPI
20190507-122021 DBG 2069:           <- getFsType cifs
??? RBK0016E: No sd card /dev/mmcblk found.
20190507-122021 MSG 4310:        ??? RBK0016E: No sd card /dev/mmcblk found. 
20190507-122021 DBG 1160:           -> exitError 107
20190507-122021 DBG 1167:           <- exitError 107
20190507-122021 DBG 2536:           -> cleanup 

```

framps, 2019-05-07:
Thank you very much for the log snippet and sending me the whole log which I actually needed to get the whole picture.

```
20190507-122021 DBG 3866:     -- Disk /dev/mmcblk0: 14,9 GiB, 15931539456 bytes, 31116288 sectors
20190507-122021 DBG 3866:     -- Units: sectors of 1 * 512 = 512 bytes
20190507-122021 DBG 3866:     -- Sector size (logical/physical): 512 bytes / 512 bytes
20190507-122021 DBG 3866:     -- I/O size (minimum/optimal): 512 bytes / 512 bytes
20190507-122021 DBG 3866:     -- Disklabel type: dos
20190507-122021 DBG 3866:     -- Disk identifier: 0x00000000
```
and
```
20190507-122021 DBG 3867:     -- /dev/mmcblk0 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```
and
```
20190507-122021 DBG 4089:           -- ls /dev/mmcblk*:
20190507-122021 DBG 4089:           -- /dev/mmcblk0
```
points out you don't have /dev/mmcblk0p1 which is the boot device or boot partition in Raspbian and is expected to exist. Do you have an idea how you managed to delete /dev/mmcblk0p1 and mount /dev/mmcblk0 as /boot? This shouldn't have happened when restoring the backup. Did you manually modify some config or SD partitioning?

raspiBackup will run again if /dev/mmcblk0p1 will exist on the SD card again. If the boot code is stored on /dev/mmcblk0 you system will come up but raspiBackup doesn't accept this partitioning because that's not standard Raspbian partitioning.

Or you have /boot on /dev/sda1 and /dev/mmcblk0 is not used at all to boot your system. Unfortunately /etc/fstab is not logged but I think then you have to edit /etc/fstab to use /dev/sda /boot instead of /dev/mmcblk0 and this should fix your issue.

biohazardxxx, 2019-05-09:
sorry for the late reply.
Yes I needed to adjust fstab file for the boot mount point:
PARTUUID=144a08c7-01  /boot           vfat    defaults          0       2
Because it was identified by PARTUUID of the broken SD
After the change it now looks like:
/dev/mmcblk0  /boot           vfat    defaults          0       2
Should I try to change to UUID="101A-1111" of the new SD in fstab?

framps, 2019-05-10:
I don't think so. Would you please run [this script](https://github.com/framps/raspiBackup/blob/master/inspectSystem.sh) and provide the output?

biohazardxxx, 2019-05-13:
Ok, done. Here the output (xx'ed credentials).

> 
> ========================= cat /etc/os*release
> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="8"
> VERSION="8 (jessie)"
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> 
> ========================= for f in /etc/*[-_]{release,version}; do echo - $f -; cat $f; echo; done
> - /etc/os-release -
> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="8"
> VERSION="8 (jessie)"
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> 
> - /etc/debian_version -
> 8.0
> 
> 
> ========================= cat /etc/fstab
> proc            /proc           proc    defaults          0       0
> /dev/mmcblk0  /boot           vfat    defaults          0       2
> # PARTUUID=109A-9113  /sd               ext4    defaults,noatime  0       1
> /dev/disk/by-uuid/xxxx-xxxx-4d13-86db-414ee4143ed9    /   ext4    defaults,noatime  0       1
> # a swapfile is not a swap partition, no line here
> #   use  dphys-swapfile swap[on|off]  for that
> # //192.168.0.1/Backups    /mnt/backup_router    cifs    defaults,noauto,nofail,username=xxxxxx,passwd=xxxxxx0    0
> //192.168.0.1/Backups    /mnt/backup_nas    cifs    defaults,user,file_mode=0777,vers=1.0,dir_mode=0777,auto,nofail,username=xxxxx,passwd=xxxxx,x-systemd.automount,x-systemd.require$
> # //media.local/Media     /media/xxxxxxxxxxxxx/Media     cifs     defaults,user,file_mode=0777,dir_mode=0777,auto,username=xxxxxxxx,password=xxxxxxx     0     0
> 
> 
> ========================= fdisk -l
> 
> Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 4096 bytes
> I/O size (minimum/optimal): 4096 bytes / 4096 bytes
> Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 4096 bytes
> I/O size (minimum/optimal): 4096 bytes / 4096 bytes
> Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 4096 bytes
> I/O size (minimum/optimal): 4096 bytes / 4096 bytes
> Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 4096 bytes
> I/O size (minimum/optimal): 4096 bytes / 4096 bytes
>

framps, 2019-06-03:
On Raspbian /etc/fstab has to have following format:

```
PARTUUID=d888a167-01  /boot           vfat    defaults          0       2
PARTUUID=d888a167-02  /               ext4    defaults,noatime  0       1
```
or in previous older versions
```
/dev/mmcblk0p1  /boot           vfat    defaults          0       2
/dev/mmcblk0p2  /               ext4    defaults,noatime  0       1
```
That's the format raspiBackup expects to see in /etc/fstab.

I have no clue how you managed to get 
```
/dev/mmcblk0 /boot vfat defaults 0 2
```
in your /etc/fstab. Please make sure your /etc/fstab conforms to the format used in Raspbian.

framps, 2019-06-10:
@biohazardxxx Are there any open questions left? Otherwise please close this issue.

biohazardxxx, 2019-06-12:
Sorry for late reply I was on holidays. 
I am not sure how can I get a PARTUUID for the /dev/mmcblk0 partition. Any hints?

framps, 2019-06-12:
Np. Try `sudo blkid`

biohazardxxx, 2019-06-13:
That unfortunately doesn't return a PARTUUID for that partition, only for my external HDD:

> pi@rpi:~ $ sudo blkid
> /dev/mmcblk0: LABEL="boot" UUID="109A-9113" TYPE="vfat"
> /dev/sda1: LABEL="rootfs" UUID="ddaf2604-fb7c-4d13-86db-414ee4143ed9" TYPE="ext4" PARTLABEL="primary" PARTUUID="7fab0989-0e3c-40df-b6c1-e38b38a01d9c"
>

framps, 2019-06-13:
PARTUUID is short for `partition uuid`. You don't have a partition on /dev/mmcblk0.

That's what you have to do:
1. save all contents of /dev/mmvblk0
2. create a partition /dev/mmvblk0p1 of 50M
3. copy the saved contents into /dev/mmcblk0p1  
4. Update /boot/cmdline.txt
5. Update /etc/fstab

Then raspiBackup will be able to backup your system.

biohazardxxx, 2019-06-17:
I will do that when I have some time to spare. Thanks for the help so far. I will close this ticket now and get back if I can't get it working.
Hope all this info helps someone else in the future :)

closed 2019-06-17 by biohazardxxx