framps/raspiBackup #141: Add 7/4/12 backup strategy

framps, 2019-05-27:
@DocYester suggested to add a 7/4/12 backup strategy in issue #128 and helped to create helper script [raspiBackup7412.sh](https://github.com/framps/raspiBackup/blob/beta/helper/raspiBackup7412.sh). This function should be integrated into next release.

framps, 2019-11-18:
Implemented 5a208a698e7151e82d29ca13244c46dac27343a2. Still some testing needed

framps, 2019-11-19:
Initial habe ich [diese Seite](https://www.linux-tips-and-tricks.de/de/raspibackupcategorie/555-raspibackup-smart-recycle-backup-strategie-7-4-12/) erstellt um die Strategie zu beschreiben. Sollte die Beschreibung unklar sein oder etwas fehlen - please let me know.

framps, 2019-11-24:
Implemented in https://github.com/framps/raspiBackup/commits/beta

christianTF, 2020-01-13:
Hi @framps,

two questions about the strategy:

Example config:
My user defines two types of backups:
- rsync hourly, called with strategy "48 2 1 0" (keep 48 daily , 2 weekly, 1 monthly backups)
- ddz weekly, called with strategy "0 1 2 0" (keep 0 daily, 1 weekly, 2 monthly backups)

1. 
Does the backup strategy observe the backup type?
If I do the hourly rsync call with `--smartRecycle --smartRecycleOptions "48 2 1 0", will it keep the ddz backups untouched, and vice versa?

2. 
The user has scheduled ddz to run weekly. What does the first "0" do with the currently processed backup? Is it sure that the currently finalized backup always stays untouched?

Thanks!
Christian

framps, 2020-01-13:
Hello @christianTF ,

> Does the backup strategy observe the backup type?

No. Not right now :disappointed: . But that's a valid point and I will check whether I can add this feature in the current (still inofficial) beta without blowing the whole beta. I keep you posted.

> What does the first "0" do with the currently processed backup? 

`0` means there is no daily backup considered to retain. `1` means there is one weekly backup retained. 

> Is it sure that the currently finalized backup always stays untouched?

Frankly I don't get your point. Please give an example.

christianTF, 2020-01-13:
> > What does the first "0" do with the currently processed backup?
> `0` means there is no daily backup considered to retain. `1` means there is one weekly backup retained.
> > Is it sure that the currently finalized backup always stays untouched?

An extreme example:
The call is `--smartRecycle --smartRecycleOptions "0 0 0 0"` to keep no retention backups.
Is it guaranteed, that the current backup, called with this options, is not deleted.

framps, 2020-01-13:
I got it now. If any parm is `0` the corresponding backup scan (daily, weekly, ...) is not executed thus this interval is ignored. But I notice I should add some additional smart recycle test for this scenario.

christianTF, 2020-01-13:
Thanks for clarifying!

framps, 2020-01-13:
Reopening until https://github.com/framps/raspiBackup/issues/196 is fixed.

framps, 2020-01-15:
#196 is now fixed

closed 2020-01-15 by framps