framps/raspiBackup #143: Not deleteing backup folder when cleaning up aborted/failed backup.

timinaust, 2019-06-08:
When a backup is cancelled the backup folder is not removed, only emptied.  The rm -r at line 2408 in the cleanup trap reports 'directory not empty'.  This is not recorded in the log that remains in the user's folder as that has already been copied.

Why: The stdin/stderr redirections 'exec 1>...' and 'exec 2>...' at line 2169 remain active and hence the raspiBackup.log file in the backup directory remains open.

Suggestion: Save the stdout and stderr just before redirecting (eg 'exec 1>&6' and 'exec 2>&7') and restoring them before the running the 'rm -r' (eg 'exec 6>&1' and 'exec 7>&2').

Running raspiBackup v0.6.4.2 on Raspbian 9/RPi3 to a remote nfs volume.

Ref: https://unix.stackexchange.com/questions/80988/how-to-stop-redirection-in-bash

framps, 2019-06-08:
Thank you very much for reporting the issue and suggesting a fix. 

Next version 0.6.4.3 is just ready to be published but will now be suspended until this issue is fixed.

framps, 2019-06-08:
@timinaust I created right now a branch with a fix for the issue you reported and based on your suggestion. Please download [this version](https://raw.githubusercontent.com/framps/raspiBackup/master_143_delete_dir_fails/raspiBackup.sh) and verify the fix.

Would be great if you will verify the fix short term - raspiBackup 0.6.4.3 is ready for release :smiley:

framps, 2019-06-09:
I spent some hours testing the fix. 

As long as you don't use a nfs mounted backup partition deletion of the directory works even using the old version. If a nfs mounted partition is used the directory is deleted only with the fix. In addition I detected the log file is incomplete now.

timinaust, 2019-06-10:
> @timinaust I created right now a branch with a fix for the issue you reported and based on your suggestion. Please download [this version](https://raw.githubusercontent.com/framps/raspiBackup/master_143_delete_dir_fails/raspiBackup.sh) and verify the fix.
> 
> Would be great if you will verify the fix short term - raspiBackup 0.6.4.3 is ready for release and should include the fix smiley

Thanks for the prompt response.  Yes this fixed version correctly deletes the directory in response to aborted (control-c) backup to nfs mounted volume, and the last three lines of the log file that is in the user home directory correctly contain ...  the copied location;  the fact that the backup failed;  the date/time the job stopped.

Great work, and very useful script.

framps, 2019-06-10:
Thank you very much you verified the fix :+1: . 

Frankly I have no clue why the old version works with local attached backup space (that's what I did in order to verify the directory is deleted correctly) but not with nfs. Looks like there is some special file handling in place when using nfs.

framps, 2019-06-10:
I merged the fix https://github.com/framps/raspiBackup/commit/e6bdb12a3c3779518309d8028f49e745439ac66d into master. The logging issue is not related to the change.

closed 2019-06-10 by framps