framps/raspiBackup #147: Anzeigefehler UI

Hofei90, 2019-06-25:
Die Empfohlenen Einstellungen von dem Hinweis der raspiBackup Webseite  waren schon eingestellt, jedoch zur Sicherheit nochmals überprüft. 
Eine Idee, an was der Anzeigefehler noch liegen kann?
[image]

framps, 2019-06-25:
Hm ... Da scheint es ein Problem mit der Spracherkennung zu geben. Wie ist denn bei Dir die Ausgabe als root von `env ¦ grep -i lang` ?

Hofei90, 2019-06-25:
Sicher mit dem komischen Zeichen? ¦
```
root@redmine_nextcloud_pi:/home/pi# env ¦ grep -i lang
env: '\302\246': No such file or directory
```

Bei `env | grep -i lang` erhalte ich keine Ausgabe

framps, 2019-06-25:
Das komische Zeichen habe ich auf dem Mobile eingegeben :disappointed: Es sollte wie Du schon richtig vermutet hast `|` sein.

Jetzt habe ich Dein Problem reproduzieren koennen. Es liegt irgendwie in der Kombination dass kein `LANG` gesetzt ist und Umlaute benutzt werden und kein englisches UTF8 installiert ist. Nach dem L kommt ein ö. 
1) Was bekommst Du denn wenn Du `env | grep -i LC_` eingibst.
2) Welche Sprachen hast Du mit raspi-config installiert?

framps, 2019-06-25:
I think I found a fix for your issue. Please test https://github.com/framps/raspiBackup/blob/development/installation/raspiBackupInstallUI.sh. But nevertheless please provide the information I asked for in previous comment.

Hofei90, 2019-06-26:
`env | grep -i LC_` ergibt keine Ausgabe

Bei folgenden Sprachen ist ein Stern:
de_DE.UTF-8 UTF-8
en_GB.UTF-8 UTF-8

Der Fehler ist im development Branch behoben.
"Lustigerweise" habe ich den selben Fehler gerade in raspi-config auch gehabt?! Egal was du gemacht hast, das könntest du dir Raspberry Pi Foundation weiter melden :)

framps, 2019-06-26:
Das habe ich auch festgestellt. Aber guter Tipp. In raspi-config wird da viel geheimnisvolles bzgl der Spracherkennung getan. Ich sehe mir das mal an.

Aber ist gut dass mit meinem Fix das Problem gelöst ist. aber was mich wundert:

Auf meinem Stretch bekomme ich
[code]
env | egrep -i "LANG|LC_"
LC_ALL=en_US.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_PAPER=de_DE.UTF-8
LC_MONETARY=de_DE.UTF-8
LANG=en_US.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_NUMERIC=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
[/code]

Was hast Du denn installiert? Was ist der Inhalt von /etc/os-release ?

Hofei90, 2019-06-27:
Auch mit der leichten Abwandlung `env | egrep -i "LANG|LC_"` gibt es bei mir keine Ausgabe.

Was meinst du mit: 

> Was hast Du denn installiert?
Welche Programme? Einsatzzweck ist, es läuft darauf Redmine mit apache2, mysql, passenger und redis.
Außerdem noch Nextcloud

```
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

uname -a
Linux redmine_nextcloud_pi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux

```

framps, 2019-06-27:
Nach einer normalen Stretch Installation sind dieses Environmentvariablen gesetzt und die Pi Foundation verlässt sich auf darauf. Irgendwie hast Du es geschafft diese Variablen rauszuschmeissen. Vielleicht hat irgendeine der installierten SW das umgeschossen?

Anyhow - ist ein guter Hinweis und jetzt ja soweit gefixed. Ich sehe noch mal nach wie das in raspi-config gemacht wird. Da ich das Problem jetzt ja jederzeit reproduzieren kann kann ich dann einen u.U. weiteren Fix in raspiBackup und/oder raspi-config ja selbst ausprobieren.

Schliesst Du den Issue dann bitte?

closed 2019-07-01 by Hofei90

framps, 2019-07-01:
Summary in English:
Due to missing locale settings LANG and LC_* the UI was not able to display text which used German Umlauts. In addition the window borders used by whiptail are garbage.