framps/raspiBackup #148: Einbau Fortschrittsanzeige

Hofei90, 2019-07-09:
Hallo,
ich würde eine Fortschrittsanzeige begrüßen, welche mir anzeigt beim Restore wie viele MB oder GB schon von der Gesamtgröße wiederhergestellt worden sind.

z.B 1,2GB von 8,3GB 
und sich diese dann stetig aktualisiert.
Gerne auch mit zusätzlichen Fortschrittsbalken.

Das ganze wär sowohl beim Restore als auch Backup sicherlich auch ein nettes Feature.

framps, 2019-07-23:
Beim Backup gibt es dazu bereits die Option `-g`. Allerdings wird beim tar nur -v benutzt, kein pv.
Beim Restore gibt es bei /boot schon Progress. Nur bei / fehlt es noch. Ich habe zugegebenermassen noch nicht daran gedacht. Am einfachsten wäre es die Logik vom Backup zu übernehmen und dabei beim tar bei / auch pv zu benutzen. 

Beim rsync ist das leider nicht ohne signifikant größere zusätzliche Laufzeit möglich denn für pv müsste erst die Menge der zu sichernden bzw zu rücksichernden Daten berechnet werden. Deshalb wird die rsync Option `--info=progress2` benutzt die auch eine gewissen Indikation liefert.

Ich erstelle dafür einen neuen Issue damit das in einer der nächsten Releases reinkommt.

framps, 2019-07-23:
Please close issue if you agree with #151

Hofei90, 2019-07-24:
Die Opiton mit `-g` hab ich übersehen.
Ich würde die Information mit `--info=progress2` in die Parameterliste https://www.linux-tips-and-tricks.de/de/raspibackup/#parameter mitaufnehmen.

closed 2019-07-24 by Hofei90