framps/raspiBackup #150: Restore fehlerhaft (rsync) Read-Only File System

Hofei90, 2019-07-09:
Before opening a big issue please check

* [ ] Did you check whether your message was listed on the messages page and the suggested action fixes your issue? [English](https://linux-tips-and-tricks.de.www289.your-server.de/en/rmessages) - [German](https://linux-tips-and-tricks.de.www289.your-server.de/de/fehlermeldungen)

If this doesn't help please provide following detailed information:

* [ ] Which operating system do you use? Provide the contents of /etc/os-release
* [ ] Provide all messages written by raspiBackup
* [ ] Attach the debug log to the issue which will be created by raspiBackup with option `-l debug`

Nachdem der Restore schon einige Zeit lief bekam ich folgende Fehlermeldung:
`raspiBackup.sh -d /dev/sdb /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/
--- RBK0009I: ***-VirtualBox: raspiBackup.sh V0.6.4.3 (2d927a2) Di 9. Jul 20:42:45 CEST 2019 gestartet.
--- RBK0128I: Logdatei ist /home/***/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0138I: Bootbackup /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 wird benutzt.
--- RBK0068I: Bootpartitionsdateien des Backups aus dem Verzeichnis /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 die mit redmine_nextcloud_pi-backup beginnen werden benutzt.
!!! RBK0018W: Ziel /dev/sdb mit 14.83 GiB ist größer als die Backupquelle mit 14.83 GiB. Die root Partition wird entsprechend vergrößert um den ganzen Platz zu benutzen.
!!! RBK0065W: Gerät /dev/sdb wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  50,2MB   46,0MB   primary  fat32        lba
 2      50,3MB  15932MB  15881MB  primary  ext4
!!! RBK0066W: Gerät /dev/sdb wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0038I: Bist Du sicher? j/N j
--- RBK0050I: Backup wird von /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 zurückgespielt.
--- RBK0051I: Master boot backup wird von /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/redmine_nextcloud_pi-backup.mbr auf /dev/sdb zurückgespielt.
--- RBK0052I: Partition(en) werden auf /dev/sdb erstellt.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sdb1 zurückgespielt.
--- RBK0054I: Zweite Partition (Rootpartition) /dev/sdb2 wird formatiert.
--- RBK0055I: Zweite Partition (Rootpartition) wird auf /dev/sdb2 zurückgespielt.
rsync: write failed on "/tmp/raspiBackup/var/lib/mysql/nextcloud/oc_activity.ibd": Read-only file system (30)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
??? RBK0035E: Backupprogramm rsync endete beim Restore mit RC 11.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft mit RC 117 beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: redmine_nextcloud_pi: raspiBackup.sh V0.6.4.3 (2d927a2) Di 9. Jul 21:40:32 CEST 2019 beendet.

Restore wurde von einem VMBox Linux Mint ausgeführt:
`NAME="Linux Mint"
VERSION="19.1 (Tessa)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.1"
VERSION_ID="19.1"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tessa
UBUNTU_CODENAME=bionic
`
Backup wurde mit rsync erstellt.

Option -l kann ich bei Bedarf morgen nachliefern?!

framps, 2019-07-23:
Das sieht mir sehr danach aus als wenn die SD Karte, die zum Restore benutzt wurde, nicht mehr in Ordnung ist, denn /boot konnte geschrieben werden und auch diverse Dateien von /. 
1) Ist dieser Fehler reproduzierbar und immer bei derselben Datei?
2) Tritt der Fehler bei einer neuen SD Karte auch auf?

Hofei90, 2019-07-24:
Fehler kommt immer innerhalb vom /temp Ordner. Jedoch bei unterschiedlichen Files.
Eine andere SD Karte habe ich leider nicht zur Hand, ein frisches raspbian buster ließ sich damit aber Problem mit balenaEtcher vom Windowsrechner aus aufspielen und funktionierte tadellos.

Die detaillierte Ausgabe durch die Funktion -l debug bekommt man die im Terminal auch zusehen oder nur im raspiBackup.log File, welches ich mir gerade überschrieben habe. Bei der Ausgabe im Terminal konnte ich keine nennenswerte unterschiede zur normalen Ausgabe feststellen.

framps, 2019-07-24:
Es ist merkwürdig dass die Partition plötzlich read-only wird. Ich bin in der Vergangenheit schon öfter einem Fehler hinterhergelaufen wo sich die Ursache in einer defekten SD Karte fand :cry: . Das könnte auch hier der Fall sein. D.h. bevor ich hier detailierter reinsehe teste bitte ob der Restore auch bei einer neuen SD Karte auftritt.

Hofei90, 2019-07-27:
Ok verständlich - weitere Tests/Versuche kann ich aber erst ab 12.08.19 durchführen. Den Issue hab ich also nicht vergessen ;)

framps, 2019-07-28:
Nur keine Eile. Und der Issue bleibt so lange offen bis Du ihn closest :smirk:

framps, 2019-08-27:
@Hofei90 Wie sieht es aus? Hast Du inzwischen weitere Tests machen koennen?

Hofei90, 2019-08-27:
Ist für nächste Woche geplant - nicht vergessen ;)

framps, 2019-08-27:
:+1: Perfekt

Hofei90, 2019-09-02:
Neue SD Karte - selber Fehler:

```
raspiBackup.sh -d /dev/sdb /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/
--- RBK0009I: ***-VirtualBox: raspiBackup.sh V0.6.4.3 (2d927a2) Mo 2. Sep 16:34:21 CEST 2019 gestartet.
--- RBK0128I: Logdatei ist /home/****/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0138I: Bootbackup /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 wird benutzt.
--- RBK0068I: Bootpartitionsdateien des Backups aus dem Verzeichnis /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 die mit redmine_nextcloud_pi-backup beginnen werden benutzt.
!!! RBK0018W: Ziel /dev/sdb mit 14.83 GiB ist größer als die Backupquelle mit 14.83 GiB. Die root Partition wird entsprechend vergrößert um den ganzen Platz zu benutzen.
!!! RBK0065W: Gerät /dev/sdb wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  15932MB  15927MB  primary  fat32        lba
!!! RBK0066W: Gerät /dev/sdb wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0038I: Bist Du sicher? j/N j
--- RBK0050I: Backup wird von /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 zurückgespielt.
--- RBK0051I: Master boot backup wird von /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/redmine_nextcloud_pi-backup.mbr auf /dev/sdb zurückgespielt.
--- RBK0052I: Partition(en) werden auf /dev/sdb erstellt.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sdb1 zurückgespielt.
--- RBK0054I: Zweite Partition (Rootpartition) /dev/sdb2 wird formatiert.
--- RBK0055I: Zweite Partition (Rootpartition) wird auf /dev/sdb2 zurückgespielt.
rsync: write failed on "/tmp/raspiBackup/opt/redmine/files/dmsf/p_stoerungen/170213114643_224_Schaltplan.pdf": Read-only file system (30)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
??? RBK0035E: Backupprogramm rsync endete beim Restore mit RC 11.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft mit RC 117 beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: redmine_nextcloud_pi: raspiBackup.sh V0.6.4.3 (2d927a2) Mo 2. Sep 16:39:10 CEST 2019 beendet.
```

Wie im Ausgangspost schon erwähnt, möchte ich es nochmal erwähnen, dass ich das ganze mit einer Virtuellen Maschine mache, kann daran der Fehler liegen?
Lieder hab ich keinen Linuxrechner zur Hand.

framps, 2019-09-02:
Ok. Vielen Dank fuer den Test mit den neuen SD Karte. Somit sind wir zu 99.9% sicher dass es nicht an der SD Karte liegt. Dass die nun auch einen Weg hat ist ja eher unwahrscheinlich :smiley: 

Ich denke /dev/sdb2 ist wohl wirklich RO gemounted. Die Frage ist warum ... raspiBackup erzeugt ein Unterverzeichnis /tmp/raspiBackup/boot, mounted da die neue Bootpartition und restored dorthin das Bootverzeichnis aus dem Backup. Danach wird /tmp/raspiBackup/boot umounted
`umount $BOOT_PARTITION &>>"$LOG_FILE"`
und nun in /tmp/raspiBackup das neue Rootfilesystem gemounted und das Rootfilesystem aus dem Backup restored.

1) Kannst Du denn /dev/sdb2 manuell unter irgendeinem Verzeichnis in /tmp mounten und was draufschreiben? Wie ist dann die Ausgabe von `sudo mount`? Merkwuerdig ist dass es auf /dev/sdb1 funktioniert. 

2) Es koennte auch noch sein dass es ein Problem beim umount der Bootpartition gibt. Dann sollte sich im Logfile fuer `umount $BOOT_PARTITION &>>"$LOG_FILE"` eine Fehlermeldung finden. Kannst Du mal dort nachsehen ob nach dem Text `Umounting partitions` irgendeine Fehlermeldung zu finden ist? Vielleicht dauert ja der umount in der VM etwas länger dass der folgende mount von root failed. Direkt nach der Meldung `RBK0055I` sollte dann im Log eine Fehlermeldung auftauchen.

Hofei90, 2019-09-02:
Sind die Befehle so 1:1 ausführbar, oder soll ich die $xxx mit irgendas ersetzen?
Das geht mir gerade "zu schnell"

framps, 2019-09-02:
Kein Problem. Teilweise habe ich auch nur _laut gedacht_ bzw _laut geschrieben_ :smile: 

Zu (1)
Bitte folgende Befehl ausführen und die Ergebnisse der Befehle zeigen nachdem die SD Karte, auf die Du versuchtest zu restoren in den USB Reader gesteckt hast und als /dev/sdb erkannt wurde:

```
sudo mkdir -p /tmp/raspiBackup/boot
sudo mount /dev/sdb1 /tmp/raspiBackup/boot
sudo echo "hello boot" > /tmp/raspiBackup/boot/hello
sudo cat /tmp/raspiBackup/boot/hello
sudo mount
sudo umount /tmp/raspiBackup/boot

sudo mount /dev/sdb2 /tmp/raspiBackup
sudo echo "hello world" > /tmp/raspiBackup/hello
sudo cat /tmp/raspiBackup/hello
sudo mount
```

Zu (2)
In der Debuglogdatei nach `Umounting partitions` suchen und die folgenden 10 Zeilen anzeigen. Dann noch nach `RBK0055I` suchen und die folgenden 10 Zeilen anzeigen.

Hofei90, 2019-09-02:
Ergebnis zu 1.:
```
***@***-VirtualBox:~$ sudo su
root@***-VirtualBox:/home/***# mkdir -p /tmp/raspiBackup/boot
root@***-VirtualBox:/home/***# mount /dev/sdb1 /tmp/raspiBackup/boot
root@***-VirtualBox:/home/***# echo "hello boot" > /tmp/raspiBackup/boot/hello
root@***-VirtualBox:/home/***# cat /tmp/raspiBackup/boot/hello 
hello boot
root@***-VirtualBox:/home/***# mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=987812k,nr_inodes=246953,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=204124k,mode=755)
/dev/sda1 on / type ext4 (rw,relatime,errors=remount-ro,data=ordered)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,nosuid,nodev,noexec,relatime,rdma)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/hugetlb type cgroup (rw,nosuid,nodev,noexec,relatime,hugetlb)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=40,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=12981)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
***:/Redmine_Backup on /media/redmine_backup_nfs type nfs4 (rw,relatime,vers=4.2,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=10.0.2.15,local_lock=none,addr=***)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=204120k,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/sr0 on /media/***/VBox_GAs_6.0.10 type iso9660 (ro,nosuid,nodev,relatime,nojoliet,check=s,map=n,blocksize=2048,uid=1000,gid=1000,dmode=500,fmode=400,uhelper=udisks2)
/dev/fuse on /run/user/1000/doc type fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=204120k,mode=700)
/dev/sdb1 on /tmp/raspiBackup/boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
root@***-VirtualBox:/home/***# umount /tmp/raspiBackup/boot/
root@***-VirtualBox:/home/***# mount /dev/sdb2 /tmp/raspiBackup
root@***-VirtualBox:/home/***# echo "hello world" > /tmp/raspiBackup/hello
root@***-VirtualBox:/home/***# cat /tmp/raspiBackup/hello 
hello world
root@***-VirtualBox:/home/***# mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=987812k,nr_inodes=246953,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=204124k,mode=755)
/dev/sda1 on / type ext4 (rw,relatime,errors=remount-ro,data=ordered)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
[... 30 lines cut ...]
20190902-163427 DBG 3389:              -- label-id: 0xa1a856ea
[... cut ...]
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
20190902-163910 DBG 1208:                 <- executeCmd 11
??? RBK0035E: Backupprogramm rsync endete beim Restore mit RC 11.
20190902-163910 MSG 3554:              ??? RBK0035E: Backupprogramm rsync endete beim Restore mit RC 11.
20190902-163910 DBG 1163:                 -> exitError 117
20190902-163910 DBG 1170:                 <- exitError 117
20190902-163910 DBG 2541:                 -> cleanup
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
```

framps, 2019-09-02:
> 20190902-163508 DBG 1188:                 -> executeCmd Command: rsync --numeric-ids -aHX --exclude=/redmine_nextcloud_pi-backup.* "/media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/$

Diese Zeile sieht mir unvollstaendig bzw falsch aus. Ich haette folgendes erwartet:
```
executeCmd Command: rsync --numeric-ids -aHX --exclude=/redmine_nextcloud_pi-backup.* "/media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 /tmp/raspiBackup
```
Kannst Du die Zeile noch mal genau prüfen ?

Hofei90, 2019-09-02:
Du hast natürlich recht - das kommt davon, wenn man die Datei mit nano öffnet

```
cat raspiBackup.log | grep "DBG 1188"
20190902-163508 DBG 1188:                 -> executeCmd Command: rsync --numeric-ids -aHX --exclude=/redmine_nextcloud_pi-backup.* "/media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/" /tmp/raspiBackup
```

framps, 2019-09-02:
Danke. Ich muss jetzt eine raspiBackup Version zusammenbauen die weitere Debugstatements hat damit wir weiterkommen. Das dauert dann aber etwas :wink:

framps, 2019-10-05:
Die aktuelle Version im Development Branch enthaelt Code der, wenn meine Vermutung stimmt, das Problem beheben sollte. Bitte testen und Feedback :wink:

Hofei90, 2019-11-04:
So welcher Commit bietet sich denn jetzt an für den Test? Immer noch der aktuelle Branch im Development?

framps, 2019-11-04:
Ja nimm den. Der aktuelle Development branch ist die Basis fuer das naechste kommende Release Beta und enthaelt an bestimmten Stellen Code der darauf wartet bis Partitionsupdates im Kernel erkannt wurden.

Hofei90, 2019-11-05:
```
***@***-VirtualBox:~/raspiBackup$ git branch
* development
  master
***@***-VirtualBox:~/raspiBackup$ sudo raspiBackup.sh -d /dev/sdb /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/
sudo: raspiBackup.sh: Befehl nicht gefunden
***@***-VirtualBox:~/raspiBackup$ sudo bash raspiBackup.sh -d /dev/sdb /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/
--- RBK0009I: ***-VirtualBox: raspiBackup.sh V0.6.4.4-dev (d1ffc9f) Di 5. Nov 14:24:56 CET 2019 gestartet.
!!! RBK0192W: =========> HINWEIS <========= 
!!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte. 
!!! RBK0192W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /home/***/raspiBackup/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
??? RBK0154E: Ein Restore ist nicht möglich wenn eine Partition von /dev/sdb gemounted ist.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft mit RC 102 beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: ***-VirtualBox: raspiBackup.sh V0.6.4.4-dev (d1ffc9f) Di 5. Nov 14:24:57 CET 2019 beendet mit Returncode 102.
***@***-VirtualBox:~/raspiBackup$ cat raspiBackup.sh -d /dev/sdb /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/
cat: Ungültige Option -- d
„cat --help“ liefert weitere Informationen.
***@***-VirtualBox:~/raspiBackup$ 

```

Muss ich die aus git geclonte Version auch installieren?
Sollte nicht das Skript direkt auch funktionieren?
Sry ich steh gerade etwas auf den Schlauch warum sich der Development Branch nicht ausführen lässt.

framps, 2019-11-05:
> Muss ich die aus git geclonte Version auch installieren?

Nein
> Sollte nicht das Skript direkt auch funktionieren?

Ja
> Sry ich steh gerade etwas auf den Schlauch warum sich der Development Branch nicht ausführen lässt.

NP. Du musst in dem Verzeichnis, in dem Du den Branch ausgechecked hast entweder `sudo bash raspiBackup.sh` benutzen oder `sudo ./raspiBackup.sh`. Bitte pruefe noch vorher ob Du die aktuelle Version benutzt mit 
```
sudo ./raspiBackup.sh --version
Version: 0.6.4.4-dev CommitSHA: d1ffc9f CommitDate: 2019-10-28 CommitTime: 19:48:40
```
Falls nicht musst Du noch `git pull` eingeben und Dir den letzten Stand vom git repo holen.

Hofei90, 2019-11-07:
Auch auf dem Development Stand kommt es noch zu dem beschriebenen Fehler:

```
<nachname>@<nachname>-VirtualBox:~/raspiBackup$ sudo ./raspiBackup.sh --version
Version: 0.6.4.4-dev CommitSHA: d1ffc9f CommitDate: 2019-10-28 CommitTime: 19:48:40
<nachname>@<nachname>-VirtualBox:~/raspiBackup$ sudo ./raspiBackup.sh -d /dev/sdb /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508/
--- RBK0009I: <nachname>-VirtualBox: raspiBackup.sh V0.6.4.4-dev (d1ffc9f) Do 7. Nov 07:39:03 CET 2019 gestartet.
!!! RBK0192W: =========> HINWEIS <========= 
!!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte. 
!!! RBK0192W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /home/<nachname>/raspiBackup/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0138I: Bootbackup /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 wird benutzt.
--- RBK0068I: Bootpartitionsdateien des Backups aus dem Verzeichnis /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 die mit redmine_nextcloud_pi-backup beginnen werden benutzt.
!!! RBK0018W: Ziel /dev/sdb mit 14.83 GiB ist größer als die Backupquelle mit 14.83 GiB. Die root Partition wird entsprechend vergrößert um den ganzen Platz zu benutzen.
!!! RBK0065W: Gerät /dev/sdb wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  49,4MB   45,2MB   primary  fat32        lba
 2      50,3MB  15932MB  15881MB  primary  ext4
!!! RBK0066W: Gerät /dev/sdb wird überschrieben mit der gesicherten Boot- und Rootpartition.
!!! RBK0069W: Bootpartition /dev/sdb1 wird formatiert und erhält die zurückgespielte Bootpartition.
!!! RBK0070W: Rootpartition /dev/sdb2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N j
--- RBK0050I: Backup wird von /media/redmine_backup_nfs/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-rsync-backup-20190707-001508 zurückgespielt.
--- RBK0052I: Partition(en) werden auf /dev/sdb erstellt.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sdb1 zurückgespielt.
--- RBK0054I: Zweite Partition (Rootpartition) /dev/sdb2 wird formatiert.
--- RBK0055I: Zweite Partition (Rootpartition) wird auf /dev/sdb2 zurückgespielt.
rsync: write failed on "/tmp/raspiBackup/usr/bin/mysql_plugin": Read-only file system (30)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
??? RBK0035E: Backupprogramm rsync endete beim Restore mit RC 11.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft mit RC 117 beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: redmine_nextcloud_pi: raspiBackup.sh V0.6.4.4-dev (d1ffc9f) Do 7. Nov 07:47:44 CET 2019 beendet mit Returncode 117.

```

framps, 2019-11-07:
Mist :cry: . Kannst Du mir das Debuglog per eMail zuschicken?

Hofei90, 2019-11-15:
Fehler **nicht** auffindbar
Keine Lösung in Sicht.

closed 2019-11-15 by Hofei90

framps, 2019-11-15:
Just an additional comment: we detected following error message:
```
[  721.983128] usb 1-2: reset full-speed USB device number 3 using ohci-pci
[  723.368551] sd 3:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[  723.368564] sd 3:0:0:0: [sdb] tag#0 Sense Key : Unit Attention [current] 
[  723.368572] sd 3:0:0:0: [sdb] tag#0 Add. Sense: Not ready to ready change, medium may have changed
[  723.368580] sd 3:0:0:0: [sdb] tag#0 CDB: Write(10) 2a 00 00 c5 c4 a8 00 00 f0 00
[  723.368587] print_req_error: I/O error, dev sdb, sector 12960936
[  723.368605] Buffer I/O error on dev sdb2, logical block 1607829, lost async page write
```
So it looks like it's somehow related to the communication between SD card and Virtual Box communication to the SD card. Maybe a race condition inside of VB when writing on the SD card.