framps/raspiBackup #151: Add a progress indication for restore

framps, 2019-07-23:
Right now option `-g` creates a progress indication for backup of /boot and / and /boot for restore.

Use pv for a tar backup and restore progress indication and for dd in restore. Add also `--info=progress2` when restoring a rsync backup.

closed 2020-01-11 by framps

skyerjoe, 2020-08-04:
Is this feature also included in the UI MENUE ?

best regards

framps, 2020-08-04:
@skyerjoe Not sure I get your point. There is no way to kick off a restore from the UI and thus there is no progress indicator.

skyerjoe, 2020-08-05:
I mean if I configure an rsync backup over the gui, do I have to add the command  "--info=progress2" manually on the command line or is this also possible in the config or the gui.

framps, 2020-08-05:
Got it now :smile: . 

Just use option `-g` when you invoke raspiBackup and then `--info=progress2` will be used when rsync is called.