framps/raspiBackup #156: RaspiBackup.vars: Restore File and directory

christianTF, 2019-08-03:
Restore directory and Restore file to
/tmp/raspibackup.vars

as talked about in
https://www.loxforum.com/forum/projektforen/loxberry/plugins/122935-plugin-loxberry-backup?p=208329#post208329

framps, 2019-08-04:
@christianTF 
I just wanted to add the information you need in raspiBackup but then I detected I don't know exactly which usage scenario you are talking about and which information you need :cry: Would you please spend some more details on your requirements ?

framps, 2019-08-31:
@christianTF Do you still need this feature? Please give more details. Oherwiese please close the issue :wink:

christianTF, 2019-08-31:
I currently have the request on the radar, therefore I close the issue 🙂

closed 2019-08-31 by christianTF