framps/raspiBackup #157: Add support option for root partition (to support backup of encrypted systems)

JsBergbau, 2019-08-07:
With dd backup method raspi backup seems hardcoded to use /dev/mmcblk0p1 and /dev/mmcblk0p2
When system is encrypted backuping /dev/mmcblk0p2 produces unusable output. In fact it contains the luks header and the encrypted data, so you could decrypt it with password. However since this data looks like random data it is not compressible at all and thus wasting space. 
So to have an unencrypted backup which is compressible please add an option to select root partition to save with dd.

framps, 2019-08-08:
raspiBackup saves the root partition which is /dev/mmcblk0p2 or if / is moved to an external storage like USB stick this partition is saved.

A backup has to be an exact copy of the system and thus the encrypted data is saved as is. If the backup contains the unencrypted data that's a security breach :open_mouth: .
I see you point compression with ddz will not save much disk space given the entropy of the data. But that's what you have to pay for when you use an encrypted partition :wink:

framps, 2019-08-31:
@JsBergbau 

> So to have an unencrypted backup which is compressible please add an option to select root partition to save with dd.

Maybe I don't get your point. Would you please elaborate on this a bit more? Or do you agree with my previous comment? Then please close this issue :wink:

JsBergbau, 2019-09-16:
> A backup has to be an exact copy of the system and thus the encrypted data is saved as is. If the backup contains the unencrypted data that's a security breach
I think it depends. If you save the backup to an encrypted NAS then it is no security breach. But I also get your point. So an option to chose would be nice, but never the default. 
However I'm closing this issue because only very little users will ever need this feature.

closed 2019-09-16 by JsBergbau