framps/raspiBackup #158: Raspibackup disregards -f option

JsBergbau, 2019-08-07:
I use raspibackup without installation.
Having this config file

```
####################################################
#
# Default Konfigurationsdatei für raspiBackup.sh
#
# (C) 2013,2019 framp at linux-tips-and-tricks dot de
#
# Weitere Details zu den Parametern finden sich auf
# http://www.linux-tips-and-tricks.de/de/raspibackup/#parameter
#
# $Date: 2019-06-07 17:54:47 +0200$
# $Sha1: 6fc3284$
#
####################################################

# folgende Zeile nicht aendern
VERSION_DECONF="0.1.2"

#####################################################
#
# Hinweis:
#
# 0 schaltet eine Option aus
# 1 schaltet eine Option an
#
#####################################################

# Pfad wo das Backupfile gespeichert wird
DEFAULT_BACKUPPATH="/backup"

# Anzahl der zu vorhaltenden Backups
DEFAULT_KEEPBACKUPS=3
# Anzahl der vorzuhaltenden Backups fuer einen speziellen Backuptyp. Falls sie 0 ist wird DEFAULT_KEEPBACKUPS benutzt
DEFAULT_KEEPBACKUPS_DD=0
DEFAULT_KEEPBACKUPS_DDZ=0
DEFAULT_KEEPBACKUPS_TAR=0
DEFAULT_KEEPBACKUPS_TGZ=0
DEFAULT_KEEPBACKUPS_RSYNC=0

# Typ des Backups: dd, tar, xbmc or rsync
DEFAULT_BACKUPTYPE="rsync"

# zip tar oder dd backup
DEFAULT_ZIP_BACKUP=0

# dd backup sichert nur den von den Partitionen belegten Platz
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0

# Durch && getrennte Befehle, die vor dem Starten des Backups auszuführen sind
DEFAULT_STOPSERVICES=":"

# Durch && getrennte Befehle, die nach dem Starten des Backups auszuführen sind
DEFAULT_STARTSERVICES=":"

# Durch && getrennte Befehle, die ausgeführt werden bevor Services gestoppt werden
DEFAULT_BEFORE_STOPSERVICES=":"

# Durch && getrennte Befehle, die ausgeführt werden nachdem die Services wieder gestartet werden
DEFAULT_AFTER_STARTSERVICES=":"

# emailadresse die das Backupergebnis erhält
DEFAULT_EMAIL=""

# Sender emailadresse die mit ssmtp benutzt wird
DEFAULT_SENDER_EMAIL=""

# Weitere Parameter für das eMail programm (Optional)
DEFAULT_EMAIL_PARMS=""

# Message level (0 = minimal, 1 = detailed)
DEFAULT_MSG_LEVEL=0

# mailprogram
DEFAULT_MAIL_PROGRAM="mail"

# Gerät wo das Backup restored wird
DEFAULT_RESTORE_DEVICE=""

# Log wird in eMail mitgeschickt
DEFAULT_APPEND_LOG=0

# Option die vom Mailprogramm benutzt wird um das Log anzuhängen (Z.B. -a oder -A)
DEFAULT_APPEND_LOG_OPTION="-a"

# Detailierte Logausgaben der Backupprogramme
DEFAULT_VERBOSE=0

# Check auf einen remoten Backupfad wird nicht vorgenommen
DEFAULT_SKIPLOCALCHECK=0

# Blocksize von dd
DEFAULT_DD_BLOCKSIZE=1M

# Weitere Parameter für dd
DEFAULT_DD_PARMS=""

# Excludeliste für das benutzte Backuprogramm
DEFAULT_EXCLUDE_LIST=""

# Notifizierung soll stattfinden wenn eine neue Scriptversion verfügbar ist
DEFAULT_NOTIFY_UPDATE=0

# Aufzurufende Erweiterungen
DEFAULT_EXTENSIONS=""

# partitionsbasierter backup modus
DEFAULT_PARTITIONBASED_BACKUP=0

# Partitionsnummern, die gesichert werden sollen beim partitionsbasierten Backup
# Um alle partitionen zu sicher benutze "*"
DEFAULT_PARTITIONS_TO_BACKUP="*"
# Partitionsnummern muessen in Klammern stehen. Siehe Beispiel in folgender Zeile fuer die Partitionen 1,2 und 5
# DEFAULT_PARTITIONS_TO_BACKUP=(1 2 5)

# Sprache der Meldungen (DE oder EN)
DEFAULT_LANGUAGE="DE"

# Systeme auf die raspiBackup mit dem Parameter -y übertragen wird. ssh Zugang muss in der authorized keys Datei definiert sein
# Beispiel: "root@raspberrypi root@fhem root@openhab root@magicmirror"
DEFAULT_DEPLOYMENT_HOSTS=""

# Es wird nicht nach einer Restorebestätigung gefragt bei dem Device. Vorsicht!
DEFAULT_YES_NO_RESTORE_DEVICE=""

# Es werden Hardlinks für Partitionsbootfiles benutzt
DEFAULT_LINK_BOOTPARTITIONFILES=1

# Hardlinks werden für rsync benutzt
DEFAULT_USE_HARDLINKS=1

# Sichere boot partition mit tar
DEFAULT_TAR_BOOT_PARTITION_ENABLED=0

# eMail wird nur gesendet wenn ein Fehler aufgetreten ist. Vorsichtig benutzen !
DEFAULT_MAIL_ON_ERROR_ONLY=0

# Versionen bei denen keine deprecated message gemeldet werden soll, getrennt durch Kommata
DEFAULT_SKIP_DEPRECATED=""

# Reporte uuid
DEFAULT_USE_UUID=1

# teste auf bad blocks wenn beim restore eine Partition formatiert wird (Das dauert lange)
DEFAULT_CHECK_FOR_BAD_BLOCKS=0

# Das root Filesystem wird beim restore automatisch angepasst
DEFAULT_RESIZE_ROOTFS=1

# Zeitstempel werden vor den Meldungen ausgegeben
DEFAULT_TIMESTAMPS=0

# System status Informationen werden im Log aufgeführt
DEFAULT_SYSTEMSTATUS=0

# Interval fuer eine Restoretest Erinnerung (Einheit: Monate)
DEFAULT_RESTORE_REMINDER_INTERVAL=6

# Anzahl der Erinnerungen an einen Restoretest
DEFAULT_RESTORE_REMINDER_REPEAT=3

```
Interesting is the line 

# Typ des Backups: dd, tar, xbmc or rsync
DEFAULT_BACKUPTYPE="rsync"

When raspiBackup is startet via ./raspiBackup.sh -f raspiBackup_de.conf
it starts backing up with dd

On another system, default backup is dd and in the config file is
`# zip tar oder dd backup
DEFAULT_ZIP_BACKUP=1
`
However raspiBackup doesn't compress. You have to add the -z switch, then it compresses.

framps, 2019-08-09:
@JsBergbau I created a fix for this issue in the development branch. Please verify the fix :wink:

framps, 2019-08-27:
@JsBergbau Would you please verify my fix for your issue?

JsBergbau, 2019-08-28:
@framps 
Thank you. "-f" Option works now as expected.

closed 2019-08-28 by JsBergbau