framps/raspiBackup #159: chown failed: Invalid argument (22)

ChrizZz90, 2019-08-23:
I changed my backup type to rscync and have issues with chown and chgrp, which is both not working due to invalid argument (22). I try to make a backup at my Synology NAS, there I have activated rsync but right now without a rsync account.

Content /etc/os-release:
`PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

Part of the Issue msg:
`rsync: chown "/backup/raspberrypi/raspberrypi-rsync-backup-20190823-075622/mmcblk0p7/home/pi/.config/Electron/Cache" failed: Invalid argument (22)
rsync: chgrp "/backup/raspberrypi/raspberrypi-rsync-backup-20190823-075622/mmcblk0p7/etc/ssl/private/.ssl-cert-snakeoil.key.Wg4wnA" failed: Invalid argument (22)
rsync: chown "/backup/raspberrypi/raspberrypi-rsync-backup-20190823-075622/mmcblk0p7/etc/xrdp/.rsakeys.ini.IDYbEW" failed: Invalid argument (22)
rsync: chown "/backup/raspberrypi/raspberrypi-rsync-backup-20190823-075622/mmcblk0p7/home/pi/.Xauthority.wpsOCT" failed: Invalid argument (22)
rsync: chown "/backup/raspberrypi/raspberrypi-rsync-backup-20190823-075622/mmcblk0p7/home/pi/.bash_history.dwjnYi" failed: Invalid argument (22)
rsync: chown "/backup/raspberrypi/raspberrypi-rsync-backup-20190823-075622/mmcblk0p7/home/pi/.bash_logout.g2Z0qK" failed: Invalid argument (22)
`
After that I have to stop the backup with STRG+C

framps, 2019-08-23:
Your backup partition is not mounted correctly. I found https://serverfault.com/questions/213647/rsync-fails-to-backup-some-files-giving-error-22 which explains some root causes for this error message.
See also https://www.linux-tips-and-tricks.de/en/synology-usage/ which has some gotchas of Synology users with raspiBackup.
How these links will help you to locate the root cause. Would be great if you update this issue when you identified the root cause and future users of Synology will be able to benefit from your investigations.

ChrizZz90, 2019-08-23:
Solution is to use nfs 3 instead of 4 with:

`sudo mount -t nfs -o nfsvers=3 NAS.IP:/volume/Backup_folder /backup_folder`

now I will try if I can use nfs 4 if I give user rights on NAS for guests.

closed 2019-08-23 by ChrizZz90

framps, 2019-08-24:
:+1: Great you found the root cause. Thank you very much for sharing the solution.