framps/raspiBackup #161: Zeile 943: printf: `)': Ungültiges Formatierungszeichen.

HB9ERD, 2019-08-25:
Ich hatte mein Initial-Backup im Januar mit einer älteren Version (damals die aktuellste, eine Beta) erstellt und wollte jetzt endlich mal das nächste Backup machen. Dazu machte ich zuerst ein Update von RaspiBackup (mit -U). Dann gab es beim Backup die o.g. Fehlermeldung.

Hier alle Meldungen:

pi@raspi3b ~ > sudo /usr/local/bin/raspiBackup.sh -o "systemctl stop cups && systemctl stop cron" -a "systemctl start cron && system,ctl start cups" /backup/
--- RBK0009I: raspi3b: raspiBackup.sh V0.6.4.3 (2d927a2) Sam Aug 24 22:45:52 CEST 2019 gestartet.
--- RBK0151I: Backuppfad /backup wird benutzt.
--- RBK0036I: Partitionslayout wird gesichert.
43+1 Datensätze ein
43+1 Datensätze aus
45190144 Bytes (45 MB, 43 MiB) kopiert, 1.93074 s, 23.4 MB/s
--- RBK0158I: rsync Backup "/backup/raspi3b/raspi3b-rsync-backup-20190824-224551" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 01:39:49.
--- RBK1001I: Speicherauslastung - Vor dem Backup - Belegt:  MB Frei:  MB - Nach dem Backup: Belegt:  MB Frei:  MB
--- RBK1000I: CPU Temperatur vor und nach dem Backup: 51.5'C - 46.2'C
--- RBK1002I: Partitionsauslastung vor dem Backup: Belegt: 98.36 GiB Frei: 10.63 GiB
--- RBK1003I: Partitionsauslastung nach dem Backup: Belegt: 26.81 GiB Frei: 82.19 GiB
/usr/local/bin/raspiBackup.sh: Zeile 943: printf: `)': Ungültiges Formatierungszeichen.
--- RBK1004I: Änderung freier Platz: 71.55 GiB (672.00 
/usr/local/bin/raspiBackup.sh: Zeile 1216: system,ctl: Kommando nicht gefunden.
!!! RBK0047W: Ein Fehler trat beim Starten von Services auf. RC 127.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup /backup/raspi3b/raspi3b-rsync-backup-20190824-224551 in wird gelöscht. Das wird etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
/usr/local/bin/raspiBackup.sh: Zeile 1216: system,ctl: Kommando nicht gefunden.
!!! RBK0047W: Ein Fehler trat beim Starten von Services auf. RC 127.
--- RBK0010I: raspi3b: raspiBackup.sh V0.6.4.3 (2d927a2) Son Aug 25 00:28:48 CEST 2019 beendet.
pi@raspi3b ~ [106]> 

Liegt der Fehler (wie so häufig) irgendwo an der Bedienung oder ist das tatsächlich ein Bug?

framps, 2019-08-25:
> ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
/usr/local/bin/raspiBackup.sh: Zeile 1216: system,ctl: Kommando nicht gefunden.

Das liegt an einem Typo bei Dir.
`sudo /usr/local/bin/raspiBackup.sh -o "systemctl stop cups && systemctl stop cron" -a "systemctl start cron && system,ctl start cups" /backup/`
muss
`sudo /usr/local/bin/raspiBackup.sh -o "systemctl stop cups && systemctl stop cron" -a "systemctl start cron && systemctl start cups" /backup/`
heissen.

>
/usr/local/bin/raspiBackup.sh: Zeile 943: printf: `)': Ungültiges Formatierungszeichen.

Das liegt vermutlich daran, dass Du noch noch eine alte Memory Extension benutzt. Das Format der Meldungen hat sich in einer Version die Du jetzt uebersprungen hast geaendert und bedeutet dass auch die neuesten Extensions benutzt werden muessen. Also diese einfach runterladen und dann sollte das Problem behoben sein.

> Liegt der Fehler (wie so häufig) irgendwo an der Bedienung oder ist das tatsächlich ein Bug?

Beides :wink: No 1 ist ein Typo bzw Bedienungsfehler waehrend No 2 ein *Feature* ist :smile: raspiBackup kann nicht pruefen ob die korrekten Extensionversionen benutzt werden. War aber auch nie geplant das zu tun.

HB9ERD, 2019-08-25:
Danke für die schnelle Antwort.

Neueste Memory Extensions runterladen? Hmm, ich finde im Benutzerhandbuch von raspiBackup dazu gar nichts und stehe ein wenig wie der Ochse vorm Berg da...  Wie muss ich da vorgehen oder wo steht das erklärt? Sorry für meine laienhafte Frage.

framps, 2019-08-25:
> ich finde im Benutzerhandbuch von raspiBackup dazu gar nichts

Es [steht schon da](https://www.linux-tips-and-tricks.de/de/raspibackup/#wrapper), und etwas versteckt dann der [eigentliche Link](https://www.linux-tips-and-tricks.de/de/13-raspberry/442-raspibackup-erweiterungen/). 

Deine Frage hat mir auch gezeigt dass die Doku zu den Extensions etwas outdated war. Ich habe sie jetzt wieder auf Vordermann gebracht. Es gibt drei verschiedene Möglichkeiten sie zu installieren und Du kannst die Dir genehmste aussuchen :wink:

framps, 2019-09-11:
@HB9ERD Wie sieht es aus? Ist die Fehlermeldung mit den neuen Extensions weg? Dann bitte den Issue schliesen :wink:

framps, 2019-09-27:
Closing because of inactivity from issue creator

closed 2019-09-27 by framps