framps/raspiBackup #162: raspiBackupWrapper mit pishrink funktioniert nicht

framps, 2019-08-29:
Der erste Punkt war das wget nach Github aus Deiner Anleitung, hier kamen nicht die Skripte sondern die Webseiten. Das war schnell gelöst, nachdem ich die RAW-Links von Github benutzte.
Aber wenn ich jetzt ein Backup per Wrapper starte, dann läuft das eigentliche Backup gut durch (dd) aber pishrink wird nicht aufgerufen:
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0167I: Eine eMail wird versendet.
--- RBK0010I: fhem: raspiBackup.sh V0.6.4.3 (2d927a2) Do 29. Aug 10:07:51 CEST 2019 beendet.
Backup succeeded :-)
sudo: pishrink: Befehl nicht gefunden

Er ruft die Funktion nicht auf, die weiter oben im Skript für pishrink definiert ist?

framps, 2019-08-29:
Fixed with https://github.com/framps/raspiBackup/commit/e4ebafb097b6df5d65b532befb2a8b674da3f5f9

upD8R, 2019-08-29:
Verified to work!

framps, 2019-08-29:
Great. Thank you very much for verifying.

closed 2019-08-29 by framps