framps/raspiBackup #165: Shrinking root partition doesn't work any more

framps, 2019-09-05:
If the backup root partition size > target SD shrinking of partition seems not to work during restore

framps, 2019-09-05:
I opened the issue on behalf of Serge and just tried to reproduce the issue - but fail :cry: Resize works for me. Would be great if Serge adds some more details about his issue.

DesertRider, 2019-09-05:
Here are some details:

Output of the restore command:
```
sudo raspiBackup.sh -1 -d /dev/sda .
--- RBK0009I: mapvpn20.corp.shq.local: raspiBackup.sh V0.6.4.3 (2d927a2) started at Thu  5 Sep 15:26:06 EDT 2019.
--- RBK0128I: Using logfile /backup/mapvpn20.corp.shq.local/mapvpn20.corp.shq.local-rsync-backup-20190905-000301/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
mmcblk0p1
mmcblk0p2
!!! RBK0143W: Target /dev/sda does not match with backup. Partitioning forced.
??? RBK0160E: Target /dev/sda with 29.22 GiB is smaller than backup source with 29.80 GiB.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed with RC 102. Check previous error messages.
--- RBK0010I: mapvpn20.corp.shq.local: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Thu  5 Sep 15:26:11 EDT 2019.

```

and attached is the raspiBackup.log and raspiBackup.msg renamed as .msg.txt)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/3580916/raspiBackup.log)
[raspiBackup.msg.txt](https://github.com/framps/raspiBackup/files/3580918/raspiBackup.msg.txt)

Tell me if you need more informations/further tests!

framps, 2019-09-05:
@DesertRider Thank you very much for providing more details about your issue.

I didn't know you use the partition oriented mode (not used by a lot of users). Please see https://github.com/framps/raspiBackup/tree/issu165 for a fix.

framps, 2019-09-06:
Just another comment: Resize is not supported when using option -P. So the only way to restore a -P backup to a smaller SD card is to use option -1 or manually create all partitions beforehand on the SD card and use option -0.

DesertRider, 2019-09-06:
Fix for #165 work fine, tested with 2 differents keys/microSD cards

framps, 2019-09-06:
@DesertRider :+1: Thank you very much for verifying.

closed 2019-09-06 by framps