framps/raspiBackup #173: Boot Devices not detected

mars-online, 2019-09-19:
The script asked me to report this issue, so I do.

Directly after a reboot of my pi a backup usually can be performed without any issue. After a few days the script exits with the following errors:

```bash
$ sudo /usr/local/bin/raspiBackup.sh -k 10   -t ddz -a "service apache2 start" -o "service apache2 stop" /backup -l Debug
```
<details><summary>output</summary>
<p>

```
--- RBK0009I: myHost: raspiBackup.sh V0.6.4.3 (2d927a2) started at Do 19. Sep 21:36:03 UTC 2019.
--- RBK0128I: Using logfile /backup/myHost/myHost-ddz-backup-20190919-213600/raspiBackup.log.                                                     --- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
??? RBK0203E: Unable to discover boot device. Please report this issue with a debug log created with option '-l debug'.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /backup/myHost/myHost-ddz-backup-20190919-213600. This will take some time. Please be patient.
--- RBK0049I: Messages saved in /home/pi/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: myHost: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Do 19. Sep 21:36:25 UTC 2019.
```
</p>
</details>

```bash
$ cat  /etc/os-release
```
<details><summary>output</summary>
<p>

```bash
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
</p>
</details>

```
$ cat raspiBackup.log
```
<details><summary>output</summary>
<p>

```bash
20190919-213603 DBG 1559:     <- substituteNumberArguments 
20190919-213603 DBG 5451:     -> check4RequiredCommands 
20190919-213603 DBG 5472:     <- check4RequiredCommands 
20190919-213604 DBG 6212:  -- RESTORE: 0 - fileParameter: /backup
20190919-213604 DBG 6231:  -- Enabling trap handler
20190919-213604 DBG 1113:  -- TRAP cleanup SIGINT SIGTERM EXIT
20190919-213604 DBG 2162:     -> setupEnvironment 
20190919-213604 DBG 2210:     -- Current logfiles: L: /backup/raspiBackup.log M: /backup/raspiBackup.msg
20190919-213605 DBG 2267:     -- LOG_OUTPUT: 2
20190919-213605 DBG 2268:     -- Using logfile /backup/myHost/myHost-ddz-backup-20190919-213600/raspiBackup.log
20190919-213605 DBG 2269:     -- Using msgfile /backup/myHost/myHost-ddz-backup-20190919-213600/raspiBackup.msg
20190919-213605 DBG 2277:     -- BACKUPTARGET_DIR: /backup/myHost/myHost-ddz-backup-20190919-213600
20190919-213605 DBG 2278:     -- BACKUPTARGET_FILE: /backup/myHost/myHost-ddz-backup-20190919-213600/myHost-ddz-backup-20190919-213600.img.gz
20190919-213605 DBG 2280:     <- setupEnvironment 
20190919-213605 DBG 1317:     -> logOptions 
20190919-213605 DBG 1319:     -- Linux myHost 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux
20190919-213606 DBG 1321:     -- Options:  -k 10 -t ddz -a service apache2 start -o service apache2 stop /backup -l Debug
20190919-213606 DBG 1322:     -- APPEND_LOG=0
[... 180 lines cut ...]
20190919-213623 MSG 2517:                    --- RBK0043I: Removing incomplete backup in /backup/myHost/myHost-ddz-backup-20190919-213600. This will take some time. Please be patient. 
20190919-213624 DBG 2518:                    -- total 2
20190919-213624 DBG 2518:                    -- drwxr-xr-x 2 pi pi     0 Sep 19  2019 .
20190919-213624 DBG 2518:                    -- drwxr-xr-x 2 pi pi     0 Sep 19  2019 ..
20190919-213624 DBG 2518:                    -- -rw-r--r-- 1 pi pi 13989 Sep 19  2019 raspiBackup.log
20190919-213624 DBG 2518:                    -- -rw-r--r-- 1 pi pi   427 Sep 19  2019 raspiBackup.msg
--- RBK0049I: Messages saved in /home/pi/raspiBackup.msg.
20190919-213624 MSG 2531:                    --- RBK0049I: Messages saved in /home/pi/raspiBackup.msg. 
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.
20190919-213624 MSG 2533:                    --- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log. 
20190919-213624 DBG 2536:                    <- cleanupBackupDirectory 
??? RBK0005E: Backup failed. Check previous error messages for details.
20190919-213624 MSG 2754:                 ??? RBK0005E: Backup failed. Check previous error messages for details. 
Invocation parms: ' -k 10 -t ddz -a service apache2 start -o service apache2 stop /backup -l Debug'
20190919-213624 DBG 2379:                    -> sendEMail 
20190919-213624 DBG 2477:                    <- sendEMail 
20190919-213624 DBG 2779:                 <- cleanupBackup 
20190919-213624 DBG 2785:                 -> cleanupTempFiles 
20190919-213625 DBG 2792:                 <- cleanupTempFiles 
20190919-213625 DBG 2568:              -- Terminate now with rc 121
--- RBK0010I: myHost: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Do 19. Sep 21:36:25 UTC 2019.
20190919-213625 MSG 2571:              --- RBK0010I: myHost: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Do 19. Sep 21:36:25 UTC 2019. 
```
</p>
</details>

framps, 2019-09-20:
I'm sorry, you found a bug. I just created a hot fix 78cfce81ec38e75eeae6c37157ee9c921966eb58 for you. Please grab raspiBackup.sh from this branch https://github.com/framps/raspiBackup/tree/issue173 and verify the fix.

mars-online, 2019-09-25:
```
$ sudo /home/pi/raspiBackup.sh -k 10   -t ddz -a "service apache2 start" -o "service apache2 stop" /backup
```
<details><summary>Seams to work</summary>
<p>

--- RBK0009I: 3hawk: raspiBackup.sh V0.6.4.3-hotfix (ec8e864) started at Mi 25. Sep 16:45:26 UTC 2019.
!!! RBK0173W: =========> NOTE  <========= 
!!! RBK0173W: This is a temporary hotfix and has to be upgraded to next available version as soon as one is available. 
!!! RBK0173W: =========> NOTE <=========
--- RBK0128I: Using logfile /backup/hostname/hostname-ddz-backup-20190925-164525/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0151I: Using backuppath /backup.
--- RBK0008I: Stopping services: 'service apache2 stop'.
Stopping web server: apache2 ... waiting ..
--- RBK0081I: Creating backup of type ddz in /backup/hostname/hostname-ddz-backup-20190925-164525.
--- RBK0085I: Backup of type ddz started. Please be patient.
</p>
</details>

closed 2019-09-25 by mars-online