framps/raspiBackup #176: Resizing /boot partition

christianTF, 2019-09-29:
Hello framp,

we currently have an issue in LoxBerry-Core to update our users Pi to LoxBerry 2.0, and possibly RaspiBackup could help us out:

Updating the Pi firmware leads to the problem, that rpi-update installs both 32- and 64-bit kernel in /boot, but on existing Raspberry’s the /boot partition is too small.
New images don’t have the problem, as Raspian already increased the /boot size, but we cannot increase it on running Raspberry’s.

To not need to drop Pi1/Zero Support, our idea was creating an dd backup with RaspiBackup, and let the user recover the image on a new SD.
The question is: Is there an option to resize the /boot partition in the DD backup (on *writing* of the image), so the finished image would recover with e.g. Win32Diskimager with a resized /boot partition?
This would be the easiest procedure for the users, as they don’t need to do any special treatment.
If there is no option, is it technically possible inside RaspiBackup, and could you point out the code part we would need to adjust?

Thanks a lot,
Christian 

@mschlenstedt
@svethi

framps, 2019-09-29:
Hello Christian,

raspiBackup doesnt support resizing of the boot partition - unfortunately. There exists option -0 for restore which forces raspiBackup to use the existing partitioning on the SD card. So you can prime an SD card with 256MB boot partition and  allocate the rest for the root partition and restore the backup. But this works only for tar and rsync backups :cry:

A dd backup is a 1:1 copy of the SD card including the partition table. I found [url=https://www.raspberrypi.org/forums/viewtopic.php?t=187999#p1349833]this post[/url] which explains all manual steps how to use a dd img (which can be the raspiBackup dd image) and to increase the size of the boot partition on a new SD card. I also found [url=https://learn.adafruit.com/resizing-raspberry-pi-boot-partition/edit-partitions]this page[/url] which uses gparted. But I think you're looking for a more user friendly way to get the boot partition resized - e.g. get support in raspiBackup for this issue.

I think the best way to solve the issue is to create a small script which uses the existing dd backup and executes the steps mentioned in the first link. Then a user has to execute this script once to create a larger boot partition and a smaller root partition and then restore the data from the two partitions in the dd backup on a new SD card. Unfortunately there doesn't exist a restore pre and post exit in raspiBackup. Latter would be the right place to kick off the conversion script.

If you wish I can create this script and commit it into the helper directory. But I will not spend any time for error checking and to make the script error prone. This will take too much time. But then you can grab this script as a starting point and either customize, enhance and integrate it into your loxberry plugin :wink:

christianTF, 2019-09-29:
Thanks for the response and the investigation.
We discuss the proposals internally, and would come back to you!
Thanks!
Christian

framps, 2019-10-21:
@christianTF Any news on your internal discussions?

christianTF, 2019-10-23:
Hello framp,
we currently suspended the resize of the /boot partition as we could work-around the Raspberry Pi 4 issue by not deploying the 64-bit kernel on LoxBerry. So we now can update older Raspberrys with their 64MB /boot partition and will investigate further when we finally can release LoxBerry 2.0.

Therefore, I close the issue, and we will continue the investigation with your proposals in a LoxBerry issue.

Thanks again!
Christian

closed 2019-10-23 by christianTF

framps, 2020-05-02:
@christianTF FYI:
I just had a user who updated stretch and wanted to know how he can use raspiBackup to get the boot partition resized to 256M. It's possible with raspiBackup:
1) Create boot partition of 256M on SD card
2) Create root partition on the SD card
3) format /boot and / accordingly
4) Restore a backup created by raspiBackup and use option -0 which forces raspiBackup to use the existing partitioning on the SD card.