framps/raspiBackup #177: Feature request: mini standalone backups for cloud upload

ephestione, 2019-09-29:
Would there be a "smart" way to build "almost comprehensive" backups that are small enough to be uploaded to the cloud?
I already make periodic backups (not with raspibackup I must admit, I use rpimage which creates a flashable img file directly from the rsync of a running system without stopping any services -ugly I know but it works for me; even if I am thinking about switching since now RaspiBackup has the restore2image tool which back at the time didn't exist.

Still, I feel the need to store backups (encrypted of course) on the cloud, and uploading a few GBs each time isn't feasible.

I already created something for my own use, which is extremely sour and brutal, but it's a start:
https://github.com/ephestione/bazidrop
anyhow, I think something way better can be done using raspibackup structure, and probably with a better background reasoning than my own.

ephestione, 2019-09-29:
In my idea, it should be a technique that, with very few efforts, should be able to restore the system to the working order of the original system, but instead of restoring the whole OS, you will be restoring the "smart backup" inside a vanilla flash of Raspbian

framps, 2019-09-29:
[pishrink](https://github.com/Drewsif/PiShrink) is a nice tool to compress an image to it's minimum size. It's not used by raspiBackup because raspiBackups purpose is to create a backup - nothing more. You use 7z in your code to compress the image. I frankly have no idea which compression method is more efficient. But please have a look into [raspiBackupWrapper](https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupWrapper.sh) which uses pishrink to compress the backup. It shouldn't be a big deal to use 7z instead of pishrink or even concatenate pishrink and 7z. If you plan any investigations on this please let us know about your results - or even better - if you create a new wrapper script using raspiBackup please create a PR.

I agree with you it would be a nice feature for raspiBackup to compress the backup but I already decided not to integrate pishrink into raspiBackup because of the [Linux philosophy](https://en.wikipedia.org/wiki/Unix_philosophy). raspiBackup just creates backups. That's it. For compression the backup there exist tons of other tools which can be used later on. Unfortunately raspiBackup cannot be used in pipes but that's why I created some wrapper scripts as a workaround.

ephestione, 2019-09-29:
Hey, thanks for the fast reply.
Yeah, 7zip helps with reducing backup size since LZMA is more efficient (albeit slower) than GZip, yet this is only a strategy, as the primary concept is that my backup tool does not replicate the whole OS, it selects just the things that you need to "rebuild" your installation from a plain raspbian image.

It will only backup /etc, /home, and some other things (like crontabs, mysql databases and list of installed packages) so you get a final backup that's less than 100MB (my systems go up to max 7MB, one of them just 400Kb).
Then, you have to download the latest raspbian image, flash it, and then, basically, decompress the old backup on it, and running apt to restore the list of packages.

Just adding pre- or post- processing to RaspiBackup wouldn't cut it I think, because I would need to remove rather than add!

Maybe this request is in fact too distant from the original concept of raspibackup, and I probably asked to add a functionality that strays quite a bit from your roadmap

framps, 2019-09-29:
> Maybe this request is in fact too distant from the original concept of raspibackup, and I probably asked to add a functionality that strays quite a bit from your roadmap

I had two primary goals in mind for raspiBackup:

1) Create a full backup to ease restore
2) Provide all information to manual restore a backup with Linux tools

There are much more high sophisticated backup and restore tools out there which need much less backup space but - to put it bluntly - disk space is cheep nowadays - and if you use rsync the backup space and time overhead is quite small :wink: 

But please don't be afraid I don't accept your feature request. I like you're idea to use 7z instead of pishrink if the compression ratio is better than the one of pishrink. Feel free to create a PR for 7z usage :smiley:

ephestione, 2019-09-29:
Yes, in fact disk space is cheap, and that's not the reason I'm looking for this functionality :wink: 
Cloud space tho, that's more expensive, and I was looking to get "smart small backups" to upload to the cloud and keep stored in someone else's servers (maintained and redundant way better than mine) in case my own local storage fails for any reason.
How small? Let's say 10MB for a not overly complicated system.

framps, 2019-09-29:
Sorry - I missed you want to use cloud space as backup space :cry: 

raspiBackup was not designed for this scenario - unfortunately. I already got some feedback this is a requirement for other people also and most probably will become more and more important in the future. When we will get Gb upload speed from your ISP you will be able to use raspiBackup. But the design of raspiBackup to create a full backup needs bandwidth which isn't available as of now :disappointed:

ephestione, 2019-09-29:
Rather than upload speed (I have 3MB/s upload speed and would be doing it during the night, no big issue) I wouldn't want to fill up all my free dropbox space with 4GB multiple backups for multiple RPi's!
And I have no intention to pay for cloud storage :smile_cat: 

Anyhow, regarding backup size reduction, pishrink and 7z work on different angles. pishrink removes empty space at the end of the image, 7z compresses data, so even if slow, I expect pishrink+7z to yield better results than just 7z

closed 2019-09-29 by ephestione

christianTF, 2019-09-29:
I can provide some input to you, @ephestione , from own experience with other projects.

Other than zip, 7zip is not capable to directly pipe compressed content to a remote file. That means, also if it seems to be piping, it creates a temporary file in /tmp that is copied to the pipe if 7zip has finished. On a Raspberry that leads to HEAVY sd write cycles, or, if /tmp is tmpfs, the 7z operation will fail because of missing “disk” (ram)  space.

Best option for Cloud Storage, if you also manage the server, is to run an rsync server on the destination, as rsync provides block based sync of files using hashes, and including compression. 

Another option would be to mount the cloud storage as directory on the Raspberry. That would require some driver for the cloud storage, or using smb or nfs. In that way, rsync type of RaspiBackup might save you Transfer load by using links.

lg, Christian

framps, 2019-09-29:
@christianTF :+1: Appreciate your comment

ephestione, 2019-09-30:
@christianTF thank you for the rich insight!
Those are indeed useful pointers to consider when planning a "all round and scalable" backup solution, which I didn't consider to be honest, because I was looking just at my personal use-case.
For example, I don't even associate "SD card" to raspberry anymore. The only SD's I use on my raspi's are those needed for bootcode.bin where the pendrive won't boot by itself, or the one holding the /boot on my raspi4 until usb boot is released.
As my intention is to have a really small backup, using tmpfs will be permissive with about 20-30MB of data that can get compressed to 7MB, and out of sheer curiosity I tested 7z with ultra settings on my 4GB (3+GB worth of real data), after quite a while it produced a 670MB archive. Alright for a tmpfs on a 4GB raspi4, but yeah, overkill for anything else.