framps/raspiBackup #179: External Root partition not saved

FrColin, 2019-10-07:
I have got a raspi2 with an SD for /boot and a SSD for /

as it is said in the FAQ raspiBackup Shoud backup the external root partition
it only backup the  Partition mmcblk0p1.
not the sda2 partition
backup type is tar 
version is 0.6.4.3
mount is :
/dev/sda2 on / type ext4 
/dev/mmcblk0p1 on /boot type vfat 

/boot/cmdline.txt is: 
dwc_otg.lpm_enable=0 console=tty1 root=/dev/sda2 rootfstype=ext4 elevator=deadline fsck.repair=yes  rootwait

framps, 2019-10-07:
Hm ... whenever I executed any tests with an external partition I used /dev/sdx**1**. It may be related to the fact you use /dev/sda**2**. Would you please provide the backup debug log?

framps, 2019-10-08:
Thank you very much for the log which helped me to find the root cause for your issue: You use the partition oriented backup mode (option -P) which does not support an external root filesystem. But from the log I can see you only have two partitions. So I'm wondering why you don't use the normal mode which supports an external root partition. Just change `PARTITIONBASED_BACKUP=1` into `PARTITIONBASED_BACKUP=0` in /usr/local/etc/raspiBackup.conf with an editor or use the installer to change the mode.

I will add a check when -P is is used whether an external root filesystem exists and write an error message in next release. 

Why did you decide to use the partition oriented mode? Is there any place I missed to document that an external root partition is not supported with option -P? Please let me know and I will also update the doc so other users of raspiBackup will not run into this error.

FrColin, 2019-10-10:
Thank you for your help , 
I miss understand the partition mode,
in my conf I put  PARTITIONBASED_BACKUP=0 and DEFAULT_PARTITIONBASED_BACKUP=0
and it is working
what is the difference beween the two variable ?
the doc https://www.linux-tips-and-tricks.de/en/backup/#butypes
may mention that the Partition Mode didn't save external root partition

framps, 2019-10-10:
Great it works now :+1: 

`DEFAULT_PARTITIONBASED_BACKUP=0` in the config file is the only update you have to do. But the other line doesn't harm - it's just meaningless :wink: 

And yes, you are right. I will update the explanation of the two modes and make clear an external partition is not supported in partition oriented backup.

Would you please close the issue now?

FrColin, 2019-10-10:
again thank you very much