framps/raspiBackup #181: NAS integration

Snowplaytv, 2019-10-15:
Könnten sie auch eine Anleitung veröffentlichen in dem erklärt wird wie man Backup speicher benutzt der nicht am Raspberry angeschlossen ist ? z.B. ein NAS oder die Fritzbox mit einer NAS Funktion ?

framps, 2019-10-15:
Im Prinzip ist das wohl moeglich. Aber dann wuerde ich nur irgendwelche Anleitungen wie man ein Laufwerk per Samba Client oder nfs Client auf einem Linux mounted kopieren. Man muss im Netz nur nach samba oder nfs suchen im Zusammenhang mit Linux oder Raspberry. Ich habe auf die Schnelle [diesen Link](https://www.elektronik-kompendium.de/sites/raspberry-pi/2102191.htm) gefunden der weitere Links zu dem Thema hat. Ansonsten ist eine gute Anlaufstelle dieses [Raspberry Forum](https://forum-raspberrypi.de/forum/). Im Tutorialteil findet man sicherlich Anleitungen und kann bei Problemen Fragen stellen.

framps, 2019-10-28:
@Snowplaytv Hast Du eine Anleitung gefunden und kannst nun erfolgreich raspiBackup mit einer NAS benutzen? Dann bitte den Issue closen :wink:

framps, 2019-11-05:
Closed because of incactivity

closed 2019-11-05 by framps