framps/raspiBackup #187: Use pigz instead of gzip to compress dd backups

framps, 2019-12-08:
pigz uses all available cores to compress/uncompress data which is much faster than gzip.

Hofei90, 2020-02-20:
Pigz kenn ich nicht, aber hier mal ein kleiner Hinweis und Warnung:
https://forum-raspberrypi.de/forum/thread/43534-kurze-programmvorstellung-pbzip2-lbzip2-komprimierung-mit-mehreren-kernen/
Die Warnung ist in Beitrag 10

framps, 2020-03-01:
Moved into backlog

closed 2020-03-01 by framps

Raptor039, 2022-01-30:
Hi @framps and many thanks for your tool! I'm using it since months and it's very reliable.

Is this option still in a work in progress state?

It would be nice and usefull in saving time to be able to use multithreaded gzip compression whend doing a dd backup images.

Hope it will be implemented soon 🙂

framps, 2022-01-30:
You're the second one who asks for this support. Frankly this is a kind of exotic feature I'll implement when I have a lot of spare time which I don't have right now. Sorry about this.