framps/raspiBackup #188: Shrinking Problem with RaspiBackupWrapper

stm-clemens, 2019-12-17:
Hey,

Well then i'll try it here :)

I try to shrink my weekly raspberry backup with pishrink via raspiBackupWrapper.
Right now it fails with the following error:

```
12-16-2019 09:56:50 --- RBK0010I: ubuntu: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Mo 16. Dez 09:56:50 CET 2019.
Backup succeeded :-)
pishrink.sh v0.1.2
pishrink.sh: Gatherin data...
md5sum: /tmp/tmp.41jRvuDZjh/etc/rc.local: No such file or directory
Creating new /etc/rc.local
mv: cannot stat '/tmp/tmp.41jRvuDZjh/etc/rc.local': No such file or directory
pishrink.sh: Checking filesystem...
e2fsck: Invalid argument while trying to open /dev/loop4
/dev/loop4:
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

pishrink.sh: Filesystem error detected!...
pishrink.sh: Trying to recover corrupted filesystem...
e2fsck 1.45.3 (14-Jul-2019)
e2fsck: Invalid argument while trying to open /dev/loop4

The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

pishrink.sh: ERROR occured in line 58: Filesystem recoveries failed. Giving up...
```

when I run pishrink by hand it works just fine.

Im on ubuntu 19.10 64bit raspi4

is there anything I can do to get it working I cant flood my server with 200gb backup files :,D

framps, 2019-12-17:
I need more information. Would you please modify the first line in pishrink 
from
`#!/bin/bash`
to
`#!/bin/bash -x`
and call the wrapper script again? You will get a lot of messages now from pishrink. Please show them to me.

stm-clemens, 2019-12-17:
```
ubuntu@ubuntu:/usr/local/etc$ sudo ./raspiBackupWrapper.sh 
+ set -euf -o pipefail
+ MYSELF=raspiBackupWrapper.sh
+ MYNAME=raspiBackupWrapper
+ VERSION=0.2.6
+ set +u
+ GIT_DATE=': 2019-08-29 11:09:42 +0200$'
+ set -u
+ GIT_DATE_ONLY='2019-08-29 11:09:42 +0200$'
++ cut -f 2 -d ' '
+ GIT_DATE_ONLY=2019-08-29
++ cut -f 3 -d ' '
+ GIT_TIME_ONLY=11:09:42
+ set +u
+ GIT_COMMIT=': a009fd8$'
+ set -u
++ cut -f 2 -d ' '
++ sed 's/\$//'
+ GIT_COMMIT_ONLY=a009fd8
+ GIT_CODEVERSION='raspiBackupWrapper.sh 0.2.6, 2019-08-29/11:09:42 - a009fd8'
+ LOOP_DISK_NAME=
+ BACKUP_MOUNT_POINT=/mnt/backup/
+ BACKUP_PATH=/mnt/backup//live/ubuntu_raspberry
+ LOOP_MOUNTED=0
+ [[ -e /bin/grep ]]
+ PATHES='/usr/local/sbin /usr/local/bin /usr/sbin /usr/bin /sbin /bin'
+ for p in $PATHES
+ /bin/grep -E -q '[^:]/usr/local/sbin[:$]'
+ [[ -z /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin ]]
+ export PATH=/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ PATH=/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ for p in $PATHES
+ /bin/grep -E -q '[^:]/usr/local/bin[:$]'
+ [[ -z /usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin ]]
+ export PATH=/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ PATH=/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ for p in $PATHES
+ /bin/grep -E -q '[^:]/usr/sbin[:$]'
+ [[ -z /usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin ]]
+ export PATH=/usr/sbin:/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ PATH=/usr/sbin:/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ for p in $PATHES
+ /bin/grep -E -q '[^:]/usr/bin[:$]'
+ [[ -z /usr/sbin:/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin ]]
+ export PATH=/usr/bin:/usr/sbin:/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ PATH=/usr/bin:/usr/sbin:/usr/local/bin:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
+ for p in $PATHES
+ /bin/grep -E -q '[^:]/sbin[:$]'
+ for p in $PATHES
+ /bin/grep -E -q '[^:]/bin[:$]'
+ trapWithArg cleanup SIGINT SIGTERM EXIT
+ local func
+ func=cleanup
+ shift
+ for sig in "$@"
+ trap 'cleanup SIGINT' SIGINT
+ for sig in "$@"
+ trap 'cleanup SIGTERM' SIGTERM
+ for sig in "$@"
+ trap 'cleanup EXIT' EXIT
+ isMounted /mnt/backup/
+ local path
+ path=/mnt/backup/
+ [[ /mnt/backup/ != '' ]]
+ mountpoint -q /mnt/backup/
+ return 0
+ WAS_MOUNTED=1
+ echo '--- /mnt/backup/ already mounted'
--- /mnt/backup/ already mounted
[... 52 lines cut ...]
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

pishrink.sh: Filesystem error detected!...
pishrink.sh: Trying to recover corrupted filesystem...
e2fsck 1.45.3 (14-Jul-2019)
e2fsck: Invalid argument while trying to open /dev/loop5

The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

pishrink.sh: ERROR occured in line 58: Filesystem recoveries failed. Giving up...
+ cleanup EXIT
+ ((  ! 1  ))
+ ((  0  ))
```

Thats the Output with -x

framps, 2019-12-17:
Thank you very much for the debug log. But unfortunately you added `-x` in raspiBackupWrapper by error :cry:  Please add this option in pishrink.

stm-clemens, 2019-12-17:
Damn :,D sorry my bad.

```
12-17-2019 17:28:47 --- RBK0010I: ubuntu: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Di 17. Dez 17:28:47 CET 2019.
Backup succeeded :-)
+ version=v0.1.2
++ pwd
+ CURRENT_DIR=/usr/local/etc
+ SCRIPTNAME=pishrink.sh
+ LOGFILE=/usr/local/etc/pishrink.log
+ should_skip_autoexpand=false
+ debug=false
+ repair=false
+ gzip_compress=false
+ getopts :sdrzh opt
+ case "${opt}" in
+ gzip_compress=true
+ getopts :sdrzh opt
+ shift 1
+ '[' false = true ']'
+ echo 'pishrink.sh v0.1.2'
pishrink.sh v0.1.2
+ src=/mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img
+ img=/mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img
+ [[ -z /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img ]]
+ [[ ! -f /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img ]]
+ ((  EUID != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v parted
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v losetup
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v tune2fs
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v md5sum
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v e2fsck
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v resize2fs
+ ((  0 != 0  ))
+ '[' -n '' ']'
+ trap cleanup ERR EXIT
+ info 'Gatherin data'
+ echo 'pishrink.sh: Gatherin data...'
pishrink.sh: Gatherin data...
++ ls -lh /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img
++ cut -d ' ' -f 5
+ beforesize=184G
++ parted -ms /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img unit B print
++ tail -n 1
+ parted_output='2:269484032B:196865932799B:196596448768B:ext4::;'
++ echo '2:269484032B:196865932799B:196596448768B:ext4::;'
++ cut -d : -f 1
+ partnum=2
++ echo '2:269484032B:196865932799B:196596448768B:ext4::;'
++ cut -d : -f 2
++ tr -d B
+ partstart=269484032
++ losetup -f --show -o 269484032 /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img
+ loopback=/dev/loop5
++ tune2fs -l /dev/loop5
+ tune2fs_output='tune2fs 1.45.3 (14-Jul-2019)
Filesystem volume name:   writable
Last mounted on:          /
Filesystem UUID:          2fae1f20-4a20-4c77-a39b-fb6480f64a0b
[... 185 lines cut ...]
+ e2fsck -y /dev/loop5
e2fsck 1.45.3 (14-Jul-2019)
e2fsck: Invalid argument while trying to open /dev/loop5

The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

+ ((  8 < 4  ))
+ [[ false == true ]]
+ error 58 'Filesystem recoveries failed. Giving up...'
+ echo -n 'pishrink.sh: ERROR occured in line 58: '
pishrink.sh: ERROR occured in line 58: + shift
+ echo 'Filesystem recoveries failed. Giving up...'
Filesystem recoveries failed. Giving up...
+ exit -9
+ cleanup
+ losetup /dev/loop5
+ '[' false = true ']'

```

framps, 2019-12-18:
Thank you very much for the log. Would you now please call pishrink directly (also with added -x) against /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img? That way I will get the difference behavior between the direct and indirect invocation of pishrink.

stm-clemens, 2019-12-19:
```
ubuntu@ubuntu:~$ sudo pishrink.sh -z /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img 
pishrink.sh v0.1.2
pishrink.sh: Gatherin data...
pishrink.sh: Checking filesystem...
writable: Inode 765430, end of extent exceeds allowed value
	(logical block 32, physical block 498720, len 88)

writable: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
pishrink.sh: Filesystem error detected!...
pishrink.sh: Trying to recover corrupted filesystem...
e2fsck 1.45.3 (14-Jul-2019)
writable contains a file system with errors, check forced.
Pass 1: Checking inodes, blocks, and sizes
Inode 765430, end of extent exceeds allowed value
	(logical block 32, physical block 498720, len 88)
Clear? yes

Inode 765430, i_blocks is 960, should be 264.  Fix? yes

Inode 4289308, end of extent exceeds allowed value
	(logical block 893, physical block 17878922, len 95)
Clear? yes

Inode 4289308, i_blocks is 7904, should be 7152.  Fix? yes

Pass 2: Checking directory structure
Entry 'replorigin_checkpoint' in /home/ubuntu/docker/postgres/pg_data/pg_logical (1644296) has deleted/unused inode 1649605.  Clear? yes

Pass 3: Checking directory connectivity
Pass 3A: Optimizing directories
Inode 6999 extent tree (at level 1) could be shorter.  Optimize? yes

Inode 23189 extent tree (at level 1) could be shorter.  Optimize? yes

Inode 35342 extent tree (at level 1) could be shorter.  Optimize? yes

Pass 4: Checking reference counts
Pass 5: Checking group summary information
Block bitmap differences:  -(498720--498806) +(531581--531582) -(594413--594414) +(2884094--2884095) -3023308 -6933103 +6933263 -(14738975--14738977) -(14754879--14754880) -(14754966--14754967) -14755015 +14766292 +(14766352--14766353) +(14766390--14766391) -(17878922--17879016)
Fix? yes

Free blocks count wrong for group #15 (1616, counted=1703).
Fix? yes

Free blocks count wrong for group #16 (9920, counted=9918).
Fix? yes

Free blocks count wrong for group #18 (7744, counted=7746).
Fix? yes

Free blocks count wrong for group #92 (17411, counted=17412).
Fix? yes

Free blocks count wrong for group #545 (8984, counted=9078).
Fix? yes

Free blocks count wrong (43262864, counted=43263046).
Fix? yes

Inode bitmap differences:  -1649509
Fix? yes

Free inodes count wrong for group #209 (3396, counted=3397).
Fix? yes

Free inodes count wrong (10936066, counted=10936067).
Fix? yes

writable: ***** FILE SYSTEM WAS MODIFIED *****
writable: 596413/11532480 files (0.2% non-contiguous), 4734133/47997179 blocks
resize2fs 1.45.3 (14-Jul-2019)
pishrink.sh: Shrinking filesystem...
resize2fs 1.45.3 (14-Jul-2019)
Resizing the filesystem on /dev/loop5 to 4293986 (4k) blocks.
Begin pass 2 (max = 1369194)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 1465)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 88729)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop5 is now 4293986 (4k) blocks long.

pishrink.sh: Shrinking image...
pishrink.sh: Gzipping the shrunk image...
pishrink.sh: Shrunk /mnt/backup/ubuntu/ubuntu-dd-backup-20191217-161206/ubuntu-dd-backup-20191217-161206.img.gz from 184G to 6,7G...
```

framps, 2019-12-19:
Thank you very much. We make progress :+1: 

Unfortunately there was no `-x` in pishrink in the `#/bin/bash` line :cry: . 
1. Would you please add `-x` in pishrink and update the previous log with the detailed log you then will get?
2. I believe you answered the prompts from pishrink on your console - right?

stm-clemens, 2019-12-20:
Again :,D

```
ubuntu@ubuntu:/mnt/backup/ubuntu/ubuntu-dd-backup-20191220-071959$ sudo pishrink.sh -z ubuntu-dd-backup-20191220-071959.img 
+ version=v0.1.2
++ pwd
+ CURRENT_DIR=/mnt/backup/ubuntu/ubuntu-dd-backup-20191220-071959
+ SCRIPTNAME=pishrink.sh
+ LOGFILE=/mnt/backup/ubuntu/ubuntu-dd-backup-20191220-071959/pishrink.log
+ should_skip_autoexpand=false
+ debug=false
+ repair=false
+ gzip_compress=false
+ getopts :sdrzh opt
+ case "${opt}" in
+ gzip_compress=true
+ getopts :sdrzh opt
+ shift 1
+ '[' false = true ']'
+ echo 'pishrink.sh v0.1.2'
pishrink.sh v0.1.2
+ src=ubuntu-dd-backup-20191220-071959.img
+ img=ubuntu-dd-backup-20191220-071959.img
+ [[ -z ubuntu-dd-backup-20191220-071959.img ]]
+ [[ ! -f ubuntu-dd-backup-20191220-071959.img ]]
+ ((  EUID != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v parted
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v losetup
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v tune2fs
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v md5sum
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v e2fsck
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v resize2fs
+ ((  0 != 0  ))
+ '[' -n '' ']'
+ trap cleanup ERR EXIT
+ info 'Gatherin data'
+ echo 'pishrink.sh: Gatherin data...'
pishrink.sh: Gatherin data...
++ ls -lh ubuntu-dd-backup-20191220-071959.img
++ cut -d ' ' -f 5
+ beforesize=184G
++ parted -ms ubuntu-dd-backup-20191220-071959.img unit B print
++ tail -n 1
+ parted_output='2:269484032B:196865932799B:196596448768B:ext4::;'
++ echo '2:269484032B:196865932799B:196596448768B:ext4::;'
++ cut -d : -f 1
+ partnum=2
++ echo '2:269484032B:196865932799B:196596448768B:ext4::;'
++ cut -d : -f 2
++ tr -d B
+ partstart=269484032
++ losetup -f --show -o 269484032 ubuntu-dd-backup-20191220-071959.img
+ loopback=/dev/loop5
++ tune2fs -l /dev/loop5
+ tune2fs_output='tune2fs 1.45.3 (14-Jul-2019)
Filesystem volume name:   writable
Last mounted on:          /tmp/tmp.PF16iL2vZg
Filesystem UUID:          2fae1f20-4a20-4c77-a39b-fb6480f64a0b
Filesystem magic number:  0xEF53
[... 247 lines cut ...]
+ endresult=17989181952
+ logVariables 309 endresult
+ '[' false = true ']'
+ truncate -s 17989181952 ubuntu-dd-backup-20191220-071959.img
+ [[ true == true ]]
+ info 'Gzipping the shrunk image'
+ echo 'pishrink.sh: Gzipping the shrunk image...'
pishrink.sh: Gzipping the shrunk image...
++ gzip -f9 ubuntu-dd-backup-20191220-071959.img
+ [[ ! -n '' ]]
+ img=ubuntu-dd-backup-20191220-071959.img.gz
++ ls -lh ubuntu-dd-backup-20191220-071959.img.gz
++ cut -d ' ' -f 5
+ aftersize=6,7G
+ logVariables 324 aftersize
+ '[' false = true ']'
+ info 'Shrunk ubuntu-dd-backup-20191220-071959.img.gz from 184G to 6,7G'
+ echo 'pishrink.sh: Shrunk ubuntu-dd-backup-20191220-071959.img.gz from 184G to 6,7G...'
pishrink.sh: Shrunk ubuntu-dd-backup-20191220-071959.img.gz from 184G to 6,7G...
+ cleanup
+ losetup /dev/loop5
+ losetup -d /dev/loop5
+ '[' false = true ']'
```
And no I didn't answer any prompts.

framps, 2019-12-20:
Thank you very much for the log. 

Unfortunately you used a different backup image :cry: but I want to compare the pishrink run log against the same image which failed with the wrapper. 

The first one you used in raspiBackupWrapper is `ubuntu-dd-backup-20191217-161206.img` which has orphaned inodes. Because it's a clone of the SD card this means the SD card a backup was created from is already corrupt and thus the image cannot be shrinked with pishrink. The image you used with pishrink `ubuntu-dd-backup-20191220-071959.img` is OK and has no orphaned inodes. 

1. Please run `sudo fsck /dev/mmcblk0p2` to make sure the filesystem on the SD card is OK. 
2. Then create a backup with raspiBackupWrapper (pishrink including `-x`). 
3. Next run pishrink (including `-x`) against the image created in previous step. 

That way I get two logs from pishrink against the same image and can find differences :wink:

framps, 2019-12-30:
Ping @stm-clemens

framps, 2020-01-02:
@stm-clemens Any update on this issue and help still needed? Otherwise I'm going to close the issue in one week.

stm-clemens, 2020-01-07:
Hey sorry.

I was on vacation.

Here are the logs from the Wrapper and the script directly

```
ubuntu@ubuntu:~$ sudo /usr/local/etc/raspiBackupWrapper.sh
--- /mnt/backup/ already mounted
01-07-2020 11:50:13 --- RBK0009I: ubuntu: raspiBackup.sh V0.6.4.3 (2d927a2) started at Di 7. Jan 11:50:13 CET 2020.
01-07-2020 11:50:14 --- RBK0128I: Using logfile /mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/raspiBackup.log.
01-07-2020 11:50:14 --- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
01-07-2020 11:50:14 --- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
01-07-2020 11:50:20 --- RBK0151I: Using backuppath /mnt/backup.
01-07-2020 11:50:20 !!! RBK0157W: No services to stop.
01-07-2020 11:50:20 --- RBK0081I: Creating backup of type dd in /mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012.
01-07-2020 11:50:20 --- RBK0085I: Backup of type dd started. Please be patient.
187746+0 records in
187746+0 records out
196865949696 bytes (197 GB, 183 GiB) copied, 4332,77 s, 45,4 MB/s
01-07-2020 13:02:33 --- RBK0078I: Backup time: 01:12:13.
01-07-2020 13:02:34 --- RBK0159I: 5 backups kept for dd backup type.
01-07-2020 13:02:34 !!! RBK0156W: No services to start.
01-07-2020 13:02:34 --- RBK0033I: Please wait until cleanup has finished.
01-07-2020 13:02:34 --- RBK0049I: Messages saved in /mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/raspiBackup.msg.
01-07-2020 13:02:35 --- RBK0026I: Debug logfile saved in /mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/raspiBackup.log.
01-07-2020 13:02:35 --- RBK0017I: Backup finished successfully.
01-07-2020 13:02:35 --- RBK0167I: Sending email.
Reading message body from STDIN because the '-m' option was not used.
If you are manually typing in a message:
  - First line must be received within 60 seconds.
  - End manual input with a CTRL-D on its own line.

Jan 07 13:02:35 localhost sendEmail[17766]: Message input complete.
Jan 07 13:02:36 localhost sendEmail[17766]: Email was sent successfully!
01-07-2020 13:02:36 --- RBK0010I: ubuntu: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Di 7. Jan 13:02:36 CET 2020.
Backup succeeded :-)
+ version=v0.1.2
++ pwd
+ CURRENT_DIR=/home/ubuntu
+ SCRIPTNAME=pishrink.sh
+ LOGFILE=/home/ubuntu/pishrink.log
+ should_skip_autoexpand=false
+ debug=false
+ repair=false
+ gzip_compress=false
+ getopts :sdrzh opt
+ case "${opt}" in
+ gzip_compress=true
+ getopts :sdrzh opt
+ shift 1
+ '[' false = true ']'
+ echo 'pishrink.sh v0.1.2'
pishrink.sh v0.1.2
+ src=/mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/ubuntu-dd-backup-20200107-115012.img
+ img=/mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/ubuntu-dd-backup-20200107-115012.img
+ [[ -z /mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/ubuntu-dd-backup-20200107-115012.img ]]
+ [[ ! -f /mnt/backup/ubuntu/ubuntu-dd-backup-20200107-115012/ubuntu-dd-backup-20200107-115012.img ]]
+ ((  EUID != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v parted
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v losetup
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v tune2fs
+ ((  0 != 0  ))
+ for command in parted losetup tune2fs md5sum e2fsck resize2fs
+ command -v md5sum
[... 520 lines cut ...]
++ tr -d B
+ endresult=20870803968
+ logVariables 309 endresult
+ '[' false = true ']'
+ truncate -s 20870803968 ubuntu-dd-backup-20200107-115012.img
+ [[ true == true ]]
+ info 'Gzipping the shrunk image'
+ echo 'pishrink.sh: Gzipping the shrunk image...'
pishrink.sh: Gzipping the shrunk image...
++ gzip -f9 ubuntu-dd-backup-20200107-115012.img
+ [[ ! -n '' ]]
+ img=ubuntu-dd-backup-20200107-115012.img.gz
++ ls -lh ubuntu-dd-backup-20200107-115012.img.gz
++ cut -d ' ' -f 5
+ aftersize=6,3G
+ logVariables 324 aftersize
+ '[' false = true ']'
+ info 'Shrunk ubuntu-dd-backup-20200107-115012.img.gz from 184G to 6,3G'
+ echo 'pishrink.sh: Shrunk ubuntu-dd-backup-20200107-115012.img.gz from 184G to 6,3G...'
pishrink.sh: Shrunk ubuntu-dd-backup-20200107-115012.img.gz from 184G to 6,3G...
+ cleanup
+ losetup /dev/loop4
+ losetup -d /dev/loop4
+ '[' false = true ']'
```

framps, 2020-01-15:
Vielen Dank fuer die Logs. Aus irgendeinem Grunde funktioniert der mount im wrapper nicht. Leider prueft pishrink da nicht den RC und bricht sofort ab :disappointed: 

Ist dieser Fehler reproduzierbar - d.h. tritt er auch bei anderen Backups auf?

framps, 2020-01-20:
> sollen wir das Gespäch auf deutsch führen oder englisch halten?

Lieber in English damit alle es lesen koennen. Aber auch gerne in Deutsch wenn es Dir lieber ist :wink: 

Dann muss ich mal versuchen das zu reproduzieren. Wird aber sicherlich etwas dauern ...

stm-clemens, 2020-01-30:
English should be fine.

Anything new with "our" problem?

Hofei90, 2020-02-01:
So mir fällts in Deutsch leichter ;)

Hast du zufällig die AVM Laborversion für die Fritzbox installiert? (sofern du eine Fritzbox hast)
In der Laborversion (=beta Version ihrer Firmware) wurde ja smb2/3 hinzugefügt, ich hatte jedoch erhebliche Probleme mit der Laborversion.
Ein Problem dabei war, dass pishrink über das Netzwerk auf der fritz.nas nicht mehr funktionierte. Also wieder stable Version eingespielt und siehe da, pishrink funktionierte wieder tadellos.
Die anderen Probleme sind an dieser Stelle unwichtig.

stm-clemens, 2020-02-03:
Hey,
ne ich habe keine Fritzbox hier. Wir haben die Digitalisierungsbox von der Telekom. Diese greift aber garnicht in die Kommunikation mit ein. Das Backup läuft vom Raspberry Pi 4 4GB Ubuntu 64bit auf einen Qnap NAS. Der Nas ist hier die Konstante. An ihm wurde nichts geändert seitdem ich von Raspbian Buster auf Ubuntu gewechselt bin. Das script habe ich auch 1zu1 rüberkopiert und auch Pishrink wurde mit der selben Anleitung installiert. Also entweder hängt es an ubuntu oder an 64bit.

framps, 2020-02-06:
> Anything new with "our" problem?

Isnt't it our issue?

Right now my priority is to get the next raspiBackup release 0.6.5 out of the door and to publish 0.6.5 beta to get early feedback :wink:

framps, 2020-02-14:
I just executed the wrapperscript successfully on the latest buster image and the image was shrinked without any error message :smiley: 

> Im on ubuntu 19.10 64bit raspi4

I don't have a Raspi4 and now will try to recreate the issue on my Raspi3 with ubuntu. Please pass me the download link you grabbed the image from and a link with installation instructions. :wink:

stm-clemens, 2020-02-17:
> Right now my priority is to get the next raspiBackup release 0.6.5 out of the door and to publish 0.6.5 beta to get early feedback 😉

@framps no problem take your time :)

I used https://ubuntu.com/download/raspberry-pi 64bit 19.10 image for my raspi4 no special instructions. Just etch it and go :)

framps, 2020-02-22:
Thank you very much for the links. 

Unfortunately I just detected I don't have any working 8GB SD card any more for the image and neither SD card to test the restore :cry:

framps, 2020-02-23:
I was able to reproduce your issue. It seems to be related somehow to the fact the Ubuntu image uses loop devices.

stm-clemens, 2020-02-24:
Yay? :,D

any idea how to fix it?

framps, 2020-02-24:
Let's put it bluntly. I have no idea as of now :cry:  

raspiBackup is targeted for Raspbian and you use Ubuntu. raspiBackup works for a lot of people using Ubuntu even it's not supported. 

raspiBackupWrapper works with a Raspbian image. So I can just reject your issue with `unsupported environment`. I don't do this because I'm interested to know the root casue which is somehow releated to the loop device usage in Ubuntu.

I just spent the whole evening on this issue - but no success :cry: . I'll continue to investigate on this issue - but with a lower priority :wink: . Will let you know when I succeeded or gave up.

stm-clemens, 2020-02-25:
Well thats not that good of a message…

But its an error with pishrink not raspibackup :P

framps, 2020-02-25:
> But its an error with pishrink not raspibackup :P

Well, may be but it also may be an issue with the Ubuntu image which is now my preference. During my tests I noticed multiple times the dd command to create the backup causes the kernel to write `BUG` messages which said there is an unresponsive core on my Raspi3 and I had to restart the Raspi in order to get it working again...

> But whats the difference between calling pishrink with the script and running it directly?

I noticed even if you create the image with raspiBackup directly (not using the wrapper) which just uses dd to create the image and then invoke pishrink in the command line against the image you get the error message from pishrink. If you execute it a second time pishrink finishes successfully.

Then I thought a good workaround would be to call pishrink twice in the wrapper script but unfortunately pishrink exits in line 59 when an error occurs and terminates the wrapper script.

So there is one ugly workaround I see for you:

1) Edit pishrink in line [59](https://github.com/Drewsif/PiShrink/blob/master/pishrink.sh#L59) and remove the exit call
2) Call pishrink a second time in line [148](https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupWrapper.sh#L148) in raspiBackupWrapper

framps, 2020-03-01:
@stm-clemens Any news on this issue from your side? Otherwise please close this issue.

framps, 2020-03-07:
@stm-clemens  I'm going to close this issue the next days if there is no reply any more from your side.

closed 2020-03-08 by framps