framps/raspiBackup #190: Option to skip boot partition backup

Talkabout, 2019-12-28:
Hi,

I am using this tools for some time on several Raspberries and it is really great. Recently I had to switch one of them to a different board (Rock64) to improve VPN performance (hardware acceleration for crypto) and unfortunately my backup process is not working. I am using rsync for incremental backups which includes the boot partition, which is not that important for me. With the Rock64 and Armbian the backup completely fails because the boot process is completely different from Raspbian. I don't expect it to work with armbian boot structure but would like to have the option to skip the boot partition backup. That way I can still use the script for my environment and do not need to switch to something different.
Thanks!

Bye

framps, 2019-12-28:
Hello @Talkabout ,

I see your point. But please keep in mind raspiBackup was designed to create fully functional backups for Raspberries. I spend a lot of time to test raspiBackup and right now rewrite the regression testpackage on qemu for the next release. In addition every time I publish a new release I execute some manual tests on my local Raspberries. Said that you can see I want to make sure raspiBackup is a reliable backup tool.
I cannot test raspiBackup with other SOCs than Raspberries because of missing HW . In addition support for various other SOCs will create a huge test effort for me I don't want to spend. Keep in mind raspiBackup is for free.

If you don't need a fully functional backup `rsnapshot` will be the right tool for you :wink:

Talkabout, 2019-12-28:
Hi @framps ,

I think there is a missunderstanding. My request is not about testing with other boards or make "non-fully-functional" backups. The only thing is to provide an option to skip the boot partition backup, everything else should work as it is. My complete environment is build upon the structure of how raspibackup is creating folders and the structures therein, so using a different backup tool is difficult because there might be no tools providing the same features (including sending mails).
If such an option is not possible could you give me some hints which parts in the code I need to change to prevent the boot partition backup? I am also fine in having a modified version of raspibackup.

Thanks!

framps, 2019-12-29:
> The only thing is to provide an option to skip the boot partition backup, everything else should work as it is.

What about restore? Do you want to restore everything manually without support from raspiBackup? Otherwise the restore logic has to be tweaked also. This  creates another issue because raspiBackup saves the partition layout with `sfdisk /dev/mmcblk0` during backup in order to recreate the same partitioning during restore. As you can see it's much more to disable/tweak in order to get raspiBackup to meet your requirements. Think of the PARTUUID update, partition resizing and so on. There exists a lot of code dealing with the boot partition either directly or indirectly which means there are a lot of places to tweak in raspiBackup.

Talkabout, 2019-12-29:
Hi @framps ,

ok, got it. Well for me the restore is nothing more but copying the backup folder to the right location, as all my Boards are booting from a NAS system wia nfsroot. That is why raspibackup restore functionality is not something I need...
Anyways thanks for your explanations even if my problem cannot be solved!

Bye

framps, 2019-12-29:
> Anyways thanks for your explanations even if my problem cannot be solved!

It's possible - in particular because you don't need the restore functionality. Everybody having  some bash knowledge should be able to strip boot partition handling code. But I'm no longer willing to spend any time in feature requests which do not give any new function for the majority of raspiBackup users. I was disappointed too many times by users requesting features which actually only solve their special request.

I now try another approach to help you to get the feature you request:
1. Create a fork of raspiBackup
2. Modify the code and remove boot handling code
3. Try hard to get it done on your own
4. If you are stuck create an issue in your fork and mention me. I then will have a look into the code and give you hints as long as my effort doesn't exceed approx 15 mins per issue.

Hope you understand my attitude.

framps, 2019-12-30:
@Talkabout Please close this issue if you don't accept my offer.

Talkabout, 2019-12-30:
I am not sure if I will attempt to change the code myself, but I think the issue itself can be closed.

closed 2019-12-30 by Talkabout