framps/raspiBackup #191: Problem unkown filesystem type 'cifs'

sponkpi, 2020-01-05:
Hallo,
ich bin mal so frei und schreibe hier in deutsch, das geht einfach schneller....
Ich habe mir heute für mein Multihostsystem iobroker, das auf drei verschiedenen Raspi läuft, das raspiBackup installiert. Ein Riesenlob dafür, da wird in Zukunft wohl einiges an Zeit gespart werden können.
Aber zu meiner Frage:
Ich habe auf einem Odroid HC1 ein fertiges Nextcloudimage von ownyourbits am laufen.
Nachdem auf den Raspi´s alles so fein geklappt hat, wollte ich natürlich auch den Odroid damit ausrüsten. Das funktioniert so weit, aber beim ersten Aufruf kommt die Fehlermeldung:

> /mount: /backup: unknown filesystem type 'cifs'.

Mache ich etwas falsch?
Geht es beim Odroid überhaupt?
Liegt es am fertigen Image?
Was kann ich tun?
Vielen Dank für deine Hilfe und schöne Grüße aus dem Sauerland.

christianTF, 2020-01-05:
Probiere das Mounten statt mit cifs mit smbfs. Ist der neuere Treiber.

framps, 2020-01-05:
> ich bin mal so frei und schreibe hier in deutsch, das geht einfach schneller....

Kein Problem. 

Zu Deiner Frage: Das sieht mir sehr danach aus dass der Eintrag in der /etc/fstab fuer /backup nicht ganz OK ist. Der Vorschlag von @christianTF kann eine moegliche Ursache sein. Ansonsten findet man im Netz wenn man nach `"unknown filesystem type 'cifs'"` sucht verschiedene andere Ursachen dafuer. Falls Du das Debuglog zur Verfuegung stellst kann ich auch da noch mal reinsehen.

sponkpi, 2020-01-06:
Hallo und vielen Dank für die Hilfe.
Als Erstes habe ich in der Zeile 
`sudo mount -t cifs -o user=xxx,pass=yyy //10.10.10.83/backup //backup` 
das "cifs" durch "smbfs" ersetzt. Das hat leider keinen Erfolg gebracht.
Mit exakt der gleichen Zeile hat es aber auf dem Raspi mit iobroker super funktioniert.
Danach fand ich einen Hinweis, noch einmal die cifs-utils zu installieren. Das habe ich mit 
`sudo apt-get install cifs-utils`
gemacht.
Danach noch einmal:
`sudo mount -t cifs -o user.......`
Und siehe da - es läuft. Mal abwarten wie es fertig aussieht...

framps, 2020-01-10:
> Und siehe da - es läuft. Mal abwarten wie es fertig aussieht...

@sponkpi Wie ist das Ergebnis? Wenn alles OK ist bitte den Issue closen :wink:

sponkpi, 2020-01-10:
Hallo,
ja - sorry, ich gebe Bericht, wenn ich nächste Woche wieder zu Hause bin.
Das Sichern hat aber tatsächlich schon mal geklappt.
Ich habe bisher nur noch nicht versucht, das Image auf eine SD Karte zu schreiben und zu überprüfen, ob die Sicherung sich wieder zurück spielen lässt.
Gebe Bescheid, wenn ich das zum Abschluss ausprobiert habe.

Schönes Wochenende an alle

framps, 2020-01-10:
Alles klar. Danke. Genaugenommen ist aber Dein Issue resolved. Wenn jetzt beim Restore Probleme auftreten sollten ware das ein neuer Issue :wink: 

Aber mach einfach wie es Dir passt. Ich moechte nur meine geloesten Issues nicht mehr als offen sehen. Ordnung muss sein :smile:

sponkpi, 2020-01-15:
Hallo, ich wollte nur kurz den Erfolg melden.
Alle Sicherungen ließen sich auf einen andere SD Karte schreiben und funktionierten anschließend.
Vielen Dank an den Autor

closed 2020-01-15 by sponkpi

framps, 2020-01-15:
@sponkpi Vielen Dank fuer die Verifikation :+1: