framps/raspiBackup #192: tar: /var/lib/samba/private/msg.sock/440: socket ignored

dcaccount, 2020-01-11:
Hello,
at the end of a tar backup (successfully finished), before the restarting of the services, I get following messages:

> tar: /var/lib/samba/private/msg.sock/440: socket ignored
> tar: /var/lib/samba/private/msg.sock/7815: socket ignored
> tar: /var/lib/samba/private/msg.sock/582: socket ignored
> tar: /var/lib/samba/private/msg.sock/1936: socket ignored
.....

What does it mean? Is it a problem? Please advise.

Thanks,
Dan

framps, 2020-01-11:
Hello Dan,

Did you stop samba before creating a backup with option -a and -o?

dcaccount, 2020-01-11:
> Hello Dan,
> 
> Did you stop samba before creating a backup with option -a and -o?

Yes, using 

> raspiBackupInstallerUI.sh

framps, 2020-01-11:
Frankly I expect this kind of messages not to appear. Are you sure you stopped samba? Please provide the debug log.

dcaccount, 2020-01-11:
I have chosen as directory where to backup an external USB disk, could it be the reason?

Please find here below the log, thanks for helping.

[raspiBackup copia.txt](https://github.com/framps/raspiBackup/files/4049881/raspiBackup.copia.txt)

framps, 2020-01-11:
Thank you very much for the log. There I find
```
-> stopServices 
--- RBK0008I: Stopping services: 'systemctl stop smbd && systemctl stop ssh && systemctl stop cron && systemctl stop lighttpd && systemctl stop bccollect && systemctl stop console-setup && systemctl stop dbus && systemctl stop dhcpcd && systemctl stop keyboard-setup && systemctl stop networking && systemctl stop nmbd && systemctl stop openvpn && systemctl stop php7.3-fpm && systemctl stop raspi-config && systemctl stop rc-local && systemctl stop rng-tools && systemctl stop rpi-eeprom-update && systemctl stop rsyslog && systemctl stop triggerhappy && systemctl stop unattended-upgrades && systemctl stop wpa_supplicant && systemctl stop dphys-swapfile && systemctl stop fake-hwclock && systemctl stop gldriver-test && systemctl stop ifupdown-pre && systemctl stop kmod-static-nodes && systemctl stop netfilter-persistent && systemctl stop bluealsa && systemctl stop bluetooth && systemctl stop alsa-restore && systemctl stop alsa-state'.
```
which is a huge list and I frankly don't think all of them have to be stopped ... but anyhow I see samba is stopped.
I've never seen this kind of samba error messages before :cry: I searched in the net for this message and it's created when you backup a system running samba. Looks like samba wasn't stopped the time the backup ran. raspiBackup stops the services and it seems samba is stopped asynchronously and at the time the backup runs is still active.

But nevertheless it's an issue with the backup of samba sockets which can be ignored. Whenever you startup the backup new sockets will be created by samba. 

I think you can just ignore the error messages. Just test the restored image thoroughly and I think you will be fine.

If you want to isolate the root cause of the error message I can create a version of raspiBackup with additional debug statements. Just let me know.

dcaccount, 2020-01-11:
Thank you so much for the reply.
If I reduced the number of services to be stopped, would it change something?
Could you please list those that must be stopped? I found a list in your website, shall I stick with only those?

dcaccount, 2020-01-12:
I have discovered that if I choose a backup type rsync instead of tar the error disappears.

Any idea?
Thanks,
Dan

framps, 2020-01-12:
> Could you please list those that must be stopped?

Just stop all services which use some kind of database or cache something in memory. For example `systemctl stop bluealsa && systemctl stop bluetooth && systemctl stop alsa-restore && systemctl stop alsa-state` I think don't have to be stopped. 

But reducing the number of services will not stop the error message to pop up. I suggest to stop all services in your command list and check whether smb is still active. Maybe it's possible to use `lsof` to find out who uses the socket. I suggest to ask for help in a Linux or Raspberry forum.

> I have discovered that if I choose a backup type rsync instead of tar the error disappears.

I have no idea. But anyhow I suggest to use rsync instead of tar to profit from the hardlink support in an ext4 filesystem.

framps, 2020-01-26:
@dcaccount Any news on this issue? Otherwise please close it :wink:

closed 2020-01-27 by dcaccount