framps/raspiBackup #193: How can I exclude a directory from backup?

dcaccount, 2020-01-12:
Hello,
I have tried to esclude one directory from backup.

In the config file I have put:
`# exclude list
DEFAULT_EXCLUDE_LIST="--exclude=/home/daniele/TBD/*"`

but it still backups up it.
Where am I wrong? Pease advise.
Thanks, Dan

framps, 2020-01-12:
Could you please show the rsync or tar command invoked by raspiBackup? You'll find it in the debug log.

dcaccount, 2020-01-12:
Thanks for helping!

> executeCmd Command: rsync --exclude="/mnt/WD/Download/ToBackup@QNAP/Raspberry"                     --exclude="/mnt/WD/Download/ToBackup@QNAP/Raspberry/rpi4/rpi4-rsync-backup-20200112-094617/raspiBackup.log"             
        --exclude="/mnt/WD/Download/ToBackup@QNAP/Raspberry/rpi4/rpi4-rsync-backup-20200112-094617/raspiBackup.msg"                     --exclude='.gvfs'                       --exclude=/proc/*                       --exclude=/lost+found/*                         --exclude=/sys/*                        --exclude=/dev/*                        --exclude=/boot/*                       --exclude=/tmp/*                        --exclude=/run/*                        --exclude=/home/daniele/TBD/*                                           --numeric-ids                   -aHAx                                                                   /                       "/mnt/WD/Download/ToBackup@QNAP/Raspberry/rpi4/rpi4-rsync-backup-20200112-094617"

framps, 2020-01-13:
Looks good to me. Actually the statement has to follow the syntax and semantic described in the man pages of rsync. It's not a raspiBackup issue. Therefor please close this  issue.

In order to test different exclude statements I suggest to proceed as follows:
1. Check the rsync man pages or visit a Linux or Raspberry forum and ask for help.
2. When you have an idea what's the root cause execute the rsync command raspiBackup will execute (see above) and add option `-v` (verbose) and `-n` (dryrun). That way you easily can check whether the directory is excluded. Maybe pipe the result in grep :wink:

closed 2020-01-13 by dcaccount