framps/raspiBackup #195: Smart Backup Strategie: Reihenfolge Service wieder starten und alte Backups löschen tauschen?

betonmoewe, 2020-01-13:
If you have a question about raspiBackup please check whether following resources answer your question first before opening an issue :

* [ ] Did you read the raspiBackup FAQ's first which answer a lot of common questions ? [English](https://www.linux-tips-and-tricks.de/en/faq) - [German](https://www.linux-tips-and-tricks.de/de/faq)

betonmoewe, 2020-01-13:
Hi,

1.) Mail geht wieder in der aktuellsten Beta ;)

2.) Ich möchte die Smart Backup Stategie nutzen (schön das dies jetzt ohne wrapper geht). Ich habe nun folgendes Anliegen:
Ich habe auf einem meiner rasperries all meine Services als docker Container implementiert (z.B. fhem) und möchte 1 x die Woche ein Backup ziehen, welches auch Docker stoppt und nach dem Backup wieder startet (unter der Woche stoppe ich Docker nicht). Dies funktioniert so weit, aber da zunächst nach erfolgreichem Backup alle alten Backups gelöscht werden (falls diese in das Löschmuster fallen) dauert es unnötig lange, bis die Services wieder verfügbar sind. Ich denke, es wäre besser diese beiden Schritte zu tauschen: zunächst nach erfolgten erfolgreichem (oder erfolglosem) Backup die Services wieder schnellstmöglichst starten und erst dann alle weiteren Schritte (wie auch das Aufräumen der alten Backups) durchführen.

Gruß
Die Betonmöwe

framps, 2020-01-13:
Moin @betonmoewe ,

Du hast absolut Recht :+1: . Die Zeit in der die Services gestoppt sind sollte minimal sein. Ich habe das wie Du schon vorgeschlagen hast eben geaendert (https://github.com/framps/raspiBackup/commit/258124c850da553aba9c4c500c833bfc192d9abc)

framps, 2020-01-13:
@betonmoewe FYI #196  Not sure whether this is an issue for you.

framps, 2020-01-15:
@betonmoewe Kannst Du bitte noch den Fix testen und dann den Issue closen?

framps, 2020-01-26:
@betonmoewe Bitte verifizieren :wink:

betonmoewe, 2020-01-27:
sorry, war die letzten Tage unterwegs

sieht jetzt gut aus

closed 2020-01-27 by betonmoewe

framps, 2020-01-27:
Kein Problem. Vielen Dank