framps/raspiBackup #196: Enable smart recycle strategy to consider backup type

framps, 2020-01-13:
As of now smart recycle considers all backup for a system and doesn't consider the backup type. If there are different types mixed (e.g. ddz and rsync) with different backup strategies

*  rsync hourly, called with strategy "48 2 1 0" (keep 48 daily , 2 weekly, 1 monthly backups)
 * ddz weekly, called with strategy "0 1 2 0" (keep 0 daily, 1 weekly, 2 monthly backups)

right now the backup strategy of rsync will include the ddz backups and vice versa.

christianTF, 2020-01-13:
BTW: If you think this is not a feature needed by the public, you don't need to implement, as I could do so also in my post-processing.

framps, 2020-01-13:
Don't worry about this issue. As you can see I reclassified this issue as a bug. The other backup strategy offered by raspiBackup - define the number of backups to keep - works on backup type. Thus the smart recycle backup strategy has to have to work on backup type also :wink: 
I don't think it's a big issue to add the backup type to be considered also in current beta. I have just to modify the regex used to parse the backup directory names to include the backup type :smile:

framps, 2020-01-15:
Fixed in https://github.com/framps/raspiBackup/commit/2299ee08fbf12f96ff453b91ce474a28a8a149cc

closed 2020-01-15 by framps