framps/raspiBackup #197: Whitespaces in backup destinations leads to error

christianTF, 2020-01-13:
A whitespace in the backup destination folder leads to an error of the backup:

I tried these parameters:

Normal call:
`sudo /usr/local/bin/raspiBackup.sh -N "fstab" -o ":" -a ":" -k 3 -t ddz -L current -l debug -m detailed -v /opt/loxberry/system/storage/usb/rootfs/Backup/Mein LoxBerry`

Result:
```Script started on 2020-01-13 14:31:39+01:00 []
??? RBK0149E: /opt/loxberry/system/storage/usb/rootfs/Backup/Mein not found.
Script done on 2020-01-13 14:31:39+01:00 [COMMAND_EXIT_CODE="108"]
 ERROR: Backup failed with error code 108
```

Call with quotes:
`sudo /usr/local/bin/raspiBackup.sh -N "fstab" -o ":" -a ":" -k 3 -t ddz -L current -l debug -m detailed -v "/opt/loxberry/system/storage/usb/rootfs/Backup/Mein LoxBerry"`

Result:
```Script started on 2020-01-13 14:33:03+01:00 []
??? RBK0149E: /opt/loxberry/system/storage/usb/rootfs/Backup/Mein not found.
Script done on 2020-01-13 14:33:04+01:00 [COMMAND_EXIT_CODE="108"]
 ERROR: Backup failed with error code 108
```

Call with Perl quotemeta:
` sudo /usr/local/bin/raspiBackup.sh -N "fstab" -o ":" -a ":" -k 3 -t ddz -L current -l debug -m detailed -v \/opt\/loxberry\/system\/storage\/usb\/rootfs\/Backup\/Mein\ LoxBerry`

Result:
```
Script started on 2020-01-13 14:33:41+01:00 []
??? RBK0149E: /opt/loxberry/system/storage/usb/rootfs/Backup/Mein not found.
Script done on 2020-01-13 14:33:42+01:00 [COMMAND_EXIT_CODE="108"]
 ERROR: Backup failed with error code 108
```

The backup destination also should support quotes in the command line.

Cross-Reference to LoxBerry Backup issue https://github.com/christianTF/LoxBerry-Plugin-LoxberryBackup/issues/24

framps, 2020-01-13:
You're rubbing salt into my wounds ... I already got an issue about white spaces in filenames a long time ago and it took me multiple days to get this fixed. I already expected there are quotes missing again to protect spaces in filenames :cry: 

But frankly - who uses spaces in filenames ? (just kidding :laughing: )

christianTF, 2020-01-13:
To show solidarity, I also fixed a whitespaces bug in paths in LoxBerry-Core, found during analyzing this issue 😉

framps, 2020-01-13:
> To show solidarity,

Thank you very much. It's a very ugly issue - in particular in bash :disappointed: 

I just checked the latest code of 0.6.4.3 (SHA 2d927a2) when RBK0149E: is written but I don't find any missing quotes. Which version (SHA) do you use?

christianTF, 2020-01-14:
`loxberry@loxberrypi4:~ $ raspiBackup.sh --version
Version: 0.6.4.3 CommitSHA: 2d927a2 CommitDate: 2019-06-17 CommitTime: 20:10:20‘`

framps, 2020-01-14:
Thank you very much for the SHA. But unfortunately I still cannot find the  root cause for the issue. In order to reproduce the issue I should know what the file which creates the issue is used for. What's the complete filename of `/opt/loxberry/system/storage/usb/rootfs/Backup/Mein` ?

christianTF, 2020-01-16:
Simply mount an external storage to `/Backup\ Directory`

I've now tried with an smb mount to symlink to `/Backup Directory`:
```bash
ln -s /opt/loxberry/system/storage/smb/homeserver/Raspberry-Backups Backup\ Directory
root@loxberry-dev:/# ls -l
lrwxrwxrwx   1 root     root        61 Jan 16 20:59 'Backup Directory' -> /opt/loxberry/system/storage/smb/homeserver/Raspberry-Backups
```
Result of calls:
```
root@loxberry-dev:/# /usr/local/bin/raspiBackup.sh -v /Backup\ Directory
??? RBK0149E: /Backup not found.
root@loxberry-dev:/# /usr/local/bin/raspiBackup.sh -v "/Backup\ Directory"
??? RBK0149E: /Backup\ not found.
root@loxberry-dev:/# /usr/local/bin/raspiBackup.sh -v "/Backup Directory"
??? RBK0149E: /Backup not found.
```
raspiBackup does neither accept escaping nor quoting. It see's only "/Backup".

framps, 2020-01-26:
This issue will not be fixed in beta 0.6.4.4. Thus I just added code in the upcoming 0.6.4.4 beta which refuses any filenames or directories which contain any spaces with an error message.

closed 2020-03-03 by framps

framps, 2021-10-03:
Check for whitespaces already added