framps/raspiBackup #198: exclude for logging raspibackup files in /var/log has double leading slash

montgomeryb, 2020-01-29:
I had several issues running this on one machine. In troubleshooting I set the log directory location to /var/log - it took multiple failed runs to see that the script is adding double slashes to the exclude directory, so tar was failing as the file was changing as it was being archived.

` tar -cpiv --one-file-system --ignore-failed-read -z -f /backups/raspiBackup/mcmpi/mcmpi-tgz-backup-20200129-140814/mcmpi-tgz-backup-20200129-140814.tgz --warning=no-xdev --numeric-owner --exclude=/backups/raspiBackup/* --exclude=//var/log/raspiBackup/mcmpi.log --exclude=//var/log/raspiBackup/mcmpi.msg --exclude=.gvfs --exclude=/proc/* --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/tmp/* --exclude=/boot/* --exclude=/run/* --exclude=/var/log/raspiBackup/* /` 
Setting 
> DEFAULT_EXCLUDE_LIST=" --exclude=/var/log/raspiBackup/*" 
allowed the script to run without any errors

framps, 2020-01-29:
Thank you very much for debugging and reporting this issue 👍 .

framps, 2020-02-06:
I created a fix in https://github.com/framps/raspiBackup/tree/issue198. Please verify.

montgomeryb, 2020-02-06:
I checked out and tried this branch. I only copied the script to the bin
directory and tried it. I ran in to a problem though. It seems to have
removed the root source (/) and the path for the log and message files
(/var/log

--- RBK0085I: Backup of type tgz started. Please be patient.
20200206-174609 MSG 3209:              --- RBK0085I: Backup of type tgz
started.
 Please be patient.
20200206-174609 DBG 1188:                 -> executeCmd Command: tar
-cpi --one-file-system                          -z              -v -f
"/backups/raspiBackup/mcmpi/mcmpi-tgz-backup-20200206-174554/mcmpi-tgz-backup
-20200206-174554.tgz"           --warning=no-xdev
--numeric-owner
                --exclude="/backups/raspiBackup/*"
 --exclude="/mcmp i.log"          --exclude="/mcmpi.msg"
 --exclude='.gvfs' --exclude=/proc/*               --exclude=/lost+found/*
                --exclude=/sys/*                --exclude=/dev/*
     --exclude=/tmp/*                --exclude=/boot/*
--exclude=/run/*
20200206-174609 DBG 1189:                 -- Redirect: & - Skips:
tar: Cowardly refusing to create an empty archive
Try 'tar --help' or 'tar --usage' for more information.
20200206-174609 DBG 1208:                 <- executeCmd 2
20200206-174609 DBG 3222:              <- tarBackup 2

framps, 2020-02-07:
Ouch :worried: - my bad. I used a bash function to remove the leading / but missed it's actually deleting all stuff leading the last / :cry: . I changed the code to use sed with a regex and now it should work. Sorry for the inconvenience. But this proves again: Don't trust a fix until it was verfified :smiley:

montgomeryb, 2020-02-08:
> Ouch 😟 - my bad. I used a bash function to remove the leading / but
> missed it's actually deleting all stuff leading the last / 😢 . I changed

Not a problem - been there before. So, that said ....

Line 3175, source is set to "" (ie empty string).
This creates a problem on line 3205 as there is no source passed to tar,
causing it to fail.

Not sure the obvious resolution to this one ... almost a catch 22?
Bryan.

framps, 2020-02-08:
:cry: Sorry Bryan for the inconvenience. I thought it's a low hanging fruit :weary: .

It's not a catch 22 - it's my ignorance and I thought there is no test required from my side. It's fixed and tested now :wink: .

Actually option `-L 0` is not documented any more and it's kept for backward compatibility. All logs are saved in the backup location together with the backup all the time. If for some reasons the backup fails and the backup directory will be cleaned up the log is copied into the home directory of the caller for further analysis. Looks like you missed this because it's not very well documented.

montgomeryb, 2020-02-08:
Yep, that one worked successfully. No problem testing this patch.

Thanks - Bryan.

framps, 2020-02-08:
Thank you very much for verifying the fix again. Sorry again for the inconvenience. 

You didn't close the issue. Is there still something missing?

closed 2020-02-08 by montgomeryb