framps/raspiBackup #200: raspiBackupInstallUI.sh issue

skyerjoe, 2020-02-04:
* [ ] Which operating system do you use? Provide the contents of /etc/os-release
**Raspibian Debain Stretch 9** 
* [ ] Provide all messages written by raspiBackup
**its not an backup problem** 
* [ ] Attach the debug log to the issue which will be created by raspiBackup with option `-l debug`
**see above**
* [ ] Version:
**0.6.4.3**

**Update Components**

Update the raspiBackupinstallUI.sh doesnt work or at least it wont be showed in the UI that is upgraded. 

best regards

framps, 2020-02-06:
> Update the raspiBackupinstallUI.sh ( 0.4 beta) over the GUI doesnt work or at least it wont be showed in the GUI that is upgraded.

I just detected you use 0.4 beta. Would you please install the latest version 0.4.3 and try again?

skyerjoe, 2020-02-09:
Hello framp 

I'm not able to update the UI but the script is still on  0.4.3 

Here is an snippet from the menue: 

 **`P1 Aktualisiere raspiBackup.sh (no upgrade available)`**
 **P2 `Aktualisiere raspiBackupInstallUI.sh (0.4-beta -> 0.4.3)`**  ->  its still the same state after running 

best regards

skyerjoe, 2020-02-09:
Ok i have to download the new raspiBackupInstallUI.sh again then it works.

But im not able to update from 0.4-beta 

 Solved so far Thnx   

best regards

framps, 2020-02-09:
I was able to reproduce the update issue and fixed it. There is a new installer 0.4.3.1 available in a new [branch](https://github.com/framps/raspiBackup/tree/issue200/installation). Please download this new installer from github and copy it in /usr/local/bin. In order to verify the fix of the update mechanism you have to modify the installer in line 30 from
```
VERSION="0.4.3.1" 	# -beta, -hotfix or -dev suffixes possible
```
to
```
VERSION="0.4.2-beta" 	# -beta, -hotfix or -dev suffixes possible
```
(fake the version).

Now invoke the installer again and update the installer. Now the update to the current installer 0.4.3 (which has the update issue) should work. You actually reinstall the buggy installer again :sad:. But that way you verified the update fix works :wink: . 

If you verify this update procedure from 0.4.2-beta to 0.4.3 works please let me know. I then will publish installer 0.4.3.1.

skyerjoe, 2020-02-13:
Jep seems working  

Snippet after Update: 

**`P1 Aktualisiere raspiBackup.sh (Kein Update verfügbar)
 P2 Aktualisiere raspiBackupInstallUI.sh (Kein Update verfügbar`**

So do u need anything else ? otherwise ill delete the file and fetch it over github from the official branch.

regards

framps, 2020-02-13:
Thank you very much for verifying the fix :+1: 

> So do u need anything else ?

Please double check raspiBackup has the correct version after update with `raspiBackup.sh --version`

> otherwise ill delete the file and fetch it over github again from the official branch.

When you double checked the version is OK I will publish the fixed installer, update this issue and then I suggest to download the fixed installer from the website instead from github.

skyerjoe, 2020-02-13:
NP  ure welcome 
raspiBackup.sh --version: 
`**Version: 0.6.4.3 CommitSHA: 2d927a2 CommitDate: 2019-06-17 CommitTime: 20:10:20**`

regards

framps, 2020-02-16:
Just published

closed 2020-02-16 by framps