framps/raspiBackup #201: SmartRecycle deleting it's own folder

AnduriI, 2020-02-10:
Hey framps,

I like your new rotation strategy and tested it. After verifying all goes right with dryrun enabled I started a manuall backup. In the process, raspiBackup startet recycling it's own working directory...

'''
pi@Pimatic:~ $ sudo raspiBackup.sh --smartRecycle --smartRecycleOptions "0 8 18 3"
--- RBK0009I: Pimatic: raspiBackup.sh V0.6.5-beta (57e1a1c) Mo 10. Feb 13:46:36 CET 2020 gestartet.
!!! RBK0165W: =========> HINWEIS <========= 
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte. 
!!! RBK0165W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0151I: Backuppfad /mnt/Backup wird benutzt.
--- RBK0008I: Services werden gestoppt: 'sudo pimatic.js stop'.
Sending SIGTERM to pimatic(11994), waiting for process exit...
Stopped.
--- RBK0081I: Backup vom Typ rsync wird in /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634 erstellt.
--- RBK0036I: Partitionslayout wird gesichert.
--- RBK0044I: Backup der Bootpartition wird in /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/Pimatic-backup.img erstellt.
43+1 Datensätze ein
43+1 Datensätze aus
45190144 Bytes (45 MB, 43 MiB) kopiert, 3,28091 s, 13,8 MB/s
--- RBK0044I: Backup des Partitionlayouts wird in /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/Pimatic-backup.sfdisk erstellt.
--- RBK0046I: Backup des Masterbootrecords wird in /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/Pimatic-backup.mbr erstellt.
--- RBK0133I: Verzeichnis /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134007 wird für Hardlinks benutzt.
--- RBK0158I: rsync Backup "/mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 00:05:58.
--- RBK0007I: Services werden gestartet: 'sudo pimatic.js start'.
Started.
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:0 Wöchentlich:8 Monatlich:18 Jährlich:3
--- RBK0217I: 3 Backups werden smart recycled. 
Pimatic-rsync-backup-20191209-020001
Pimatic-rsync-backup-20191223-020002
Pimatic-rsync-backup-20191230-020002
Pimatic-rsync-backup-20200106-020001
Pimatic-rsync-backup-20200113-020002
Pimatic-rsync-backup-20200120-020001
Pimatic-rsync-backup-20200127-020001
Pimatic-rsync-backup-20200203-020001
Pimatic-rsync-backup-20200210-020001 Backups werden aufgehoben.
--- RBK0220I: Smart Backup Strategie löscht Backup /mnt/Backup/Pimatic/Pimatic-rsync-backup-20191216-020002.
--- RBK0220I: Smart Backup Strategie löscht Backup /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134007.
--- RBK0220I: Smart Backup Strategie löscht Backup /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
/usr/local/bin/raspiBackup.sh: Zeile 1166: /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg: Datei oder Verzeichnis nicht gefunden
--- RBK0049I: Meldungen wurden in /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg gesichert.
/usr/local/bin/raspiBackup.sh: Zeile 1166: /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg: Datei oder Verzeichnis nicht gefunden
--- RBK0017I: Backup erfolgreich beendet.
/usr/local/bin/raspiBackup.sh: Zeile 1166: /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg: Datei oder Verzeichnis nicht gefunden
/usr/local/bin/raspiBackup.sh: Zeile 2561: /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg: Datei oder Verzeichnis nicht gefunden
--- RBK0167I: Eine eMail wird versendet.
/usr/local/bin/raspiBackup.sh: Zeile 1166: /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg: Datei oder Verzeichnis nicht gefunden
--- RBK0010I: Pimatic: raspiBackup.sh V0.6.5-beta (57e1a1c) Mo 10. Feb 13:58:37 CET 2020 beendet mit Returncode 0.
/usr/local/bin/raspiBackup.sh: Zeile 1166: /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200210-134634/raspiBackup.msg: Datei oder Verzeichnis nicht gefunden
'''

framps, 2020-02-10:
Thank you very much for testing the beta. Would you please provide the debug log?

AnduriI, 2020-02-10:
how do I get the debuglog to save at /home/pi? '-l Debug' seems to save it in the later deleted dir...

AnduriI, 2020-02-10:
found the option...
[Pimatic.log](https://github.com/framps/raspiBackup/files/4181199/Pimatic.log)

AnduriI, 2020-02-10:
well... I also had a look in the new pimatic.msg and it has no sign of that problem... maybe the only thing that was trying to write in the folder were the messages to that .msg/.log file...

framps, 2020-02-10:
Thank you very much for the log. 

There is no folder deleted :wink: .  Root cause for the message is that you use log option 1 (/var/log/raspiBackup/Pimatic.log) but during final cleanup the default msg file location in the backup dir is used which doesn't exist. Frankly I discourage to use log option 1 and use the default (option 2). But nevertheless this will be fixed. I updated the issue title to reflect the root cause of the issue :wink: .

AnduriI, 2020-02-10:
Hi framps,

I think you missunderstood me. I had to change the log location to option 1 to get a log file at all. Before that it was set to 2 and there were these messages "file not found". So I think the smartrecycle removes the directory and it's logs before raspibackup moves its own logfiles to /home/pi/

framps, 2020-02-10:
Thank you very much for the update. This explains the different log outputs I noticed but didn't understand. 

Just to get an idea about the severity of this issue: As far as I understand the smartrecycle result is OK. But for some reasons messages are written access to the msg file is impossible. Correct?

AnduriI, 2020-02-10:
that's correct, your smartrecycle works fine and also says to delete the current folder (because there was already one backup from that day). This results in deleting it's own logfile while still trying to write output to said logfile - and the corresponding errors.
Maybe you can circumvent this by protecting the current backup from the smartrecycle system at all (and delete it if needed with the next run)

framps, 2020-02-10:
> your smartrecycle works fine and also says to delete the current folder (because there was already one backup from that day).

I think that's the root cause for the issue. Looks like I never tested with two backups created on the same day :cry: when the last backup will be deleted. Good catch :+1:

framps, 2020-02-11:
@AnduriI Your issue is an edge issue which only happens if you request no daily backups and weekly backups and create a second backup the same day in the week. You faced this issue because you did some testing which is fine but a normal user will setup cron and create one backup a week only.

I can create a hack to fix this issue but that's not the right way to go from an implementation point of view. Unfortunately a right fix will blow up the current beta :cry:  Thus I will create an development issue to fix this in next release .

Any objections from your side?

AnduriI, 2020-02-11:
thats totally fine for me. It is not urgent in any way, I just wanted to inform you about that problem to be solved in 1.0 ;-)

closed 2020-02-11 by AnduriI

framps, 2020-02-11:
> I just wanted to inform you about that problem to be solved in 1.0 ;-)

Thank you very much