framps/raspiBackup #202: Failure during restore with raspiBackupRestore2Image.sh

WAdama, 2020-02-12:
Beta-Version 6.5
Restore using raspiBackupRestore2Image.sh fails with following messages:
??? RBK0111E: Error occured when partitions were created. RC 32
sfdisk error.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed with RC 112. Check previous error messages.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/4193330/raspiBackup.log)

framps, 2020-02-12:
Thank you very much for the debug log. Unfortunately the error happens when there are not much debug statements :disappointed: . Do you use raspiBackupRestore2Image the first time or did you ran it with raspiBackup 0.6.4.3 already successfully ?

WAdama, 2020-02-12:
I used it with 0.6.4.3 sucessfully, but to be sure, I tested it today again. With 0.6.4.3 it worked without error.

framps, 2020-02-12:
Thank you very much you double checked it works with 0.6.4.3. raspiBackupRestore2Image unfortunately is not included in the regression test package and therefore I didn't catch this regression :cry: 

I will now to setup an environment to recreate and finally fix the regression issue. 

You prove it's a good idea for users of raspiBackup to test a beta and make sure there are no regressions :+1:

WAdama, 2020-02-12:
You'Re quite welcome. If there's something I can do to help just mention it.
The Raspi with the Beta is not my only one, so there's no problem to run the restore to image on another one, so unnecessary urgency... :-)

framps, 2020-02-12:
I appreciate your help. As usual I will ask you to verify the fix and close the issue :smiley: 

But 0.6.5 will not be published until this issue is fixed even it's a kind of edge issue.

WAdama, 2020-02-13:
Say the word, I will be ready... :)

framps, 2020-02-13:
Congratulations. :+1: You found a valid regression :+1:  

It's only one single line which I missed to add when I refactored some code :sob: 

I fixed this in the [beta branch](https://github.com/framps/raspiBackup/tree/beta). Please grab the latest raspiBackup version from there and verify the fix. Please also check the shrinked image can be started (That's what I didn't test).

I also added some code which masquerades some sensitive info in the debug log (I noticed you xxx ed some information in the log which is valid :wink: )

WAdama, 2020-02-14:
A first test worked, a image has been created. The image seems valid, you can access its content.
I haven't tested if it works until now. This has to wait for tomorrow.

framps, 2020-02-14:
Great. Thank you very much. :+1:  Take your time. I'm confident the the final test - to boot the shrinked image - will succeed. But the devil is a squirrel :hamster:

WAdama, 2020-02-15:
Hi, I'm afraid bad news...
The image won't start. You can see two or three attemps to read from the SD card but that's all.

framps, 2020-02-15:
😢 OK. I'll check what's wrong. As I said ... Devil and squirrel ...

framps, 2020-02-15:
That's what I did to verify this issue:

1) I setup a buster-lite on a brand new SD card, installed current raspiBackup version 0.6.4.3 and grabbed raspiBackuprestore2Image.sh from git master. 
2) Next I installed current pishrink from git. 
3) Then created a tar backup with raspiBackup. 
4) Later on I started raspiBackuprestore2Image.sh to create a shrinked image. 
5) Then dumped the shrinked image with dd on another brand new SD card. 
6) Finally plugged in this SD card into a Raspberry and booted the system successfully. The system  rebooted once when the root filesystem was resized but then everything was OK.

Did you use a plain new SD card for the restore of the pishrinked image?

WAdama, 2020-02-15:
Yes, with 0.6.4.3 it works quite well. Only with 0.6.5.0 the problem occurs.
I write the image with Win32DiskImager on the card. I wipe the SD card before.

framps, 2020-02-15:
Oops ... I missed wer're talking about the 0.6.5 beta :cry: . Will retry with the beta now

framps, 2020-02-21:
I found the root cause of the issue. Because of a change in the partitioning the mbr is no longer restore. Unfortunately ths partuuid is stored in the mbr and thus the partuuid used in /etc/fstab and /boot/cmdline.txt doesn't match any more which cause the boot to hang.

There now exists new beta code with a fix in raspiBackuprestore2Image.sh.

@WAdama Please verify the fix.

WAdama, 2020-02-21:
That's sounds good...
And again, tomorrow will the day... :-)

WAdama, 2020-02-22:
@framps I can second your result.
I made a new backup with the new versions and then created an image.
The image works perfect. It booted and could start all services.
So I think that is it...

framps, 2020-02-22:
Perfect :+1: Thank you very much for verifying. 

This actually was a showstopper for me to publish 0.6.5 ... but now is fixed :smile:

WAdama, 2020-02-22:
You're welcome...

I already thought so....

closed 2020-02-23 by framps