framps/raspiBackup #204: Colored Messages in the email notifications

uclara, 2020-02-13:
Hi,

I World like to suggest a small change.
It would be much easier to read the email notifications if e.g. error messages would be red colored and warning messages in orange (as yellow is nearly not readable). Info
Messages could be standard or colored in dark green

Ulli

framps, 2020-02-13:
Nice idea. :+1:

framps, 2020-02-22:
In the mean time somebody asked for [support of Telegram](https://github.com/framps/raspiBackup/issues/205). Wouldn't his be an alternative for you?

uclara, 2020-02-23:
Hi, the base of suggestion is the possibility to distinguish between information, warnings and errors. I think if the coloring of the messages will be implemented for the Mail notifications it can be used also with Telegram.

Ulli

framps, 2020-02-23:
I just created a feature prototype branch https://github.com/framps/raspiBackup/tree/issue204_coloring_feature which has coloring implemented for email and console. Just use he new option `--coloring` with parameter `M` for mail coloring and `C` for console coloring. As of now the default is `MC` but will be changed to `C` finally.

@uclara Please check out the new code, test and provide feedback :wink:

uclara, 2020-02-24:
Hi,
that was really fast :-)

What's the best way to update an existing raspiBackup instalation with the prototype version?

Ulli

framps, 2020-02-24:
> that was really fast :-)

I have some spare time right now :wink: 
 
> What's the best way to update an existing raspiBackup instalation with the prototype version?

I don't suggest to replace the current version. It's a prototype version and the code will make it into the next official version.

Just execute following commands in your home directory. This allows you to 
```
wget https://raw.githubusercontent.com/framps/raspiBackup/issue204_coloring_feature/raspiBackup.sh
chmod +x raspiBackup.sh
sudo ./raspiBackup.sh
```
You may user option `--coloring` (see above for parameters). But as of now the default is to send a colored mail. 

This way you can temporarily test the prototype without changing your current setup.

uclara, 2020-02-27:
Hi,

it seems working as expected.
I tested only the console output, as my mail notification stopped with the prototype procedure you suggested above. Shall I reconfigure the mail settings after downloading the prototype version?

Ulli

framps, 2020-02-27:
> my mail notification stopped with the prototype procedure you suggested above.

You just updated raspiBackup and no config and your eMail configuration is not touched. Thus eMail should work. 

Please test with current version 0.6.4.3 again. If then the eMail works there is an issue in the beta with the eMail. Please provide the debug log of the beta run where the eMail doesn't work. But please make sure you masquerade all sensitive information (email etc) first :wink: .

uclara, 2020-03-02:
Hi,

as I had a mess with different versions I decided tu update to the newest beta FlightAware-LAN: raspiBackup.sh V0.6.5-beta (e5fd976) and the coloring ist seems to work correctly in the console log and also in the mail.

Thank you!

PS: only difference that I noted is that warnings in the console are colored in yellow, and in the email in orange instead.

framps, 2020-03-03:
Thank you very much for your test and feedback :+1: 

> as I had a mess with different versions I decided tu update to the newest beta

I'm sorry there is no support to update to the latest beta version included in raspiBackup. I will create this feature in one of the next releases. It's too late - but there is one way to update the beta:

1. Revert to version 0.6.4.3 with option -V
2. Update to the latest beta version with option -U

> warnings in the console are colored in yellow, and in the email in orange instead.

On the console there is no way to create orange text (see [VT100 display attributes](http://www.termsys.demon.co.uk/vtansi.htm)). I can change the color in the eMail to yellow also but I think yellow has not much contrast to the white background so I decided to use orange instead (See [html color codes](https://htmlcolorcodes.com/)). 

If everything is fine now for you please close the issue.

uclara, 2020-03-03:
Thank you for the great support!

closed 2020-03-03 by uclara