framps/raspiBackup #205: Proposal for a new feature: Telegram notifications

davepee, 2020-02-15:
Add Telegram notifications as another option to email. See https://core.telegram.org/ and https://pypi.org/project/telegram-send/ please. Thx.

framps, 2020-02-16:
I did some initial investigations on this feature request and it seems not to be that difficult to add some code to write the messages to Telegram. Most of the work is to configure telegram. I found two ways to send the messages:

1. `telegram-send -f msg --config myTelegram.conf`
2. `telegram-send --config framp.conf --stdin < msg`

First one sends the message in a file whereas the second one sends all messages as text. Latter one has the drawback the messages are wrapped whereas the first one requires an external editor in order to open the message file.

@davepee What's your opinion as an experienced Telegram user :wink: ? Which one makes sense to be implemented in a raspiBackup prototype for you?

framps, 2020-02-16:
@Hofei90 As far as I know you're also a Telegram user. What's your opinion on this feature request?

Hofei90, 2020-02-16:
Die Idee find ich super! Bin selbst auch aktuell am überlegen meine Skripte die Funtion nachzurüstende, bei Fehlern oder bestimmten Ereignissen eine Nachricht über Telegram zu senden.
Ich werde zwar dafür meine eigene Telegram API verwenden, aber das ist wieder ein anderes Thema. Das gepostet Python Modul scheint einfach in der Verwendung zu sein, wenn auch ich es jetzt nicht selbst probiert habe.

——

I really like this idea!  I am also currently feeling my script to send the function to be retrofitted when perceiving or perceiving a message via telegram.
  I'm going to use my own my own Telegram API, but that's another topic.  That means the Python module becomes easy to use, even if I haven't tried it myself.

davepee, 2020-02-16:
@framps I think sufficient and easiest option is `telegram-send "some text"` provided the user has already configured via `telegram-send --configure` (or your suggestion with `--config myTelegram.conf`). The entire protocol could be viewed directly on the raspi. For me is better a simple announcement than the attached file ;)

framps, 2020-02-16:
> I think sufficient and easiest option is `telegram-send "some text"` provided the user has already configured via `telegram-send --configure` 

Frankly I like this approach by not passing a config file and just call telegram-send with the messages. But whenever I invoke this command a bot will be created and receives the messages. But what about users who want to get the message to their Telegram id? They have to have to use a config - at least that's what I understand.

> The entire protocol could be viewed directly on the raspi. For me is better a simple announcement than the attached file ;)

Sound's like you have another feature request: Don't receive all messages but just a short success, warning or failure message :wink:

davepee, 2020-02-16:
> But what about users who want to get the message to their Telegram id? 
> 
Telegram notifications are about BOTs. A similar approach uses (for example) Openhab2 (https://www.openhab.org/addons/bindings/telegram/).

> Sound's like you have another feature reques
>
I didn't realize this :) I'm not able to evaluate what what does it involve.

framps, 2020-02-16:
I just pushed a [new branch](https://github.com/framps/raspiBackup/tree/issue205_telegram_feature) which is a prototype for the telegram notification feature. 

New options to use to get telegram notifications about the backup result (just a success or failure message):
1. `--telegramToken <TOKEN>`
2. `--telegramChatID <CHATID>`

@davepee  Please provide feedback :wink: 

## Note
There is no masquerading done to hide the token and chatid in the prototype in the debug log :exclamation:  This will be added later on.

Hofei90, 2020-02-17:
Hallo,
also bisher erhalte ich nur die Meldung:

> raspberrypi: Backup nicht erfolgreich!!!.
Den abschließenden Punkt und 2 der 3 Ausrufezeichen können denke ich entfernt werden.

Außerdem würde ich es begrüßen, wenn man mit einer weiteren Option bestimmen kann, ob man nur den Status ob **erfoglreich** oder **nicht erfoglreich** gesendet bekommen will, oder auch zusätzlich die Ausgabe der Fehlermeldungen.

framps, 2020-02-17:
#### Previous comment from Hofei90 xlated:
I receive a failure message only `raspberrypi: Backup failed`

I think final dot and exclamation marks can be removed.

I addition I'd like to have an option to define whether a success and/or failure message should be sent or the messages should be appended.

framps, 2020-02-17:
> I receive a failure message only `raspberrypi: Backup failed`

Hm ... I also got a success message when the backup was OK. Let me check.

> I think final dot and exclamation marks can be removed.

Agree

> I addition I'd like to have an option to define whether a success and/or failure message should be sent or the messages should be appended.

What about 
`--telegramNotification="sfg"`
where `s` enables a success, `f` enables failure notification and `g` attaches the log?
Default will be `sf`.

Hofei90, 2020-02-17:
Stopp ich hab mich falsch ausgedrückt, ich konnte auch noch keine erfolgreiche Nachricht erhalten. Da bisher das Backup  immer fehlschlug weil ich noch keinen externen Datenträger eingebunden habe. Das liegt also nicht an dir. Ich wollte damit sagen, ich habe  nur Nachrichten geprüft wo das Backup fehl schlug.

Warum „g“ für log, da kann ich gerade keinen Zusammenhang herstellen? Wär es nicht besser mit „l“ = log oder „v“ = verbose

framps, 2020-02-17:
#### Previous comment from Hofei90 xlated:

Some clarification: I wasn't able to get a successful backup through because of missing external space. I just tested the failure message. Test of success message is still missing.

Why do you use "g" for log? Wouldn't "l" = log or "v" = verbose better?

framps, 2020-02-17:
> I just tested the failure message. Test of success message is still missing.

You made my day :+1: 

> Why do you use "g" for log? Wouldn't "l" = log or "v" = verbose better?

"l" is commonly confused with "1" - depending on the character font you use. That's why I choosed g from lo**g** .

framps, 2020-02-17:
There is a new prototype available :wink: 

New option `--telegramNotifications` accepts `SFM`. I decided to use M instead of L because it's the message file which is uploaded and not the debug log.

Hofei90, 2020-02-17:
Also die drei Ausrufezeichen erhalte ich nach wie vor bei einem Fehlschlag.

Die 3 Optionen funktionieren, jedoch habe ich noch einen Änderungswunsch:
Kannst du die angehängte Datei raspiBackup.msg als Text mit senden und nicht als Dateianhang.
Mit dem iPhone z.B lässt sich die Datei in Telegram nicht betrachten. Liegt wohl an der Dateiendung, dies aber nur eine Vermutung.

davepee, 2020-02-17:
I have a moment for testing. But I received no telegram message when backup is finished sucessfully (works in Fake mode?):

`./raspiBackup.sh -F --telegramToken '1xxxxxxx:AAxxxxx9-3vM3UjrrTxxxxxxxxxxRs' --telegramChatID '\-3xxxxxxx5'`

1xxxxxxx:AAxxxxx9-3vM3UjrrTxxxxxxxxxxRs
-3xxxxxxx5
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.4.3-dev (01f2caa) started at Po úno 17 20:51:56 CET 2020.
!!! RBK0192W: =========> NOTE  <========= 
!!! RBK0192W: This is a development version and should not be used in production. 
!!! RBK0192W: =========> NOTE <=========
--- RBK0128I: Using logfile /root/temp/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
!!! RBK0124W: **Fake mode on**.
--- RBK0151I: Using backuppath /backup.
--- RBK0008I: Stopping services: 'systemctl stop tvheadend && systemctl stop openhab2 && systemctl stop mosquitto && systemctl stop apache2 && systemctl stop mariadb'.
--- RBK0081I: Creating backup of type rsync in /backup/raspberrypi/raspberrypi-rsync-backup-20200217-205154.
--- RBK0036I: Saving partition layout.
--- RBK0133I: Using directory /backup/raspberrypi/raspberrypi-rsync-backup-20200113-175545 for hardlinks.
--- RBK0158I: Creating native rsync backup "/backup/raspberrypi/raspberrypi-rsync-backup-20200217-205154".
--- RBK0085I: Backup of type rsync started. Please be patient.
--- RBK0078I: Backup time: 00:00:01.
--- RBK0159I: 3 backups kept for rsync backup type.
--- RBK0007I: Starting services: 'systemctl start mariadb && systemctl start apache2 && systemctl start mosquitto && systemctl start openhab2 && systemctl start tvheadend'.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.4.3-dev (01f2caa) stopped at Po úno 17 20:53:34 CET 2020.
--- RBK0049I: Messages saved in /root/temp/raspiBackup.msg
--- RBK0026I: Debug logfile saved in /root/temp/raspiBackup.log.
--- RBK0017I: Backup finished successfully.

framps, 2020-02-17:
@davepee Seems you didn't use option `--telegramNotifications S`. Otherwise you don't get any notification. I will add a default `SF` later on - but as of now now there is no default in the prototype.

davepee, 2020-02-17:
You're right, sorry I overlooked _"Default will be_":) Now I got: `Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: chat not found".`. ChatID is correct. Verified via https://api.telegram.org/bottoken/getUpdates and `OpenHab2`.

Hofei90, 2020-02-17:
You get your Telegram Chat ID with the RawBot and you need a Telegram Bot Token, this you get from BotFather during creating new Bot

framps, 2020-02-17:
> Also die drei Ausrufezeichen erhalte ich nach wie vor bei einem Fehlschlag.

That's the way to mark the backup failure. It's the normal failure message also sent in the email or written on the console.
 
> Die 3 Optionen funktionieren, jedoch habe ich noch einen Änderungswunsch:
> Kannst du die angehängte Datei raspiBackup.msg als Text mit senden und nicht als Dateianhang.

The messages written as text are unreadable in telegram :cry:

davepee, 2020-02-17:
> You get your Telegram Chat ID with the RawBot and you need a Telegram Bot Token, this you get from BotFather during creating new Bot
>
Yes, my Token and ChatID (group chatID with `-` string) is correct. I use it in other applications without a problem.

Hofei90, 2020-02-17:
Try without Quotes ' '.
So it works for me:
` sudo bash raspiBackup.sh -F -a -o --telegramToken <token> --telegramChatID <id> --telegramNotifications SFM <backuppath>`

framps, 2020-02-17:
>  ChatID is correct. Verified via https://api.telegram.org/bottoken/getUpdates and `OpenHab2`.

I get ```{"ok":false,"error_code":404,"description":"Not Found"}```

which doesn'd sound OK :cry: 

> You're right, sorry I overlooked _"Default will be_":) Now I got: `Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: chat not found".`. ChatID is correct. Verified via https://api.telegram.org/bottoken/getUpdates and `OpenHab2`.

davepee, 2020-02-17:
> Try without Quotes ' '.
>
It is not right way :(

`./raspiBackup.sh -F --telegramToken 1xxxxxxx:AAxxxxx9-3vM3UjrrTxxxxxxxxxxRs --telegramChatID \-3xxxxxxx5`

??? RBK0090E: Option --telegramChatID requires a parameter. If parameter starts with '-' start with '\-' instead.
--- RBK0091I: Invoke 'raspiBackup.sh -h' to get more detailed information of all script invocation parameters.

`\` do next line in command...

framps, 2020-02-17:
@davepee I used a bot get_id to retrieve my ChatID. 

> --telegramChatID \-3xxxxxxx5

I don't think a telegram id starts with a dash. (I'm a Telegram noob)

davepee, 2020-02-17:
> I don't think a telegram id starts with a dash.
>
[Yes](https://www.openhab.org/addons/bindings/telegram/).

 _That is the chatId. Note that if using a Telegram group chat, the group chatIds are prefixed with a dash that must be included in the config file. (e.g. bot1.chatId: -22334455)"_

framps, 2020-02-17:
> `\` do next line in command...

You have to put the option argument in quotes :wink: But again - I don't think a clientid starts with a dash.

davepee, 2020-02-17:
Some loop is here or I am tired :)

> You have to put the option argument in quotes wink But again

Yes, I Know :) See previous [comment](https://github.com/framps/raspiBackup/issues/205#issuecomment-587143061) and [reply](https://github.com/framps/raspiBackup/issues/205#issuecomment-587153876).

>  I don't think a clientid starts with a dash.
>
And [this](https://github.com/framps/raspiBackup/issues/205#issuecomment-587160251) again:)

framps, 2020-02-18:
@davepee Should work now with the latest version with a leading backslash, e.g. `--telegramChatID '\-3xxxxxxx5'` :wink:

Hofei90, 2020-02-19:
Ja, jetzt funktioniert es mit und ohne vorangestellten "-". Vor dem Update gabs hier tatsächlich ein Problem.
Das angehängte Message File gefällt mir immer noch nicht. Mir wär die Ausgabe in Textform lieber.

----

Yeah, now it works with or without a prefixed "-". Before the update there was actually a problem here.
I still do not like the attached message file. I would prefer the output in text form.

framps, 2020-02-19:
Lastest version now accepts also notification option `m` which will push the messages into telegram and doesn't use a file.

framps, 2020-02-20:
@davepee @Hofei90 As far as I can see the telegram prototype works fin. Any other comments? Otherwise I'll put this PR on my queue, make the code more robust and make sure the tokens and ids will not be displayed in the debug log or any messages and finally will merge the code in version 0.6.5.1.

Hofei90, 2020-02-20:
Its work for me.

But please add in Configfile too

framps, 2020-02-20:
> But please add in Configfile too

Good point. Will do :+1:

davepee, 2020-02-21:
I'm sorry for the delay with my feedback. I was busy on a business trip. Now I have tested all parameters and everything works fine :)

framps, 2020-02-21:
@davepee NP. Glad you're also fine with the prototype :smiley: . 

When I added the missing pieces (config file constants for telegram and masquerade the token and id)  I'll push another version. Would be great if you and @Hofei90 then can do a final test.

framps, 2020-02-22:
Just added config options for telegram and token and id masquerading. Please take your time to test the latest (hopefully latest :smile: ) prototype version.

davepee, 2020-02-23:
> Just added config options for telegram and token and id masquerading. Please take your time to test the latest (hopefully latest  ) prototype version.
>
Yes, latest version works well for me :)

davepee, 2020-02-23:
I thought one more improvement :) Send a message when raspiBackup starts. The script stops several services and the backup takes several minutes (sometimes hours). The services are unavailable at the time and it would be good to know that this is related to running the backup. It is not important but it would be nice :)

framps, 2020-02-23:
Thank you very much for testing.

You have a valid request. The purpose of the eMail is just to inform you raspiBackup ran and the status of the run. With a messenger notification it now makes sense to send a start notification. But that's not related to this feature request which just deals with Telegram support.

Please open a new feature request for this raspiBackup start notification :wink:

framps, 2020-02-24:
@davepee I just updated the prototype and added option --notifyStart :laughing: It's also already rolled in into the beta because I think it's a useful feature.

davepee, 2020-02-26:
> @davepee I just updated the prototype and added option --notifyStart  It's also already rolled in into the beta because I think it's a useful feature.
>
Perfect:) It works very well!

davepee, 2020-02-26:
I now upgared from 0.6.4.3-dev to version 0.6.5-beta with -U option (if I'm not mistaken it is new feature too:)) But there `Unknown option --notifyStart`. Do you know about it?

framps, 2020-02-26:
I didn't publish the latest beta code :cry: Will do it soon. But please note as of now telegram support is not in beta. But I will check whether it's acceptable to roll in the telegramm feature into beta 0.6.5 instead of waiting for 0.6.5.1.

davepee, 2020-02-26:
No problem ;)

framps, 2020-02-26:
I reviewed the changes for telegram and notifyStart and rolled the code into the beta. It's not documented as of now - but you can give it a try :wink: 

In order to get the new beta you have to revert to 0.6.5.3 with option -V and use option -U again.

davepee, 2020-02-28:
I'm not sure of the upgrade procedure. Revert to 0.6.5.3? In my case was previous version 0.6.4.3-dev. After downgrade and upgrade I have again 0.6.5-beta without `--notifyStart` option. Could you send me a direct link to new beta please? :)

framps, 2020-02-28:
My fault. I had a typo in the filename when I uploaded the new beta code :disappointed: . It's fixed now and you should get the latest beta code with -V and -U. 

But you also can execute following steps if you want:
```
wget https://www.linux-tips-and-tricks.de/raspiBackup_beta.sh -O raspiBackup.sh
sudo mv raspiBackup.sh /usr/local/bin
sudo chmod 755 /usr/local/bin/raspiBackup.sh
sudo chown root.root /usr/local/bin/raspiBackup.sh
```

Sorry for the inconvenience.

davepee, 2020-02-28:
Ok Thanks:) I now downloaded from link. But this version not works so well. Instead of start notify message I get `!!! RBK0228W: Sent telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: message text is empty".`

framps, 2020-02-28:
Hm ... works for me. Before I check the debug log please let's make sure we use the same code level. 
```
raspiBackup.sh --version
Version: 0.6.5-beta CommitSHA: 9dcb120 CommitDate: 2020-02-28 CommitTime: 13:48:21
```

That's the code I downloaded from [here](https://www.linux-tips-and-tricks.de/raspiBackup_beta.sh) with wget. Please double check. If you still get the same error with this code level please provide the debug log for further analysis.

framps, 2020-03-04:
@davepee Would you like you to confirm telegram support works in the beta before I make the beta an official version? Latest version works without any issue:
```
raspiBackup.sh --version
Version: 0.6.5-beta CommitSHA: 23ff9f3 CommitDate: 2020-03-03 CommitTime: 17:35:05
```

Just revert to 0.6.4.3 with -V and update with option -U to get this latest version.

framps, 2020-03-04:
:cry: Thank you very much for your test. 

Would you please provide the debug log ? Please check there is no sensitive information in there (raspiBackup just masqueraded some details like telegram token and chatid...but there may still be some other sensitive info)

davepee, 2020-03-04:
I hope I anonymized enough and text will be readable :)
```
20200304-223017 DBG 2535:      --- LOG_OUTPUT: 3
20200304-223017 DBG 2536:      --- Using logfile /root/Test/raspiBackup_beta.log
20200304-223017 DBG 2537:      --- Using msgfile /root/Test/raspiBackup_beta.msg
20200304-223017 DBG 2545:      --- BACKUPTARGET_DIR: /backup/raspberrypi/raspberrypi-dd-backup-20200304-223008
20200304-223017 DBG 2546:      --- BACKUPTARGET_FILE: /backup/raspberrypi/raspberrypi-dd-backup-20200304-223008/raspberrypi-dd-backup-20200304-223008.img
20200304-223018 DBG 2548:      <-- setupEnvironment 
20200304-223018 DBG 2675:      --> colorAnnotation 1 --- RBK0009I: raspberrypi: raspiBackup_beta.sh V0.6.5-beta (23ff9f3) started at St bře  4 22:30:18 CET 2020.
20200304-223018 DBG 2699:      <-- colorAnnotation 
--- RBK0009I: raspberrypi: raspiBackup_beta.sh V0.6.5-beta (23ff9f3) started at St bře  4 22:30:18 CET 2020.
20200304-223018 MSG 7281:  --- RBK0009I: raspberrypi: raspiBackup_beta.sh V0.6.5-beta (23ff9f3) started at St bře  4 22:30:18 CET 2020. 
20200304-223018 DBG 2752:      --> sendTelegramm 
20200304-223018 DBG 2725:          --> sendTelegramMessage 
20200304-223020 DBG 2734:          --- Telegram response:
20200304-223020 DBG 2734:          --- {"ok":false,"error_code":400,"description":"Bad Request: message text is empty"}
20200304-223021 DBG 2743:          --- Error sending msg: {"ok":false,"error_code":400,"description":"Bad Request: message text is empty"}
20200304-223021 DBG 2675:              --> colorAnnotation 1 !!! RBK0228W: Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: message text is empty".
20200304-223021 DBG 2699:              <-- colorAnnotation 
[1;33m!!! RBK0228W: Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: message text is empty".[0m
20200304-223021 MSG 2744:          !!! RBK0228W: Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: message text is empty". 
20200304-223021 DBG 2747:          <-- sendTelegramMessage 
20200304-223021 DBG 2725:          --> sendTelegramMessage --- RBK0009I: raspberrypi: raspiBackup_beta.sh V0.6.5-beta (23ff9f3) started at St bře  4 22:30:18 CET 2020.
20200304-223021 DBG 2725:          --> !!! RBK0228W: Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: "Bad Request: message text is empty".
20200304-223023 DBG 2734:          --- Telegram response:
20200304-223023 DBG 2734:          --- {"ok":true,"result":{"message_id":320,"from":{"id":xxx,"is_bot":true,"first_name":"xxx","username":"xxx"},"chat":{"id":-@@@@@@@@5,"title":"xxx","type":"group","all_members_are_administrators":true},"date":1583357423,"text":"--- RBK0009I: raspberrypi: raspiBackup_beta.sh V0.6.5-beta (23ff9f3) started at St b\u0159e  4 22:30:18 CET 2020.\n!!! RBK0228W: Sent to telegram failed. curl RC: 0 - HTTP CODE: 400 - Error description: \"Bad Request: message text is empty\"."}}
20200304-223023 DBG 2738:          --- Message sent
20200304-223023 DBG 2675:              --> colorAnnotation 1 --- RBK0229I: Telegram notified.
20200304-223024 DBG 2699:              <-- colorAnnotation 
--- RBK0229I: Telegram notified.
20200304-223024 MSG 2739:          --- RBK0229I: Telegram notified. 
20200304-223024 DBG 2747:          <-- sendTelegramMessage 
20200304-223024 DBG 2768:      <-- sendTelegramm true
20200304-223024 DBG 2675:      --> colorAnnotation 1 !!! RBK0165W: =========> NOTE  <========= 
20200304-223024 DBG 2675:      --> !!! RBK0165W: This is a betaversion and should not be used in production. 
20200304-223024 DBG 2675:      --> !!! RBK0165W: =========> NOTE <=========
20200304-223024 DBG 2699:      <-- colorAnnotation 
[1;33m!!! RBK0165W: =========> NOTE  <========= [0m
[1;33m!!! RBK0165W: This is a betaversion and should not be used in production. [0m
[1;33m!!! RBK0165W: =========> NOTE <=========[0m
20200304-223024 MSG 7293:  !!! RBK0165W: =========> NOTE  <=========  
20200304-223024 MSG 7293:  !!! RBK0165W: This is a betaversion and should not be used in production.  
20200304-223025 MSG 7293:  !!! RBK0165W: =========> NOTE <========= 
20200304-223025 DBG 2675:      --> colorAnnotation 1 --- RBK0128I: Using logfile /root/Test/raspiBackup_beta.log.
20200304-223025 DBG 2699:      <-- colorAnnotation 
--- RBK0128I: Using logfile /root/Test/raspiBackup_beta.log.
20200304-223025 MSG 7297:  --- RBK0128I: Using logfile /root/Test/raspiBackup_beta.log. 
20200304-223025 DBG 2675:      --> colorAnnotation 1 --- RBK0116I: Using config file /usr/local/etc/raspiBackup_beta.conf.
20200304-223025 DBG 2699:      <-- colorAnnotation 
--- RBK0116I: Using config file /usr/local/etc/raspiBackup_beta.conf.
20200304-223026 MSG 7300:  --- RBK0116I: Using config file /usr/local/etc/raspiBackup_beta.conf. 
20200304-223026 DBG 7301:  --- Read config /usr/local/etc/raspiBackup_beta.conf : 
20200304-223026 DBG 7301:  --- UUID=4c8ec1d6-f786-467e-b8b8-d184de4d8f37
20200304-223026 DBG 1451:      --> logOptions 
20200304-223027 DBG 1453:      --- Options:  -F --notifyStart --telegramToken 1@@@@s(46) --telegramChatID \-@@@@@@@@5 --telegramNotifications SFm
20200304-223027 DBG 1454:      --- APPEND_LOG=0
20200304-223027 DBG 1455:      --- APPEND_LOG_OPTION=-a
20200304-223027 DBG 1456:      --- BACKUPPATH=/backup
20200304-223027 DBG 1457:      --- BACKUPTYPE=dd
20200304-223027 DBG 1458:      --- AFTER_STARTSERVICES=
20200304-223027 DBG 1459:      --- BEFORE_STOPSERVICES=
[... cut ...]
20200304-223058 DBG 3128:              --- =================================================================================================================================================================================
20200304-223058 DBG 3129:              --- =================================================================================================================================================================================
20200304-223059 DBG 3130:              --- =================================================================================================================================================================================
```

framps, 2020-03-04:
Thank you very much for the log which helped me to find the root cause for the issue :+1: In contrast to my test you don't request an eMail  but I request both an email and a telegram notification which makes sense if you just want to get a telegram notification.

It's now fixed :smiley: 
```
 ./raspiBackup.sh --version
Version: 0.6.5-beta CommitSHA: 1170419 CommitDate: 2020-03-04 CommitTime: 23:54:34
```
I also noticed the masquerading of sensitive information for telegram is still not complete :cry: . The masquerading issue is an ongoing task for me and I'll not fix this for 0.6.5 but will keep this in mind for future masquerading enhancements.

davepee, 2020-03-05:
Perfect, I'm glad I helped a little. Now it works!
But I have one cosmetic detail. When I used `-m` option I got message below. Is this necessary in parallel with the `raspberrypi: Backup started.` report? :)

`--- RBK0009I: raspberrypi: raspiBackup_beta.sh V0.6.5-beta (1170419) started at Čt bře  5 07:59:15 CET 2020.
--- RBK0229I: Telegram notified.`

Masquerading is sufficient there is no problem to fine tune by user :)

framps, 2020-03-05:
I see your point. As you said it's just a cosmetic issue. In order to get this fixed I have to reshuffle some code but I don't want to do this now. Beta is frozen and only serious issues will be fixed. I will fix this in the next release.

Please close the issue if you're fine now.

framps, 2020-03-07:
@davepee Any chance to close this issue? I'd like to publish the beta the next days :smiley:

davepee, 2020-03-07:
Yes, of course. Thanks:)

closed 2020-03-07 by davepee