framps/raspiBackup #207: Any issues with a subdirectory in Backup Path or Configure major options

mink1956, 2020-02-24:
Respectfully,

Apologies if this has already been discussed, but a search did not turn up a similar result.

I have tested defining backup targets as a subdirectory of local mount point in the
 **Backup path for backups** 
entry in raspiBackup Configuration Tool. example:
**/xxx/backups-lmsx**
Where the mount point is **/xxx** , and **/xxx/backups-lmsx** is a subdirectory on the mounted external local volume.

Since the root mount resides on the SD card, while the subdirectory is on USB drive, USB/SSD or USB/HHD, then one may backup multiple Pi machine versions, each on a different Sd card, providing care is taken to have the same top level, root target backup mount point. 

Each Pi Machine version would have its own subdirectory pointed to, in its Installation and Configuration settings.
This should also work for a network mount point as long as all the different Pi machine versions had the same /etc/fstab entries pointing to the same network mount point, 

Combining this approach with the GUI **_SD Card Copier_** makes for a robust experimental versioning system.  I have tested switching subdirectories by rerunning the configuation tools and it works. 

Is there a problem I've missed with this approach to allow testing of different machine versions on SD cards to the same USB drive? I've written a detailed tutorial about this, pending confirmation and further testing. This also allows two identical machines to serve as backups of each other in case of hardware failure. This assumes care is taken in fstab and static IP setup.

Thank you kindly for your time and consideration.

framps, 2020-02-24:
raspiBackup appends the hostname to the backup path and in this directory creates a directory with a timestamp. See [Backup directory structure](https://www.linux-tips-and-tricks.de/en/backup/#directory) for details. There you have `/mnt/backup/raspberrypi/raspberrypi-dd-backup-20160415-222900`. Backup path is `/mnt/backup`. Hostname of the Raspberry is `raspberrypi`. The final backup directory then consists of a combination of hostname, backup type and backup date.

So as long as all your Raspberries have different hostnames (they should have :wink: ) you can store all backup in the same backup path.

> ... or Configure major options ...

I'm sorry. I don't get your point about the options and configuration :worried:

mink1956, 2020-02-25:
Thank you for quick answer. It seems a trivial question now, since after all everything in linux is a file, I just hadn't seen an explicit example. 

Each SD Card machine version has same the RPi4 OS, and may (or may not) have same Static IP/Host name, but will be defined to backup to different subdirectory of the USB drive backup Folder. I swap the SD Card to run the different machine for testing or restoration. The USB mount point is defined via PARTUUID of course.

I am using
**Installation and Configuration Tool** | **Configure major options** | **Backup path for backups**
for raspibackup 
from invoking raspiBackupInstallUI.sh
to set the backup path for each set of SD cards for a given machine series. Essentially this is just inserting an extra subdirectory between the mount point and the subdirectory that raspbiBackup builds. I was asking if this could create unforeseen problems. 

I suppose I could just directly edit the config file on the master SD Card of each machine series. All the SD Card machines are sharing the USB/SSD dirive top level target for their backups.
I then clone with SD Card Copier to create a backup SD Card, 
Each machine series has a different subdirectory defined here:
[image]
where /xxx is the USB drive mount point, and ../backup-lmsx, 
For example is the subdirectory defined for this machine series.
```
1:~ $ ls -lh /nfs
total 32K 
drwxrwsr-x 12 pi   audio 4.0K Feb 16 13:42 audio
drwxr-xr-x  2 root root  4.0K Feb 21 11:50 backup-lms
drwxr-xr-x  3 root root  4.0K Feb 22 05:00 backup-lmsx 
drwxr-xr-x  2 root root  4.0K Feb 22 14:13 backup-pihole 
```
lms is the production series, lmsx the experimental series. Pihole is for the other pi in case of hardware failure of the network security pi, so the LMS pi can be swapped in. Yes I suppose that could be done as an automatic thing, but a bit much for my pea brain right now. I don't know scripting yet.

At some point, I may start again using the SD Card just as a boot loader and put / on the SSD, so this would not work, Hm, a different partion for each machine series' / (root). No that seems to much asking for trouble. The SD Card boots are reasonably fast and allow a nice backup system.

Oh BTW

Respectfully, I ran into an issue with a line in the /etc/fstab for optimizing SD Cards from your topic
see [this ](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=266127)
Thank you kindly,
cheers
Alan

framps, 2020-02-25:
As far as I understand you may have the same hostname for different Raspberries. Therefore you have to introduce an additional level in the backup path which makes sure you can have backups with same hostnames. As far as I understand you use `/backup/<someName>` and have every system to set <someName> which will be unique (lms, lmsx, pihole, ...). Then you are fine. 

Note: You don't have to modify the backup path in the config. You can just pass the backup path in the invocation of raspiBackup in your cron tab.

Example:

On lms system:
```
0 1     * * 6   root    /usr/local/bin/raspiBackup.sh /backup/lms
```
On lmsx system:
```
0 1     * * 6   root    /usr/local/bin/raspiBackup.sh /backup/lmsx
```
You may also use one of the other config files to overwrite the backup path.  See [here](https://www.linux-tips-and-tricks.de/en/backup/#parameters) the details about the possible configuration files and their precedence. There exist multiple ways to get it done :wink: 

Thank you for pointing me to your post. Looks like it's a very nasty issue.

mink1956, 2020-02-25:
Thank you for the detailed answer. Your understanding of the environment is essentially correct. Also thank you for the tips about cron. I have used them. It is nice to have an explicit, clear example! 

Yes the  fstab  tmpfs issue with LogitechMediaServer was confusing. It may have been a regression with an RPi system update, as I thought LMS worked through a reboot initially, but hard to remember.  I had installed several pluggins that day and initially suspected them. Finally, after much searching for answers, I used a pre-LMS 'golden' SD Card that was a fairly fresh install with configuration of ssh, vnc, users, etc (but had the modified fstab) and reinstalled LMS. When that also failed on reboot, I suspected the fstab mods, I can live without that line.<G> Still seems rather odd.

You correctly warn people to test the backup. I found that the config script offered  a vnc-server service for selection that wasn't started. I did not know this so checked it. The raspiBackub.sh script threw an error when it hit that service and failed. Since it happened that the LMS was stopped before the vnc, and in the start order it was AFTER the vnc module, it was stopped and never restarted. Thus the effect was to stop the media server. I did not see this at first since Real VNC must be using a different service program and I could still log in to it fine. Anyway, editing out the incorrect vnc server program from the config file fixed this. Curious that the script thinks this vnc service is available to stop, but misses the one that is running, I suppose. Perhaps something is off in the system's configuration. Perhaps this should be a separated issue, but may not really be worth it since all is working now. Not sure how important to stop the correct vnc server for the backup anyway. Is there a discussion of issues relating to not stopping services during backup?

Thank you again,
cheers

framps, 2020-02-25:
> Is there a discussion of issues relating to not stopping services during backup?

No. As I documented there are users who don't stop any services and the restored backup works perfectly. It totally depends on your environment which services you have to stop in which sequence. Either you know how these services depend on or you test carefully. 

As far as I understand you're fine with my answers to your question :wink: Please close the issue.

closed 2020-02-25 by mink1956