framps/raspiBackup #212: Octopi / Octoprint

NightSkySK, 2020-03-04:
I've got installed on my raspberry [Octoprint](https://github.com/foosel/OctoPrint) which is a kind of print server for 3D printer. Is it possible to configure the raspiBackup to not allow the scheduled backup procedure if the print is ongoing (Octoprint is sending gcode to the printer)?

Secondly, even after read FAQ as I'm a beginner user, I'm not able to choose in which sequence and which service should be stopped before the backup procedure will start. Here is the list of services which are active on my installation:
 [image]

framps, 2020-03-08:
Any update? Otherwise please close this issue :wink:

NightSkySK, 2020-03-12:
Thank you for your answer and hints. However, I'm still not on the level to write bash code which would do what is needed in this case. Let's hope that some more advance user will create it and share it with the community, till than I will stick to run manual backups as it already makes job much easier for me.

framps, 2020-03-12:
I don't think there will be somebody from the community dealing with your issue. It's a very specific issue. 

Frankly I think if you schedule the backup at 5 am in the morning there is a very low probability there is some printing activity.

Please close this issue if you're fine now with my answers.

closed 2020-03-12 by NightSkySK