framps/raspiBackup #213: mail not working on newest beta

AnduriI, 2020-03-06:
hey framps,

after using the smartRecycle on some of my raspberries I decided to give this feature to all of them (only 2 of them were missing). They worked before and did a recent backup (doing a weekly backup for all).
After the update they both complained about mail not being installed and stopped the raspiBackup.sh.

```
pi@Octopi:~ $ sudo raspiBackup.sh -F --smartRecycle --smartRecycleOptions "0 8 18 3"
--- RBK0009I: Octopi: raspiBackup.sh V0.6.5-beta (1170419) Fr 6. Mär 14:57:10 CET 2020 gestartet.
!!! RBK0165W: =========> HINWEIS <========= 
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte. 
!!! RBK0165W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /home/pi/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
!!! RBK0201W: 8-) Freundlicher Hinweis: Führe einen Restoretest durch. Du wirst noch 2 mal erinnert werden.
!!! RBK0124W: Simulationsmodus an.
??? RBK0039E: Mail Program mail ist nicht installiert um eMail zu senden.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0049I: Meldungen wurden in /home/pi/raspiBackup.msg gesichert.
--- RBK0010I: Octopi: raspiBackup.sh V0.6.5-beta (1170419) Fr 6. Mär 14:57:11 CET 2020 beendet mit Returncode 124.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
```

The moment I comment out the DEFAULT_EMAIL it works again.

By the way, also the restore seems to not worked (never needed to use that before):
```
$ sudo raspiBackup.sh -V
--- RBK0117I: Aktuelle Scriptversion: 0.6.5-beta 0.1.3 0
grep: /usr/local/bin/raspiBackup.0.6.5-beta: Datei oder Verzeichnis nicht gefunden
/usr/local/bin/raspiBackup.sh: Zeile 3238: [$version]=$versionFile: Falscher Feldbezeichner.
grep: 0.1.3: Datei oder Verzeichnis nicht gefunden
/usr/local/bin/raspiBackup.sh: Zeile 3238: [$version]=$versionFile: Falscher Feldbezeichner.
grep: 0.sh: Datei oder Verzeichnis nicht gefunden
/usr/local/bin/raspiBackup.sh: Zeile 3238: [$version]=$versionFile: Falscher Feldbezeichner.
--- RBK0118I: Verfügbare Scriptversionen:
--- RBK0119I: 0: 0
--- RBK0119I: 1: 0
--- RBK0119I: 2: 0.6.4
--- RBK0119I: 3: 0.6.4.1-beta
--- RBK0119I: 4: 0.6.4.2
--- RBK0119I: 5: 0.6.4.3
--- RBK0119I: 6: 0.6.4.3-beta
--- RBK0119I: 7: $MYHOMEURL/de/versionshistorie
--- RBK0119I: 8: $MYHOMEURL/de/versionshistorie
--- RBK0119I: 9: $MYHOMEURL/de/versionshistorie
--- RBK0119I: 10: $MYHOMEURL/en/versionhistory
--- RBK0119I: 11: $MYHOMEURL/en/versionhistory
--- RBK0119I: 12: $MYHOMEURL/en/versionhistory
--- RBK0122I: Auswahl der Version die wiederhergestellt werden soll (0-12)
6
--- RBK0122I: Auswahl der Version die wiederhergestellt werden soll (0-12)
6
--- RBK0122I: Auswahl der Version die wiederhergestellt werden soll (0-12)
5
--- RBK0122I: Auswahl der Version die wiederhergestellt werden soll (0-12)
4
--- RBK0122I: Auswahl der Version die wiederhergestellt werden soll (0-12)
4
```

AnduriI, 2020-03-06:
Just to add, this version of 0.6.5-beta works on my others:
```
sudo raspiBackup.sh -F
--- RBK0009I: Pimatic: raspiBackup.sh V0.6.5-beta (57e1a1c) Fr 6. Mär 14:36:12 CET 2020 gestartet.
!!! RBK0165W: =========> HINWEIS <========= 
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte. 
!!! RBK0165W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /home/pi/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
!!! RBK0124W: Simulationsmodus an.
--- RBK0151I: Backuppfad /mnt/Backup wird benutzt.
--- RBK0008I: Services werden gestoppt: 'sudo pimatic.js stop'.
Sending SIGTERM to pimatic(769), waiting for process exit...
Process didn't shutdown in time, sending signal SIGTERM to pimatic(769), waiting for process exit...
Process didn't shutdown in time, sending signal SIGQUIT to pimatic(769), waiting for process exit...
Stopped.
--- RBK0081I: Backup vom Typ rsync wird in /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200306-143611 erstellt.
--- RBK0036I: Partitionslayout wird gesichert.
--- RBK0133I: Verzeichnis /mnt/Backup/Pimatic/Pimatic-rsync-backup-20200302-020001 wird für Hardlinks benutzt.
--- RBK0158I: rsync Backup "/mnt/Backup/Pimatic/Pimatic-rsync-backup-20200306-143611" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 00:00:01.
--- RBK0007I: Services werden gestartet: 'sudo pimatic.js start'.
Started.
--- RBK0159I: 20 Backups werden für den Backuptyp rsync aufbewahrt.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
--- RBK0049I: Meldungen wurden in /home/pi/raspiBackup.msg gesichert.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0167I: Eine eMail wird versendet.
--- RBK0010I: Pimatic: raspiBackup.sh V0.6.5-beta (57e1a1c) Fr 6. Mär 14:36:53 CET 2020 beendet mit Returncode 0.
```

AnduriI, 2020-03-06:
I forgot to mention that the mail client itself is tested and working:
```
echo "Mail-Inhalt" | mail -s "Betreff" [MyMail@Adress]
```
just sent the correct mail to my inbox.

framps, 2020-03-06:
Thank you very much you reported these issues :+1: . The second one is a nasty one which was introduced when I started to add version check support for config files. It's not used right now but some code is already there. This screwed up the -V functionality :cry: 

The first one is a last change I thought is no issue ... but as you know - errare human est ...

Both issues are fixed in the current beta code. Unfortunately you have to download the code manually to update your local beta:

```
wget https://www.linux-tips-and-tricks.de/de/downloads/raspibackup-beta-sh/download -O raspiBackup.sh
sudo mv raspiBackup.sh /usr/local/bin
sudo chmod +x /usr/local/bin/raspiBackup.sh
sudo chown root.root  /usr/local/bin/raspiBackup.sh
```

Sorry for the inconvenience - and thank you again for reporting these nasty issues.

framps, 2020-03-11:
Verified by @AnduriI and closing on behalf of him.

closed 2020-03-11 by framps