framps/raspiBackup #214: Rsync backup not working on NUC

eloo, 2020-03-15:
Hi,

i'm using RaspiBackup already on an RPI and a OdroidC2 with DD and now i want to use the your script for my new NUC server.

Because i have an SSD installed with a higher capacity (64gb) i'm trying to use rsync instead of dd (dd is working).

But sadly the script is complaining when i try to backup using rsync.

Maybe i have some misconfiguration and you can assist me?
Here is the log output:
```
--- RBK0009I: NUC: raspiBackup.sh V0.6.5-beta (2ee324d) So 15. Mär 13:08:36 CET 2020 gestartet.
!!! RBK0165W: =========> HINWEIS <========= 
!!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte. 
!!! RBK0165W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /mnt/data/Backup/NUC/NUC-rsync-backup-20200315-130835/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0200I: /boot und / befinden sich auf derselben Partition .
Partition 2 beginnt nicht an einer physikalischen Sektorgrenze.
Partition 2 beginnt nicht an einer physikalischen Sektorgrenze.
??? RBK0013E: Es existieren mehr als zwei Partitionen, die nur mit dem Backuptype DD oder DDZ oder der Option -P gesichert werden können.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup /mnt/data/Backup/NUC/NUC-rsync-backup-20200315-130835 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0049I: Meldungen wurden in /home/user/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /home/user/raspiBackup.log gesichert.
--- RBK0010I: NUC: raspiBackup.sh V0.6.5-beta (2ee324d) So 15. Mär 13:08:59 CET 2020 beendet mit Returncode 115.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0167I: Eine eMail wird versendet.
```

And here my fstab
```
UUID=bcea8f8b-4ebf-48a3-922e-3b2bbbde8de3 /               ext4    errors=remount-ro 0       1
UUID=5c4bb914-6dc6-4bac-a839-a3ec18de4ea1 none            swap    sw              0       0
192.168.1.200:/export/red /mnt/data nfs4 rsize=8192,wsize=8192,timeo=14,intr,nofail 0 0
```

And my config
```
####################################################
#
# Default Konfigurationsdatei für raspiBackup.sh
#
# (C) 2013,2019 framp at linux-tips-and-tricks dot de
#
# Weitere Details zu den Parametern finden sich auf
# http://www.linux-tips-and-tricks.de/de/raspibackup/#parameter
#
# $Date: 2019-05-05 17:06:29 +0200$
# $Sha1: 0884c1a$
#
####################################################

# folgende Zeile nicht aendern
VERSION_DECONF="0.1.3"

#####################################################
#
# Hinweis:
#
# 0 schaltet eine Option aus
# 1 schaltet eine Option an
#
#####################################################

# Pfad wo das Backupfile gespeichert wird
DEFAULT_BACKUPPATH="/mnt/data/Backup"

# Anzahl der zu vorhaltenden Backups
DEFAULT_KEEPBACKUPS="3"
# Anzahl der vorzuhaltenden Backups fuer einen speziellen Backuptyp. Falls sie 0 ist wird DEFAULT_KEEPBACKUPS benutzt
DEFAULT_KEEPBACKUPS_DD=0
DEFAULT_KEEPBACKUPS_DDZ=0
DEFAULT_KEEPBACKUPS_TAR=0
DEFAULT_KEEPBACKUPS_TGZ=0
DEFAULT_KEEPBACKUPS_RSYNC=0

# Typ des Backups: dd, tar, xbmc or rsync
DEFAULT_BACKUPTYPE="rsync"

# zip tar oder dd backup
DEFAULT_ZIP_BACKUP="0"

# dd backup sichert nur den von den Partitionen belegten Platz
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0

# Durch && getrennte Befehle, die vor dem Starten des Backups auszuführen sind
DEFAULT_STOPSERVICES=":"

# Durch && getrennte Befehle, die nach dem Starten des Backups auszuführen sind
DEFAULT_STARTSERVICES=":"

# Durch && getrennte Befehle, die ausgeführt werden bevor Services gestoppt werden
DEFAULT_BEFORE_STOPSERVICES=""

# Durch && getrennte Befehle, die ausgeführt werden nachdem die Services wieder gestartet werden
DEFAULT_AFTER_STARTSERVICES=""

# emailadresse die das Backupergebnis erhält
DEFAULT_EMAIL="my-email@gmail.com"

# Sender emailadresse die mit ssmtp benutzt wird
DEFAULT_SENDER_EMAIL=""

# Weitere Parameter für das eMail programm (Optional)
DEFAULT_EMAIL_PARMS=""

# Message level (0 = minimal, 1 = detailed)
DEFAULT_MSG_LEVEL="0"

# mailprogram
DEFAULT_MAIL_PROGRAM="mail"

# Gerät wo das Backup restored wird
DEFAULT_RESTORE_DEVICE=""

# Log wird in eMail mitgeschickt
DEFAULT_APPEND_LOG=0

# Option die vom Mailprogramm benutzt wird um das Log anzuhängen (Z.B. -a oder -A)
DEFAULT_APPEND_LOG_OPTION="-a"

# Detailierte Logausgaben der Backupprogramme
DEFAULT_VERBOSE=0

# Check auf einen remoten Backupfad wird nicht vorgenommen
DEFAULT_SKIPLOCALCHECK=0

# Blocksize von dd
DEFAULT_DD_BLOCKSIZE=1M

# Weitere Parameter für dd
DEFAULT_DD_PARMS=""

# Excludeliste für das benutzte Backuprogramm
DEFAULT_EXCLUDE_LIST=""

# Notifizierung soll stattfinden wenn eine neue Scriptversion verfügbar ist
DEFAULT_NOTIFY_UPDATE=1

# Aufzurufende Erweiterungen
DEFAULT_EXTENSIONS="mem temp disk"

# partitionsbasierter backup modus
DEFAULT_PARTITIONBASED_BACKUP="0"

# Partitionsnummern, die gesichert werden sollen beim partitionsbasierten Backup
# Um alle partitionen zu sicher benutze "*"
DEFAULT_PARTITIONS_TO_BACKUP="1"
# Partitionsnummern muessen in Klammern stehen. Siehe Beispiel in folgender Zeile fuer die Partitionen 1,2 und 5
# DEFAULT_PARTITIONS_TO_BACKUP=(1 2 5)

# Sprache der Meldungen (DE oder EN)
DEFAULT_LANGUAGE="DE"

# Systeme auf die raspiBackup mit dem Parameter -y übertragen wird. ssh Zugang muss in der authorized keys Datei definiert sein
# Beispiel: "root@raspberrypi root@fhem root@openhab root@magicmirror"
DEFAULT_DEPLOYMENT_HOSTS=""

# Es wird nicht nach einer Restorebestätigung gefragt bei dem Device. Vorsicht!
DEFAULT_YES_NO_RESTORE_DEVICE=""

# Es werden Hardlinks für Partitionsbootfiles benutzt
DEFAULT_LINK_BOOTPARTITIONFILES=0

# Hardlinks werden für rsync benutzt
DEFAULT_USE_HARDLINKS=1

# Sichere boot partition mit tar
DEFAULT_TAR_BOOT_PARTITION_ENABLED=0

# eMail wird nur gesendet wenn ein Fehler aufgetreten ist. Vorsichtig benutzen !
DEFAULT_MAIL_ON_ERROR_ONLY=0

# Versionen bei denen keine deprecated message gemeldet werden soll, getrennt durch Kommata
DEFAULT_SKIP_DEPRECATED=""

# Reporte uuid
DEFAULT_USE_UUID=1

# teste auf bad blocks wenn beim restore eine Partition formatiert wird (Das dauert lange)
DEFAULT_CHECK_FOR_BAD_BLOCKS=0

# Das root Filesystem wird beim restore automatisch angepasst
DEFAULT_RESIZE_ROOTFS=1

# Zeitstempel werden vor den Meldungen ausgegeben
DEFAULT_TIMESTAMPS=0

# System status Informationen werden im Log aufgeführt
DEFAULT_SYSTEMSTATUS=0

# Interval fuer eine Restoretest Erinnerung (Einheit: Monate)
DEFAULT_RESTORE_REMINDER_INTERVAL=6

# Anzahl der Erinnerungen an einen Restoretest
DEFAULT_RESTORE_REMINDER_REPEAT=3

# GENERATED - DO NOT DELETE 
UUID=0f3f6b79-f4b6-477f-8c86-77d64bf2761b
```

As you can see i'm using your beta right now. but with the latest stable i have the same issue.

Thanks for your help
Best regards
eloo

framps, 2020-03-15:
There is a new option `--ignoreAdditionalPartitions` (config: `DEFAULT_IGNORE_ADDITIONAL_PARTITIONS=1` ) in the beta available (I just detected it's not documented as of now :cry: ) which allows to use rsync even there are more partitions on your SD card. **BUT**: First partition has to be /boot and the second has to be / **AND** all other partitions are just ignored and not saved. If that's acceptable for you just try this new option.

framps, 2020-03-21:
@eloo Any update on this issue?

eloo, 2020-03-22:
Hi, sorry for the late response but i'm quite busy right now.
The idea of using rsync or tar is simply to test the behaviour and backup time right now. 
IMHO the dd backup is taking really long because its not incremental, what the rsync backup would be.
So i'm checking out my options for creating a backup strategy. maybe something like twice a year using dd and monthly or weekly using rsync.

but i will try the `DEFAULT_IGNORE_ADDITIONAL_PARTITIONS` and give feedback.

Thanks for your help so far

framps, 2020-03-23:
Take your time :wink: . Just wanted to know whether there is something on my part to do.

framps, 2020-04-08:
This issue is now 3 weeks old. Any updates? Otherwise I'll close this issue in 1 week cause of inactivity.

eloo, 2020-04-13:
Okay, i have just triggered the backup. seems to work so far. 
now i just need to check the result and the time. 
but with your new option is starting at least :+1:

eloo, 2020-04-16:
looks like the rsync backup has made some problems.. looked like it was freezing and then i've aborted it. i will give it later a try again.
now im trying the tar option

framps, 2020-04-16:
Hm ... looks like it's your local environment which causes the rsync to hang. In order to find the root cause for the rsync issue I need the debug log which you find in the backup location :wink:

eloo, 2020-04-16:
okay.. will check this on my next try. but i guess i need to extract that log before i cancel the backup because it will "cleanup" the unfinished backup i guess?

framps, 2020-04-16:
No. If you cancel the backup the logs are saved in users home directory before the backup directory is cleaned up. You will get message RBK0026I which tells you where you find the logs :wink:

eloo, 2020-04-16:
ah nice.. then i will have a look asap

eloo, 2020-05-03:
Okay, i've just checked the logs and tried rsync again..
It just looks like it takes simply too long.
The job was started at 0500+2 and was yet still not finished 1600+2.
The current backup size was around 10gb only.. i've also seen that my backup device has a really high IO load.. maybe its for the rsync nature by calculating the hashes and my target is too slow.

With tar is looks pretty good.. backup time is around 30 minutes :+1: 
I will check the tar backup soon but so long i'm fine

framps, 2020-05-03:
That's a drawback of rsync: If the nunber of files is very high a lot of IO is required. But I'm frankly surprised a local attached SSD is too slow. Please close the issue of you're fine now.

eloo, 2020-05-03:
only my drive to backup is ssd..
the destination drive is a normal hdd.

yep for me its fine with the DEFAULT_IGNORE_ADDITIONAL_PARTITIONS option.
looking forward for a release with this option

closed 2020-05-03 by eloo