framps/raspiBackup #215: Synology mount backup failure

MaxBec, 2020-03-16:
Hey guys,

i just installed the newest beta, mounted my nas to my pi (is working, i can create new files etc) and started raspiBackup.sh. Unfortunately it is not working:

`--- RBK0009I: openhab: raspiBackup.sh V0.6.5-beta (2ee324d) started at Mon 16 Mar 2020 09:24:01 AM CET.
!!! RBK0165W: =========> NOTE  <=========
!!! RBK0165W: This is a betaversion and should not be used in production.
!!! RBK0165W: =========> NOTE <=========
--- RBK0128I: Using logfile /backup/openhab/openhab-rsync-backup-20200316-092400/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
!!! RBK0209W: UUIDs are not unique on devices and/or partitions an may cause issues. In case of error messages check them with 'sudo blkid' and make them unique.
??? RBK0039E: Mail program mail not installed to send emails.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /backup/openhab/openhab-rsync-backup-20200316-092400. This may take some time. Please be patient.
--- RBK0049I: Messages saved in /root/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /root/raspiBackup.log.
--- RBK0010I: openhab: raspiBackup.sh V0.6.5-beta (2ee324d) stopped at Mon 16 Mar 2020 09:24:05 AM CET with rc 124`

framps, 2020-03-16:
> ??? RBK0039E: Mail program mail not installed to send emails.

If you want to send eMails you have to have an eMail client installed and configured. Default is to use mail. See option -s on [the raspiBackup user's guide](https://www.linux-tips-and-tricks.de/en/backup/#parameters) :wink:

MaxBec, 2020-03-18:
Ok i was able to fix that issue.
What about this?
> !!! RBK0209W: UUIDs are not unique on devices and/or partitions an may cause issues. In case of error messages check them with 'sudo blkid' and make them unique.

And i now have the problem that the  backup is stopping sudddenly. I started it 8 hours ago... but when i try to run it again it says:
??? RBK0015E: There is already an instance of raspiBackup up and running

framps, 2020-03-18:
> What about this?
> 
> > !!! RBK0209W: UUIDs are not unique on devices and/or partitions an may cause issues. In case of error messages check them with 'sudo blkid' and make them unique.

This is a message which warns that you can get restore errors because of duplicate UUIDs. Please show the output of `sudo blkid -o udev` and I can tell you exactly what's going on here :wink: 

> And i now have the problem that the backup is stopping sudddenly. I started it 8 hours ago... but when i try to run it again it says:
> ??? RBK0015E: There is already an instance of raspiBackup up and running

Did you get any error message when it stopped? 

You can start raspiBackup only once. According to the error message it's still running. Please check with `sudo ps -ef | grep raspiBackup`. If there is no output for some reasons the lock file was not deleted and you have to delete it manually. `sudo rm /var/lock/raspiBackup`.

framps, 2020-03-21:
@MaxBec Any update on this issue?

framps, 2020-03-31:
I removed the check which causes this RBK0209W message from beta because it's a test for an edge case and creates a lot of questions and is very unproductive. Just Update to the lastest beta version with options `-S -U` and the message will be gone.

framps, 2020-04-08:
This issue is now 3 weeks old. Any updates? Otherwise I'll close this issue in 1 week cause of inactivity.

closed 2020-04-23 by framps