framps/raspiBackup #217: Time of regular backup doesn't fit

godlyblade, 2020-04-13:
First of all, THANK YOU for this amazing tool!

There is just one thing, I can't understand. RaspiBackup starts the regular backup process exact one hour after the setup time.

This are my current time zone settings:
```
pi@raspberrypi:~ $ sudo raspi-config
Current default time zone: 'Europe/Berlin'
Local time is now:      Mon Apr 13 12:52:10 CEST 2020.
Universal Time is now:  Mon Apr 13 10:52:10 UTC 2020.
```
Now, when I setup the regular backup to start at `13:00`, it will start at `14:00` local time / time of my macBook. So there is an offset of exact 1 hour and I thought about time zone settings, but they seems top be correct.

My information
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

The e-mail report says:
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.4.3 (d2acd2d) started at Mon 13 Apr 14:00:01 CEST 2020.
... (this e-mail is received at about 14:15 local time)

Best Regards!

framps, 2020-04-13:
Would you please show the contents of cat /etc/cron.d/raspiBackup ?

godlyblade, 2020-04-13:
Here it is:
```
pi@raspberrypi:~ $ sudo cat /etc/cron.d/raspiBackup
#
# Crontab entry for raspiBackup.sh
#
# (C) 2017-2019 framp at linux-tips-and-tricks dot de
#
# Create a backup once a week on Sunday morning at 5 am (default)
#
02 14 * * * root /usr/local/bin/raspiBackup.sh 
```
This is my current time setup...
[image]

framps, 2020-04-13:
Looks good so far. Would you please show the exact name of the backup directory created by the backup? [See here](https://www.linux-tips-and-tricks.de/en/backup/#directory) for the backup directory structure. My is called `asterix.framp.de-rsync-backup-20200411-010004` for example. At the end you have the time the backup was created. My backup was created at 01AM.

> Now, when I setup the regular backup to start at 13:00, 

How do you do this? According /etc/cron.d/raspiBackup and your screenshot you set the backup time to 2PM.

godlyblade, 2020-04-13:
Backup directory structure
`/backup/raspberrypi/raspberrypi-ddz-backup-20200413-140001/raspberrypi-ddz-backup-20200413-140001.img.gz`

> How do you do this? According /etc/cron.d/raspiBackup and your screenshot you set the backup time to 2PM.

Sorry, I changed the time between my posts to test something. The screenshot just should say, that the _time of regular backup_ is different to the time mentioned here `/etc/cron.d/raspiBackup`

framps, 2020-04-13:
Hm ... frankly I'm now confused :cry: 

The contents of /etc/cron.d/raspiBackup tells us the backup should be created at 2PM. The backup directory name tells us that the backup was created at 2PM as defined in cron. So everything is OK.

Where did you configure the backup should be created at 1PM?

godlyblade, 2020-04-13:
Okay, sorry for my confusing information, especially the screenshot. Give me a second, I will share a video with you!

godlyblade, 2020-04-13:
[Dropbox](https://www.dropbox.com/sh/ezaxut6lyvrxll0/AADowp5GL25dE_03HHjSR2Y9a?dl=0)

Please see the video... There you can see, that I configure the backup to start at 17:20 but it starts at 18:20.

framps, 2020-04-13:
Thank you very much for the video. I now got your issue :wink: 

You changed now the time to 17:20. Please check what's in /etc/cron.d/raspiBackup. Please check you have `20 17` in there. If yes it's not an issue with raspiBackup but with cron.

I found [Why are cronjobs running at the wrong time?](https://serverfault.com/questions/231220/why-are-cronjobs-running-at-the-wrong-time). Frankly I expect raspi-config to restart cron to activate the new timezone. But please check and follow the answers given on the page.

godlyblade, 2020-04-13:
You are right, in `/etc/cron.d/raspiBackup` I have `20 17` Sorry for labeling it as an issue...

Gotcha! A reboot did the job! Thank you for your support, I will donate you a coffee sir :)

closed 2020-04-13 by godlyblade

framps, 2020-04-13:
> Sorry for labeling it as an issue...

No problem. It's a nasty raspi-config issue and I created an [issue there](https://github.com/RPi-Distro/raspi-config/issues/111). Maybe they add a restart of cron whenever the timezone is changed.