framps/raspiBackup #218: raspiBackupRestore2Image - Restore failed with RC117 - Rootpartition creation failed with RC1

floviolleau, 2020-04-17:
Hello,

Thanks for your excellent job done with this script! It is a very good thing to be able to backup it while the pi is running.

I'm trying to put in place the backup of my Rpi's.

What I did is installing raspiBackup by using the UI installer `raspiBackupInstallUI.sh`.

I set the configuration to tar mode and activated the compression.
For now, I ran it one time by using `sudo raspiBackupWrapper.sh` and I generated the content below:

[image]

I'm successful able to open the .img file and the .tgz file with 7-zip.

I activated raspiBackupRestore2Image at the end of wrapper but no dd image was generated and I missed to save the output.

So, I ran it after directly `sudo raspiBackupRestore2Image.sh /media/nfs-mountplace/backups/server/server-tgz-backup-20200417-035353` and save the output below. Because of the clean up, the .dd file is removed and the RC117 error is a little be obscure.

What I'm trying to perform is to generate the .tgz and next produce a .dd image file from the .tgz,  save both and be able to use it to restore my Pi later. I don't want to restore now my pi. Maybe I misunderstanding the usage of `sudo raspiBackupWrapper.sh`?

Thanks helping me on that issue.
Kind regards

**Operating system I use:**
```
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

<details>
  <summary>All messages written by raspiBackup</summary>

  ```
sudo raspiBackupRestore2Image.sh /media/nfs-mountplace/backups/server/server-tgz-backup-20200417-035353
raspiBackupRestore2Image.sh v0.1.4, 2019-05-01/10:29:49 - f9df3c1
===> Backup source disk size: 122112 (MiB)
0+0 records read
0+0 recors written
0 bytes copied 0,000765468 s, 0,0 kB/s
Checking that no-one is using this disk right now ... OK

Disk /dev/loop0: 119,3 GiB, 128043712512 bytes, 250085376 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0x738a4d67.
/dev/loop0p1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 256 MiB.
/dev/loop0p2: Created a new partition 2 of type 'Linux' and of size 119 GiB.
/dev/loop0p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x738a4d67

Device       Boot  Start       End   Sectors  Size Id Type
[... 597 lines cut ...]
20200417-191126 DBG 2053:                       -- debugfs /sys/kernel/debug debugfs rw,relatime 0 0
20200417-191126 DBG 2053:                       -- systemd-1 /media/nfs-mountplace autofs rw,relatime,fd=36,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
20200417-191126 DBG 2053:                       -- systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=41,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
20200417-191126 DBG 2053:                       -- sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
20200417-191126 DBG 2053:                       -- configfs /sys/kernel/config configfs rw,relatime 0 0
20200417-191126 DBG 2053:                       -- /dev/mmcblk0p1 /boot vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro 0 0
20200417-191126 DBG 2053:                       -- tv:/media/Backup\0402 /media/nfs-mountplace fuse.sshfs rw,nosuid,nodev,noexec,noatime,user_id=0,group_id=0,allow_other 0 0
20200417-191126 DBG 2053:                       -- fusectl /sys/fs/fuse/connections fusectl rw,relatime 0 0
20200417-191126 DBG 2053:                       -- tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=94828k,mode=700,uid=1000,gid=1000 0 0
20200417-191126 DBG 2059:                       <- isMounted 1
20200417-191126 DBG 2596:                    -- Deleting dir /tmp/raspiBackup
rmdir: unable to delete '/tmp/raspiBackup': Folder is not empty
umount: /dev/loop0p1: not mounted.
umount: /dev/loop0p2: not mounted.
??? RBK0077E: Restore failed with RC 117. Check previous error messages.
20200417-191126 MSG 2606:                    ??? RBK0077E: Restore failed with RC 117. Check previous error messages. 
20200417-191126 DBG 2611:                    <- cleanupRestore 117
20200417-191126 DBG 2785:                    -> cleanupTempFiles 
20200417-191126 DBG 2792:                    <- cleanupTempFiles 
20200417-191126 DBG 2568:                 -- Terminate now with rc 117
--- RBK0010I: server: raspiBackup.sh V0.6.4.3 (d2acd2d) stopped at friday 17 april 2020, 19:11:26 (UTC+0200).
20200417-191126 MSG 2571:                 --- RBK0010I: server: raspiBackup.sh V0.6.4.3 (d2acd2d) stopped at friday 17 april 2020, 19:11:26 (UTC+0200). 

  ```
</details>

framps, 2020-04-18:
> What I did is installing raspiBackup by cloning the respository from master branch and then using the UI installer `raspiBackupInstallUI.sh`.

You don't need to clone the repo. Instead you should follow [these instructions](https://www.linux-tips-and-tricks.de/en/quickstart-rbk/) to install raspiBackup. Please reinstall raspiBackup this way to make sure you use the correct files and code levels. The git contents should be identical but the devil is a squirrel :wink: 

> I activated raspiBackupRestore2Image at the end of wrapper but no dd image was generated and I missed to save the output.

That's bad. Please save the output. 

> What I'm trying to perform is to generate the .tgz and next produce a .dd image file from the .tgz, save both and be able to use it to restore my Pi later. I don't want to restore now my pi. Maybe I misunderstanding the usage of `sudo raspiBackupWrapper.sh`?

That's correct. raspiBackupWrapper helps to invoke pishrink or raspiBackupRestore2Image together with raspiBackup.

RC117 tells you the restore failed. You should also get a message which tells you exactly the RC of the used restore tool. Are you sure your SD card you try to restore is OK?

> Where should I specify `--smartRecycle` option? Inside the wrapper?

That's up to you. But I suggest to add the corresponding option in the config file. Lookup the config options [here](https://www.linux-tips-and-tricks.de/en/backup/#parameters)

floviolleau, 2020-04-18:
Hi,

Thanks for your detailled and quick answer.

I will reinstall it today and get back to you.

As per the logs output I put in my first message (under spoiler messages), the RC117 is an error code from raspiBackup.

I got the RC117 when I ran `raspiBackupRestore2Image` only so for now, I'm not performing any kind of restore.

Maybe I misunderstanding the usage of `sudo raspiBackupRestore2Image` as well?
From what I understood, it uses the tar backup and generate the dd image without changind the sd card?

Lastly, from what I understood in the docs, I must set `DEFAULT_SMART_RECYCLE` to true and `DEFAULT_SMART_RECYCLE_DRYRUN` to false?

Thanks

framps, 2020-04-18:
> I got the RC117 when I ran `raspiBackupRestore2Image` only so for now, I'm not performing any kind of restore.

Sorry. I missed this :cry:  

There was an issue with raspiBackupRestore2Image #202 which was fixed in the beta branch. Looks like you mixed the 0.6.5 beta version of raspiBackup and the 0.6.4.3 version of `raspiBackupRestore2Image`. 

So you have to install the beta with option -U when you installed 0.6.4.3 and then grab `raspiBackupRestore2Image` from [here](https://github.com/framps/raspiBackup/blob/beta/helper/raspiBackupRestore2Image.sh).

> Lastly, from what I understood in the docs, I must set `DEFAULT_SMART_RECYCLE` to true and `DEFAULT_SMART_RECYCLE_DRYRUN` to false?

Yes, but use `0` for false and `1` for true.

framps, 2020-04-23:
@floviolleau Does it work now for you? If yes please close this issue. If no ... provide debugging information :wink:

floviolleau, 2020-04-25:
Hi,

Sorry for the delay. I will try tomorrow and get back to you.

Thanks

floviolleau, 2020-04-29:
Hi,

I just retried and it failed. All scripts are in beta mode.

PSB

<details>
  <summary>raspiBackup.msg</summary>

  ```
--- RBK0009I: server: raspiBackup.sh V0.6.5-beta (26e48d7) started at mercredi 29 avril 2020, 23:41:41 (UTC+0200).
!!! RBK0165W: =========> NOTE  <========= 
!!! RBK0165W: This is a betaversion and should not be used in production. 
!!! RBK0165W: =========> NOTE <=========
--- RBK0128I: Using logfile /home/user/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /media/nfs-mountplace/backups/server/server-tgz-backup-20200429-030002/server-tgz-backup-20200429-030002.tgz.
--- RBK0068I: Using bootpartition backup files starting with /media/nfs-mountplace/backups/server/server-tgz-backup-20200429-030002 from directory server-backup.
!!! RBK0143W: Target /dev/loop0 does not match with backup. Partitioning forced.
!!! RBK0065W: Device /dev/loop0 will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/loop0:
Number  Start   End       Size      Type     File system  Flags
 1      4,19MB  273MB     268MB     primary               lba
 2      273MB   128044MB  127771MB  primary
--- RBK0066I: Device /dev/loop0 will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/loop0p1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/loop0p2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N --- RBK0050I: Restoring backup from /media/nfs-mountplace/backups/server/server-tgz-backup-20200429-030002.
!!! RBK0139W: Partition creation ignores errors.
--- RBK0053I: Restoring first partition (boot partition) to /dev/loop0p1.
--- RBK0054I: Formating second partition (root partition) /dev/loop0p2.
??? RBK0185E: Rootpartition creation failed with RC 1.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed with RC 117. Check previous error messages.
--- RBK0010I: server: raspiBackup.sh V0.6.5-beta (26e48d7) stopped at jeudi 30 avril 2020, 00:03:24 (UTC+0200) with rc 117.
??? RBK0005E: Backup failed. Check previous error messages for details.
  ```
</details>

<details>
  <summary>raspiBackup.log</summary>

  ```
20200429-234141 DBG 2533:      --- +================================================================================================================================================+
20200429-234141 DBG 2534:      --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20200429-234141 DBG 2535:      --- +================================================================================================================================================+
20200429-234141 DBG 2536:      --- | ===>  Viele sensitive Informationen werden in dieser Logdatei maskiert. Vor dem Verteilen des Logs sollte es trotzdem ueberprueft werden  <=== |
20200429-234141 DBG 2537:      --- +================================================================================================================================================+
20200429-234141 DBG 2539:      --- LOG_OUTPUT: 3
20200429-234141 DBG 2540:      --- Using logfile /home/@USER@/raspiBackup.log
20200429-234141 DBG 2541:      --- Using msgfile /home/@USER@/raspiBackup.msg
20200429-234141 DBG 2549:      --- BACKUPTARGET_DIR: 
20200429-234141 DBG 2550:      --- BACKUPTARGET_FILE: 
20200429-234141 DBG 2552:      <-- setupEnvironment 
20200429-234141 DBG 2679:      --> colorAnnotation 1 --- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5-beta (26e48d7) started at mercredi 29 avril 2020, 23:41:41 (UTC+0200).
20200429-234141 DBG 2703:      <-- colorAnnotation 
--- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5-beta (26e48d7) started at mercredi 29 avril 2020, 23:41:41 (UTC+0200).
20200429-234141 DBG 2679:      --> colorAnnotation 1 !!! RBK0165W: =========> NOTE  <========= 
20200429-234141 DBG 2679:      --> !!! RBK0165W: This is a betaversion and should not be used in production. 
20200429-234141 DBG 2679:      --> !!! RBK0165W: =========> NOTE <=========
20200429-234141 DBG 2703:      <-- colorAnnotation 
!!! RBK0165W: =========> NOTE  <========= 
!!! RBK0165W: This is a betaversion and should not be used in production. 
!!! RBK0165W: =========> NOTE <=========
20200429-234141 DBG 2679:      --> colorAnnotation 1 --- RBK0128I: Using logfile /home/@USER@/raspiBackup.log.
20200429-234141 DBG 2703:      <-- colorAnnotation 
--- RBK0128I: Using logfile /home/@USER@/raspiBackup.log.
20200429-234141 DBG 2679:      --> colorAnnotation 1 --- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
20200429-234141 DBG 2703:      <-- colorAnnotation 
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
[... 718 lines cut ...]
20200430-000324 DBG 2321:                              <-- isMounted 1
20200430-000324 DBG 3191:                          --- Deleting dir /tmp/raspiBackup
umount: /dev/loop0p1: not mounted.
umount: /dev/loop0p2: not mounted.
20200430-000324 DBG 2679:                              --> colorAnnotation 1 ??? RBK0077E: Restore failed with RC 117. Check previous error messages.
20200430-000324 DBG 2703:                              <-- colorAnnotation 
??? RBK0077E: Restore failed with RC 117. Check previous error messages.
20200430-000324 DBG 3206:                          <-- cleanupRestore 117
[... cut ...]
??? RBK0005E: Backup failed. Check previous error messages for details.
20200430-000324 DBG 2795:                          --> sendEMail 
20200430-000324 DBG 2899:                          <-- sendEMail 
20200430-000324 DBG 3023:                      --- Masquerading some mount options
20200430-000324 DBG 3041:                      --- Masquerading home directory name
20200430-000324 DBG 3046:                      --- Masquerading hostname
20200430-000324 DBG 3051:                      --- Masquerading non local IPs
  ```
</details>

framps, 2020-04-30:
Thank you very much for the logs. I found
```
mkfs.ext4: Erreur d'entrée/sortie while writing out and closing file system
```
which is 
```
mkfs.ext4: IO error while writing out and closing file system
```
That's an error reported from mkfs when creating the ext4 filesystem in the dd file which should get the backup. Unfortunately it's not clear whether it's a read or write error. I guess it's a write error. 

The image was create on  `/media/nfs-mountplace` which was mounted
```
systemd-1 /media/nfs-mountplace autofs rw,relatime,fd=42,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
```

This is not an issue with raspiBackup. It seems to me there happened some kind of nfs communication error which cause mkfs to fail. Please try again but  use a local disk instead of an nfs mounted partition to check whether the issue is related to the usage of an nfs mounted partition.

floviolleau, 2020-04-30:
Hi,

Thanks for the answer. I will check that later as I need to buy a disk to be able to test it because I'm almost full on my rpi.

Thanks for you time.

framps, 2020-04-30:
Well, if you have a spare SD card big enough to keep the backup just plug in this card with an USB adapter :wink:

floviolleau, 2020-04-30:
Well, I don't have a USB adapter.... Usually I use the embedded SD card reader of a laptop :/

framps, 2020-04-30:
I see. Or just use any USB pen drive. I have multiple of them for data xfer or OS installation.

framps, 2020-05-13:
@floviolleau Any update on this issue?

floviolleau, 2020-05-15:
Hi,

I will receive soon a new SSD to add to my pi :). I only have small USB drives (4Go) and the pi as more stuffs to backup. Moreover, now, my backups started to failed (error RC109) so it's a tar issue as per the documentation (I use tar + compression for now) and I don't know why it started to have this issue despite I didn't changed anything...

Any ideas?

Thanks for your help

framps, 2020-05-15:
raspiBackup just uses the standard Linux tools to create backups. If tar complaints you should use option `-v` to get more detail information from tar what's going on and why it complaints.

framps, 2020-06-01:
@floviolleau : Any update on this issue from your side :wink:

floviolleau, 2020-06-01:
I received my drive yesterday 😉👍. I will try by EOW

framps, 2020-06-27:
Any news on this issue?

framps, 2020-07-11:
This issue will be close by me in 1 week because of inactivity if there is no update any more.

closed 2020-07-11 by framps