framps/raspiBackup #219: raspiBackupRestore2Image.sh not working

johnnypea, 2020-04-20:
Firstly, thank you for this marvelous tool. I use it for my RPis backups over the year now.

I store my backups in mounted network share (Synology NAS, samba). And I use the "tar" option. Recently I found out about about "raspiBackupRestore2Image.sh" helper script and I wanted to use it to create a generic image from one of my backups. I would like to use it with Etcher. But I can't get it to work.

I tried to run it from one of the RPis which I backup:
```
bash raspiBackupRestore2Image.sh /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855
```

master
```
raspiBackupRestore2Image.sh v0.1.4, 2019-05-01/10:29:49 - f9df3c1
===> Backup source disk size: 61052 (MiB)
0+0 records in
0+0 records out
0 bytes copied, 0.000640019 s, 0.0 kB/s
Checking that no-one is using this disk right now ... OK

Disk /dev/loop0: 59.6 GiB, 64017661952 bytes, 125034496 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0xed0159a9.
/dev/loop0p1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 256 MiB.
/dev/loop0p2: Created a new partition 2 of type 'Linux' and of size 59.4 GiB.
/dev/loop0p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0xed0159a9

Device       Boot  Start       End   Sectors  Size Id Type
/dev/loop0p1        8192    532480    524289  256M  c W95 FAT32 (LBA)
/dev/loop0p2      540672 125034495 124493824 59.4G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Invalid argument
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
===> Reloading new partition table
===> Restoring backup into /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855/MyRPi-backup.dd
--- RBK0009I: MyRPi: raspiBackup.sh V0.6.4.3 (d2acd2d) started at Mon 20 Apr 16:47:02 CEST 2020.
--- RBK0128I: Using logfile /root/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855/MyRPi-tar-backup-20200420-142855.tar.
--- RBK0068I: Using bootpartition backup files starting with /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855 from directory MyRPi-backup.
!!! RBK0143W: Target /dev/loop0 does not match with backup. Partitioning forced.
!!! RBK0065W: Device /dev/loop0 will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/loop0:
Number  Start   End      Size     Type     File system  Flags
 1      4.19MB  273MB    268MB    primary               lba
 2      277MB   64018MB  63741MB  primary
!!! RBK0066W: Device /dev/loop0 will be overwritten with the saved boot and root partition.
--- RBK0038I: Are you sure? y/N --- RBK0050I: Restoring backup from /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855.
--- RBK0051I: Restoring mbr from /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855/MyRPi-backup.mbr to /dev/loop0.
!!! RBK0139W: Partition creation ignores errors.
--- RBK0053I: Restoring first partition (boot partition) to /dev/loop0p1.
--- RBK0054I: Formating second partition (root partition) /dev/loop0p2.
--- RBK0055I: Restoring second partition (root partition) to /dev/loop0p2.
--- RBK0184I: Rootpartition check started.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0076I: Restore finished successfully.
--- RBK0010I: MyRPi: raspiBackup.sh V0.6.4.3 (d2acd2d) stopped at Mon 20 Apr 16:48:44 CEST 2020.
===> Shrinking Image /mnt/remote-dietpi/backups/MyRPi/MyRPi-tar-backup-20200420-142855/MyRPi-backup.dd
[... 111 lines cut ...]
pishrink.sh: Trying to recover corrupted filesystem ...
e2fsck 1.44.5 (15-Dec-2018)
e2fsck: No such file or directory while trying to open 
Possibly non-existent device?
pishrink.sh: ERROR occured in line 64: Filesystem recoveries failed. Giving up...
??? Error 247 received from piShrink

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): 
Expert command (m for help): 
Enter the new disk identifier: 
Disk identifier changed from 0xcb7b86f7 to 0xcb7b86f7.

Expert command (m for help): 
Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

--- Cleaning up
```

Thank you in advance.

johnnypea, 2020-04-20:
Oh, I just found this ticket https://github.com/Drewsif/PiShrink/issues/133, probably the root of a problem.

framps, 2020-04-20:
Yes, you are right. See https://github.com/Drewsif/PiShrink/issues/134 for a fix.

closed 2020-04-20 by framps