framps/raspiBackup #220: Restore DD Backup

tlrider99, 2020-04-21:
Hi ,

My restore does not work. After the DD image has been written to a new SD card and I insert it into the Raspi, the boot process always stops at the following point:
[image]

framps, 2020-04-21:
This boot issue usually has following root causes:

1. The SD card you restored to is not OK.
2. The image was not created successfully. 
3. The SD card you created the backup from has bad blocks.

tlrider99, 2020-04-21:
I tried different SD cards, different backups and 3 imaging programs, all with the same result.
I use DD backup related to the restore on windows.
The Log seems to be ok.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4511669/raspiBackup.log)

framps, 2020-04-21:
> I tried different SD cards, different backups and 3 imaging programs, all with the same result.

Well, I don't know which other imaging programs you used but I assume all of them create a dd backup. Given the fact that all create a backup which doesn't boot I think your SD card of your system has bad blocks and will die sooner or later.

I suggest to create a backup with raspiBackup by using tar or rsync asap as long as the SD card is alive.

framps, 2020-05-13:
@tlrider99 Any update on this issue?

framps, 2020-05-19:
Closing because of inactivity

closed 2020-05-19 by framps