framps/raspiBackup #222: Kommando nicht gefunden

Hofei90, 2020-05-08:
> 
* [ ] Which operating system do you use? Provide the contents of /etc/os-release

```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
`
curl -sSLO https://www.linux-tips-and-tricks.de/raspiBackupInstallUI.sh && sudo bash ./raspiBackupInstallUI.sh`

Nachdem oben gezeigter Code zum Starten der Installation im UI Modus ausgeführt wird, jedoch noch bevor sich die GUI startet ist folgende Fehlermeldung sichtbar:
`./raspiBackupInstallUI.sh: Zeile 1190: =: Kommando nicht gefunden.`

framps, 2020-05-08:
So wie es aussieht fehlt bei Dir in der Konfigdatei eines oder mehrere Definitionen die vorhanden sein muessen. Im Log kann man sehen welche es ist denn es werden die folgenden Optionen der Reihe nach durchgegangen (immer mit vorangestelltem DEFAULT_):

```
MSG_LEVEL|KEEPBACKUPS|BACKUPPATH|BACKUPTYPE|ZIP_BACKUP|PARTITIONBASED_BACKUP|LANGUAGE|STARTSERVICES|STOPSERVICES|EMAIL|MAIL_PROGRAM
```

Pruefe mal entsprechend Deine Konfigdatei.

Hofei90, 2020-05-08:
Existiert denn schon eine Konfigdatei zum Zeitpunkt der erstmaligen Installation von raspiBackup?
Denn der Fehler kommt ja, bei der Installation mit dem UI Modus

Aber es fehlt in der Tat MSG_LEVEL, und die Reihenfolge passt auch nicht. ich hab dir die Konfig mal per Mail gesendet

framps, 2020-05-08:
Bei der Erstinstallation von 0.6.5 wird eine DefaultKonfigdatei der Version 0.1.4 benutzt. Dort steht MSG_LEVEL drin [siehe hier](https://github.com/framps/raspiBackup/blob/master/config/raspiBackup_de.conf).

Hofei90, 2020-05-08:
Alle Einträge sind doch vorhanden. Falschinfo vorhin.

Kann es was damit zu tun haben, dass ich die vorherige Version von raspiBackup erst mit dem UI Installer deinstalliert habe dann neu gestartet und dann Installiert habe.

Mit einem anderen Pi habe ich diese Fehlermeldung nicht.

framps, 2020-05-08:
Ich denke nicht. Ich habe eben bei mir auch raspiBackup mit dem Installer deinstalliert und neu installiert. Allerdings hatte ich irgendwann mal in der Beta die Option rausgenommen aus der Config und dann wieder aufgenommen als es Fehler gab. Vermutlich war es sogar derselbe Fehler den Du reportest. 
Das ist ein gewissen Nachteil einer Beta: Es ist eben Beta und enthaelt u.U. noch Fehler. D.h. ich denke Du hast die Config irgendwann mal ohne MSG_LEVEL gehabt. Dann den Installer aufgerufen und die Fehlermledung erhalten. Dann deinstalliert und neue Konfig installiert und seitdem ist der Fehler weg.

Hofei90, 2020-05-08:
Gut möglich dass das so lief. Dann legen wir das hier mal zu den Akten als mysteriöse digitale Vorkommen

closed 2020-05-08 by Hofei90

framps, 2020-05-08:
Jetzt erinnere ich mich auch genauer. Die Option habe ich beim Entwickeln der Beta irgendwann rausgenommen da sie ja eigentlich immer an ist und eine gewisse Zeit speater als ich den Installer testete habe ich den Fehler festgestellt und die Option wieder in die Config aufgenommen. In genau diese zeitliche Luecke bist Du reingefallen :cry: