framps/raspiBackup #223: raspiBackup.log wird nicht erstellt / kann nicht gelesen werden

Hofei90, 2020-05-08:
```

--- RBK0009I: redmine_nextcloud_pi: raspiBackup.sh V0.6.5 (7752af8) Fr 8. Mai 07:46:35 CEST 2020 gestartet.
--- RBK0128I: Logdatei ist /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
!!! RBK0201W: 8-) Freundlicher Hinweis: Führe einen Restoretest durch. Du wirst noch 2 mal erinnert werden.
--- RBK0151I: Backuppfad /media/NAS/SD-Card wird benutzt.
--- RBK0008I: Services werden gestoppt: 'systemctl stop sps_backup && systemctl stop redis-server && systemctl stop gitea && systemctl stop apache2 && systemctl stop mariadb'.
Warning: Stopping sps_backup.service, but it can still be activated by:
  sps_backup.timer
--- RBK0081I: Backup vom Typ dd wird in /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633 erstellt.
--- RBK0085I: Backuperstellung vom Typ dd gestartet. Bitte Geduld.
15193+1 Datensätze ein
15193+1 Datensätze aus
15931539456 Bytes (16 GB, 15 GiB) kopiert, 685,348 s, 23,2 MB/s
--- RBK0078I: Backupzeit: 00:11:25.
--- RBK0007I: Services werden gestartet: 'systemctl start mariadb && systemctl start apache2 && systemctl start gitea && systemctl start redis-server && systemctl start sps_backup'.
--- RBK0159I: 3 Backups werden für den Backuptyp dd aufbewahrt.
--- RBK0205I: Ältestes Backup /media/NAS/SD-Card/redmine_nextcloud_pi in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0049I: Meldungen wurden in /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log gesichert.
--- RBK0010I: redmine_nextcloud_pi: raspiBackup.sh V0.6.5 (7752af8) Fr 8. Mai 07:58:23 CEST 2020 beendet mit Returncode 0.
--- RBK0017I: Backup erfolgreich beendet.
sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden

```

Das Backup selbst wurde erfolgreich erstellt, jedoch kommt am Ende die Fehlermeldung 
sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden.
Auch ist die Datei in dem Ordner tatsächlich nicht vorhanden. die *.msg und die *.img Datei jedoch schon.

Version: raspiBackup.sh V0.6.5 (7752af8)
Aktuelles Buster Lite auf Raspberry Pi 3B+

framps, 2020-05-08:
Hm ... das Problem wurde mir schon im Beta gemeldet und eigentlich sollte es gefixed sein :disappointed:  Kannst Du mir das DebugLog zukommen lassen (per eMail wenn Du willst)?

Hofei90, 2020-05-08:
Ist das DebugLog noch wo extra zu finden? Denn das Log ist ja nicht im Ordner zu finden

framps, 2020-05-08:
raspiBackup schreibt in einer Meldung wo das Log abgelegt wird. 
```
--- RBK0049I: Messages saved in /disks/silver/backup/rsync/asterix/asterix-rsync-backup-20200503-101248/raspiBackup.msg.                                                                                                    
--- RBK0026I: Debug logfile saved in /disks/silver/backup/rsync/asterix/asterix-rsync-backup-20200503-101248/raspiBackup.log.        
```
Ansonsten kann es noch in /root oder dem home Verzeichnis des Aufrufers - also ~ liegen.

Merkwuerdig dass ich dieses Poblem nicht nachvollziehen kann bei mir :angry:

Hofei90, 2020-05-08:
Ja, in des Message hab ich auch die Zeile:
`--- RBK0026I: Debug Logdatei wurde in /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log gesichert.
`
Aber dort ist keine Log Datei vorhanden, denn dann kommt ja der Fehler zustande:
`sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
sed: /media/NAS/SD-Card/redmine_nextcloud_pi/redmine_nextcloud_pi-dd-backup-20200508-074633/raspiBackup.log kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden`

Habe gerade nochmals ein Backup durchgeführt mit explizit angegebenen -l DEBUG. 
Die Logdatei ist während dem gesamten Kopiervorgang mit dd vorhanden, erst nach erfolgreicher Beendigung verschwindet sie aus dem Backup Ordner

framps, 2020-05-08:
Danke fuer die Analyse. Ich erstelle mal einen dd Backup und verfolge das. Ich habe nur mit rsync getestet (geht schneller). Vielleicht ist da ein Zusammenhang mit dem dd Backup.

framps, 2020-05-08:
Eben habe ich ein dd Backup erstellt und keinen Fehler bekommen. Ohne Log komme ich nicht weiter. Dann nehmen wir die Holzhammermethode die ich zwar nicht liebe denn es ist muehsam sich da durchzuwuehlen aber es gehjt nicht anders:
```
sudo -i
script raspiBackup.script
raspiBackup.sh .... 
exit
```
Dann wirst Du eine Menge Infos auf den Screen bekommen. Das wird aber gleichzeitig in raspiBackup.script reingeschrieben. Diese Datei mir dann bitte per eMail (vielleicht gzipped) zwecks Analyse zuschicken.

framps, 2020-05-12:
Ich vermute es hat wohl was mit alten Optionen zu tun. Ich teste ja immer mit einem neu aufgesetzten raspiBackup. Kannst Du mal Deine Konfiguration raspiBackup.conf zeigen?

framps, 2020-05-15:
Ein weiterer benutzer hat dasselbe Problem gefunden und hat mir sein Log zugeschickt. Demnach liegt das Problem in der Implementierung von `masqueradeNonlocalIPs`. Wenn Du die Zeile `masqueradeNonlocalIPs $LOG_FILE` auskommentierst wird das Log vermutlich erhalten bleiben.

framps, 2020-05-19:
Looks like this is some kind of interference of perl behavior and remote backuppartition usage. The nasty perl code was replaced with a plain bash implementation in master and works perfekt. The code was rolled in into the latest raspiBackup revision available on the download page.

Users facing the issue should update the current 0.6.5 version with options `-U -S`.

@Hofei90 Thank you very much for your help on this issue. Please verify and close the issue.

closed 2020-05-19 by Hofei90